tagyoureit/nodejs-poolController #203: COM port access denied on Windows

happyiphone, 2020-08-12:
**Describe the bug**
Everything installed on a windows machine. Rs485 is installed on COM6 (I can open it with RealTerm just fine and send commands and I can see the boards TX light up so there's connection to it) but when I run npm start the web server starts but at the time of opening the COM port it says Access Denied. Not sure I need something else (I also tried the CMD windows (and powershell) with Admin rights to no avail)

**Screenshots**
[image]

**Packet Capture**
Follow the instructions to complete a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) and attach the resulting zip/log files.

**Desktop (please complete the following information):**
Windows 10

rstrouse, 2020-08-12:
You don't currently have RealTerm open do you?  This is acting like COM6 is open by another program.  I assume this a USB to RS485 device so try closing all applications that may be hooking the port.  This includes any autosync programs for finding the device the plugging it back in.  Then start njspc as Admin.

happyiphone, 2020-08-12:
Yes, I've worked with Serial ports for several years in the past so I've been careful not to have the port open anywhere else. I thought it was just because I wasn't running the command box without admin rights but even with those rights it is still giving me Access Denied :(

rstrouse, 2020-08-12:
Just for shucks I plugged in a spare dongle I had laying around.  What do you see in the device manager? This really does look like something has the interrupt hooked.  I know this is cliche but try restarting your machine, then restart njsPC.

tagyoureit, 2020-08-12:
[WIKI](https://github.com/tagyoureit/nodejs-poolController/wiki/RS-485-Adapter-Details) article for how to setup & test your RS485 and it includes links for the FTDI and CH340 drivers for Windows.

happyiphone, 2020-08-13:
I think the adapter is shot. I even use another Mac to test and while that one connected to the device, it kept saying constantly that it was recovering from a lost connection. Waiting on the RA485 to USB replacements to keep testing more.

theRealMCQwerty, 2020-08-13:
@happyiphone I was getting this same error while [Grooting](https://github.com/tagyoureit/nodejs-poolController/issues/189#issuecomment-673178062).

I was only connected to a single IntelliValve but the following fixed it.

1) Updating config.json to `"rs485Port":"\\\\.\\COMX"` where `X` is the number of your COM port
2) Rebooting my IntelliValve. (In your case you may need to reboot your panel)

I have been happily Grooting for many hours straight now with no COM port errors. 

Your situation may be different so your mileage may vary but worth a shot if you are waiting for a new dongle.

happyiphone, 2020-08-13:
I've seen floating around somewhere, but didn't thought much of it, I'll give it a shot right now.

happyiphone, 2020-08-13:
While it didn't help in this case I appreciate the tip.Thanks!

happyiphone, 2020-08-16:
Losing all hope by now, yesterday I got 3 different JBKeys and tried all 3 and have the same results with all three. Installed a new Raspberry Pi 3 with Ubuntu server and reinstalled everything. I can open the port with Node-red just fine (even send the init command and receive a response) but when I run the npc I only get several retries but no actual messages sent or received. I recorded a bunch of messages but not sure this will help. Not sure where to go from there.
[2020-08-16_00-35-36.zip](https://github.com/tagyoureit/nodejs-poolController/files/5079793/2020-08-16_00-35-36.zip)

happyiphone, 2020-08-16:
Just as a follow up if I run this with Node-Red and send the query values (thanks for that pointer I would have never guessed that init command!) I get this result back:

[165,23,15,16,2,29][1,45,0,0,0,0,0,0,0,0,3,0,64,4,72,72,0,0,56,50,0,0,0,0,0,247,235,0,13][4,88]

And I know that this is valid because if I parse byte by byte I can correlate valid values from the device so is not that I'm having issues with the Rs485 port, cable, USB key, etc.