tagyoureit/nodejs-poolController #205: Virtual controller does not recover if shut down unexpectedly

jarrah42, 2020-08-18:
**Describe the bug**
After restarting the controller when it is shut down unexpectedly (pm2 stop or crash), it is no longer able to control the pump.

**To Reproduce**
Steps to reproduce the behavior:
1. Start the controller and a pump
2. Stop the controller using `pm2 stop` or kill it
3. Restart the controller
4. It is no longer possible to start the pump, even though it is still configured

**Expected behavior**
The controller should reset its state and the state of the pumps (e.g. to off) when it is first started.

**Workaround**
The only way to get the controller working again is to stop it, remove both the poolConfig.json and poolState.json files, restart the controller, and reconfigure the UI.

tagyoureit, 2020-08-19:
Can you please attach a packet capture?  I need to go outside to my pumps to replicate this but the air quality here in California is really dangerous right now so trying to stay indoors. Thx.

jarrah42, 2020-08-19:
[replay-1.zip](https://github.com/tagyoureit/nodejs-poolController/files/5098756/replay-1.zip) packet capture when pump can be started/stopped

[replay-3.zip](https://github.com/tagyoureit/nodejs-poolController/files/5098757/replay-3.zip) packet capture after restarting controller; unable to start pump

jarrah42, 2020-08-24:
I updated from next and the pump doesn't start at all now. Here's the log:

[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/5118640/replay.zip)

jarrah42, 2020-08-24:
I can confirm that master is able to start the pump, but next doesn't.

tagyoureit, 2020-08-25:
Hm, I put your config files in my app and started up (on `next`) and it is trying to communicate with the pumps.  (I'm not outside now.)

However, in your logs I see: `[8/24/2020, 10:59:22 AM] error: undefined`.  I think we're hitting an error and we should be restarting but probably don't have the right context.  I just pushed up a change to restart the pump cycle differently.  Can you try again?  The app should go through the pump cycles now but I may need to do some more digging to find out why the undefined error happened (if it can be reproduced consistently).

jarrah42, 2020-08-25:
I've noticed that if the pump stops for a while the chlorinator starts spitting out messages like:

warn: Message aborted after 3 attempt(s): 16,2,80,17,28,143,16,3
warn: error with chlorinator: Message aborted after 3 attempt(s): 16,2,80,17,28,143,16,3

Not sure if that has anything to do with it. Anyway, I'll give it a try shortly and let you know. Thanks!

tagyoureit, 2020-08-25:
That could mean the error is coming from the comms/messages components more than the pumps/chlorinators themselves.

jarrah42, 2020-08-25:
The latest version seems to have resolved the issues. The pump starts now, and when I stop the controller and restart it, I can restart the pump. The only unusual thing I saw was this message when I set the chlorinator:

[8/25/2020, 5:09:24 PM] error: schlor.get is not a function

Thanks!

closed 2020-09-06 by tagyoureit