tagyoureit/nodejs-poolController #208: docker, docker-compose, and kubernetes/helm support

andylippitt, 2020-09-04:
I've created a docker/kube configuration that will allow for hosting the api, the web client, and an ssh tunnel to a pool-connected-raspberrypi

andylippitt, 2020-09-04:
I've rearranged projects a bit, so I'm not sure if a pull request is approriate. please see: https://github.com/andylippitt/nodejs-pool

tagyoureit, 2020-09-04:
Thanks for taking this on.  I'm not a Docker guy (want to learn more) so will dig into this a bit.  Fundamentally, how does this differ from what @wurmr did [here](https://github.com/tagyoureit/nodejs-poolController/issues/119#issuecomment-630574898)?

andylippitt, 2020-09-05:
a Dockerfile (which @wurmr built) is the first step. It describes how to build the system, sort of like old Makefiles.

The next step is trying to figure out how to host those images that are built with the Dockerfile.
There are two prevailing methods, docker-compose and kubernetes. Both of them allow you to describe how images interact and should be run in a production environment, how to update the various bits, etc. 

In both of those environments, you're usually inside of some virtual machine or container or imaginary network and can't access any physical hardware, so you'll need to escape with a network solution to get access to the hardware.

I expanded a fair bit on the idea of using socat on a raspberry pi to publish the ports. My system requires only an ssh key, and user@piserver plus the name of a serial device, and it will create an ssh tunnel that works with your controller.  If someone already has a network device, the ssh tunnel is easy enough to disable, but probably useful in most installs.

So my docker-compose setup describes how to run the controller, webclient, and the ssh tunnel in a single package, complete with easy access to the configuration separate from the published images. This is basically setup for simple, quick development. the default configuration I setup all points to localhost:8080 and localhost:4200 like your defaults.

In the kubernetes side of things, I arguably undid some of the recent work you've done separating the webclient from the api. but only in that I copy the pre-built parcel output from react directly into the controller server. I removed the server side of the react client running at 8080. this requires a one liner in my PR about adding a static serving path to express.

The other major thing was the addition of the ncc build tool. npm and node_modules dependency tress have become black holes. your current node_modules path after an `npm i` on the controller weighs >150k files and nearly 1GB of storage.  This is quite a thing to ask of a little raspberry pi.  The ncc tool does to the controller what parcel does to react. It collects all the modules that are actually in use, and concatenates them into a small set of files. the output of which weighs about 8MB for the controller. just `node index.js` and you have no other dependencies other than config. this gives us tiny output, that starts instantly, but it also frees the hosting environment from requiring npm.

my final image for the unified set weighs only 65MB for the entire image. https://hub.docker.com/layers/andylippitt/pool_unified/0.26/images/sha256-e100f9d844e10d73d0618580c115c149ab21854ec3eb28ad80e72e74eac9fc84?context=explore

tagyoureit, 2020-09-07:
I'm leaning towards including the dockerfile and dockerignore.  That ncc plugin looks promising but it is just too new, has too many open issues, and it running on beta dependencies (and has over 90 itself).  

Can you modify the PR to just include those two files.  I'd like to move most of the instructions to the wiki.  Are [these](https://github.com/andylippitt/nodejs-pool) all the users need to do to roll their own?  If you want to include instructions for how to include ncc for a smaller package please do so and we can revisit including that if it becomes stable in the future.

andylippitt, 2020-09-08:
A couple options on ncc. I have it set as a new npm run command, therefore shouldn’t change anyone’s existing processes unless they change their calls to npm. If you’d still like to keep it out of package.json and node_modules, it could be added back in as part of the (or one of several) dockerfile.

andylippitt, 2020-09-08:
They are all the instructions needed for those familiar with docker-compose and helm charts. There’s a lot of reading between zero and there, I could find some good examples to link to, but would mostly be basics in docker documentation.

The dockerfiles describe solid build routines (understandable if your not ready to trust ncc), and can gain you entrance to free autobuilds from hub.docker.com because of open source status. So every commit to github, or release might trigger a new build and publishing of a set of images to a docker account. That should probably be your’s, not mine.

The bits in my repo are relevant only to the docker-compose and kuberenetes crowds, and arguably a thing to leave to others after you include the dockerfiles, but they belong in your repo for sure. If you want to control version numbers and use docker as more of a release, distribution mechanism, you’re more than half way there, just need some accounts and github triggers.

wurmr, 2020-09-08:
FWIW the images published here are only 54MB not using `ncc`. The Dockerfile I published just uses the standard build system with `npm ci`.

https://hub.docker.com/repository/docker/wurmr/nodejs-pool-controller/tags?page=1

Another thing to keep in mind is if you want multi-arch images you won't get arm32v6 images using the above Dockerfile.

I currently run my stack inside a docker swarm using compose files that are easily built out. I agree there is benefit to giving people example compose and/or kub configs.

Finally, In the spirit of Kub/Swarm I think there is benefit to not unifying the API and Web (as this project has done). That allows you to scale/deploy them independently. It also stays with the Docker idea of one container one task.

tagyoureit, 2020-09-08:
There are quite a few people out there using @rstrouse's dashpanel instead of my webclient.  I do like the idea of providing the base files but we should allow people the choice of which client to install.

andylippitt, 2020-09-09:
The recombination of the two systems as part of the build is trivial and is not in the Dockerfiles that I PRed. Instead it happens only in a separate image I’m building called unified. Though, that addition weighs only 6mb of static html and scripts sitting around unused if undesired, not a lot of baggage if you are using a different client. I intend to use the API directly and ignore the react client myself.

If there is a more standard way to get reasonable image sizes, I’m all for it. Not advocating for ncc, just small output.

Re, multi-arch. I’m successfully building for amd64 and arm/v7 with buildx. I haven’t tried to target the v6 chips, wonder how many of those devices can competently run docker or containerd.  @wurmr  Your Dockerfile seems to suggest the problem is a missing node arm6 build? I’d argue that’s probably out of scope for solving within this repo.

wurmr, 2020-09-09:
> Re, multi-arch. I’m successfully building for amd64 and arm/v7 with buildx. I haven’t tried to target the v6 chips, wonder how many of those devices can competently run docker or containerd. @wurmr [...]

arm32v6 is the Pi Zero and Pi Zero W, the reason I build to that is that is the exact device I have hanging off my pool equipment. It handles containderd/docker loads fine for projects like this. The nice thing about that device in this application is I'm getting the power for it right off the low voltage on the Pentair panel.

--

I agree with the effort. I think the Dockerfile should be in the repo, that allows for using github actions to trigger a buildx run (or whatever CI system you'd want to tie in). My external repo and dockerhub solution is very manual because I cannot get triggered events from this repo.

As far as helm charts and the like, I love it. I think charts, example kubernetes configs and swarm/compose configs always help people get up and going and lower the barrier of entry. It also allows for much easier troubleshooting because typically folks will follow the examples and if they have issues you can help steer them back toward the known working configurations.

tagyoureit, 2020-09-17:
I included these files with the 6.1.0 release I pushed last night.  Thanks to both of you.

closed 2020-09-17 by tagyoureit