tagyoureit/nodejs-poolController #210: Not able to get Chlorinator data

fgonza2, 2020-09-08:
Hi, first of all, this is awesome work, thanks a lot!!

I am seeing an issue where i can see any data from my chlorinator.   I get the following in the web client:

`Intellichlor--40 (1)
Off
Salt
0 ppm
Current Output
0 %
Setpoint
undefined% / undefined%
Status
Ok`

On the JSON GET REST call:
 `
{
    "id": 1,
    "lastComm": 1599586453670,
    "targetOutput": 17,
    "saltLevel": 0,
    "saltRequired": 0,
    "status": {
      "val": 0,
      "name": "ok",
      "desc": "Ok"
    },
    "type": {
      "val": 0,
      "name": "pentair",
      "desc": "Pentair"
    },
    "name": "Intellichlor--40",
    "equipmentType": "chlorinator"
  }
`

All other functions seem to be working ok.  

**Pool Equipment**
 - Controller: IntelliTouch
 - Controller Model: EasyTouch2 4
 - Controller Software Rev:
     UIC: Not Avail - UOC: 2.160
 -Controller Bootloader Rev:
     UIC: Not Avail - UOC: 1.010
 - Pump(s) manufacturer and model: IntelliFlo VSF
 - Chlorinator:  IntelliChlor-40
 - Heater(s): gas

**Desktop (please complete the following information):**
 - OS: MacOS
 - Browser Safari

**Additional context**
I can control and see the chlorination level directly from the easytouch panel directly

Not sure if this is a bug or something i am missing on the configuration somewhere.   Thanks a lot

tagyoureit, 2020-09-09:
Can attach a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) so we can review.  Alternatively, if you are still on the 6.0.0 release you can check out the `next` branch which has a ton of bug fixes and likely includes this one.  The master branch should be updated with the latest code in the next week or so.

fgonza2, 2020-09-13:
seems to be working now.   I am working on an integration with home assistant

closed 2020-09-13 by fgonza2

tagyoureit, 2020-09-14:
We've been working on MQTT (see `mqtt` branch) and have had very many discussions about this in the Gitter rooms.  Users of both HomeSeer and HomeAssistant are successfully using this.  What type of integration are you working on?

fgonza2, 2020-09-15:
hi, i did an integration via restapi see [screenshot](https://i.imgur.com/kjT2oxq.png).   I am also using an RS-485 to wifi box, which is rugged to put in the pool equipment box and needs no maintenance vs. a RPI [see here](https://www.amazon.com/BGNing-USR-W610-Ethernet-Wireless-Converter/dp/B07DNWM62H/ref=sr_1_7?dchild=1&keywords=rs485+wifi&qid=1600183257&sr=8-7)

i am interested in the MQTT option as well assuming in can post update messages on its own rather than polling like it is now via rest.   I could find clear documentation, somehow it says that it needs to be updated for 6.0 to work so i didnt go deeper.   Can you point me to the documentation for PQTT or gitter chat?   i'd be happy to help and contribute as well

tagyoureit, 2020-09-15:
Nice implementation.  And cool wifi device - I hadn't seen that before.  If you go to [gitter](https://gitter.im/nodejs-poolController/Lobby) you can search or just scroll up to the last few conversations on MQTT.  [MQTT documentation](https://github.com/tagyoureit/nodejs-poolController/wiki/Bindings-Integrations-in-2.0#mqtt) was from the very first release so the basics are there but it is much more complete now.

tagyoureit, 2020-09-18:
I just pushed up 6.1 to master and MQTT is included with it.  [Directions in Wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/Bindings-Integrations-in-2.0#mqtt).