tagyoureit/nodejs-poolController #211: IntelliCenter firmware 1.045 allows a circuit to be turned off in a circuit group

george-basco, 2020-09-10:
Recent firmware, e.g. 1.045, allows a circuit to be turned **ON**, **OFF** or **IGNORE** in a circuit group.  As far as I can tell there is no support for this and current code assumes that circuits can only be turned on.

Here is a screenshot of the intellicenter.com interface:
[image]

Here are the packets from setting the circuit "Main Suction S" to **ON**, then **OFF**, and finally **IGNORE** from the intellicenter.com interface:
[circuits.log](https://github.com/tagyoureit/nodejs-poolController/files/5204052/circuits.log)

And here are replay files with "Main Suction S" set to **IGNORE** as shown above:
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/5204035/replay.11.zip)

**Pool Equipment**
 - Controller: IntelliCenter
 - Controller Model: IntelliCenter i10D
 - Controller Firmware Version: 1.045

rstrouse, 2020-09-11:
Any clue as to the purpose of the ignore?  Wouldn't this simply be the same as not including the circuit?  Off makes sense but I am trying to figure out what ignore might possibly do other than nothing.  The stubs already exist in njspc to handle this when the firmware is actually released but I might go after it before then if I can find a string of milliseconds to get my head around the intended function.

george-basco, 2020-09-11:
Didn't make sense to me either.  Maybe so you can temporarily ignore a circuit without having to remove it?

rstrouse, 2020-09-11:
Maybe...  Still the same number of clicks.  Maybe it makes more sense in the mobile app where you don't have to press save.  Oh wait for circuit groups you do have to press save.  May have to get those darn kids involved... where is that mystery machine when you need it?

george-basco, 2020-09-17:
FYI, 1,047 is available for download from their website.

rstrouse, 2020-09-27:
I added the requisite changes to support this new function.

rstrouse, 2020-09-27:
I am closing this.  Open a new issue if you find issues with the implementation.

closed 2020-09-27 by rstrouse