tagyoureit/nodejs-poolController #212: Running /state/circuit/setState produces an error but still works

stupid48, 2020-09-11:
**Describe the bug**

Trying to run /state/circuit/setState via curl to turn a circuit on but getting an error.  The circuit still turns on however so the function is working correctly.  Just curious what it means....:

{"stack":"TypeError: circuit.get is not a function\n    at /opt/nodejs-poolController/web/services/state/State.ts:56:72\n    at runMicrotasks (<anonymous>)\n    at processTicksAndRejections (internal/process/task_queues.js:93:5)","message":"circuit.get is not a function","level":"\u001b[31merror\u001b[39m"

**To Reproduce**
Steps to reproduce the behavior:
1. Run: 

curl -X PUT "http://192.168.1.50:3000/state/circuit/setState/" -H  "accept: application/json" -H  "Content-Type: application/json" -d "{\"id\":3,\"isOn\":1}"

2. Error:

{"stack":"TypeError: circuit.get is not a function\n    at /opt/nodejs-poolController/web/services/state/State.ts:56:72\n    at runMicrotasks (<anonymous>)\n    at processTicksAndRejections (internal/process/task_queues.js:93:5)","message":"circuit.get is not a function","level":"\u001b[31merror\u001b[39m"

**Expected behavior**
The on/function works but produces an error response. 

**Pool Equipment**
 - Controller: IntelliTouch
 - Controller Model: EasyTouch2 8P
 - Controller Firmware Version: [e.g. 1.047]
 - Pump(s) manufacturer and model: IntelliFlow 2 VST 011056
 - Chlorinator: IntelliChlor-40
 - Heater(s): solar

tagyoureit, 2020-09-13:
This was fixed/refined in 6.0.1 (currently the `next` branch - soon to be pushed to `master`).  However, you have some malformed JSON.  You can't have double quotes inside double quotes without escaping them.

Try `curl -X PUT "http://192.168.1.50:3000/state/circuit/setState/" -H "accept: application/json" -H "Content-Type: application/json" -d "{\"id\":3,\"isOn\":1}"`

closed 2020-09-13 by tagyoureit

stupid48, 2020-09-13:
Thanks. That’s weird because I copied the URL from your Swagger. Thanks for the info though.

tagyoureit, 2020-09-14:
Did you get it from here: https://tagyoureit.github.io/nodejs-poolcontroller-api/#/state/put_state_circuit_setState_

or did you import the swagger into another app?  I'm not sure exactly what/how Postman will translate the calls so it's good to be aware of.

stupid48, 2020-09-14:
Yes I got it from you’re swagger definition. It works but just has the error you fixed. Thanks for your help.