tagyoureit/nodejs-poolController #214: temperature calibration changes are not tracked correctly in sys.equipment.tempSensors

george-basco, 2020-09-18:
``tempSensors.setCalibration()`` is not getting called on a user change.  This causes calibration to be out of sync until next reload.
https://github.com/tagyoureit/nodejs-poolController/blob/41ab9efb0eb275bd97edf59d56d315f86c5438ba/controller/boards/IntelliCenterBoard.ts#L798

Should be something like:

```
sys.equipment.tempSensors.setCalibration('water1', sys.general.options.waterTempAdj1 = parseInt(obj.waterTempAdj1, 10))
```

Maybe an issue here as well?
https://github.com/tagyoureit/nodejs-poolController/blob/41ab9efb0eb275bd97edf59d56d315f86c5438ba/controller/comms/messages/config/ExternalMessage.ts#L623

george-basco, 2020-09-18:
A separate issue is that IntelliCenter firmware 1.047 broadcasts adjusted temps vs. true temps.  This results in a doubling of the adjustment.  The OCP broadcasts the adjusted value and then njspc adjusts it again.  Not sure if this is new to 1.047.

https://github.com/tagyoureit/nodejs-poolController/blob/41ab9efb0eb275bd97edf59d56d315f86c5438ba/controller/comms/messages/status/EquipmentStateMessage.ts#L399

rstrouse, 2020-09-21:
Yes this is new in 1.047. It appears that it now operates like IntelliTouch.  I’ll post a fix this week after I get back in town.

rstrouse, 2020-09-25:
George,

I am going to need some help here.  I need to know what the action 168 message looks like when you change the calibration for waterSensor2 and solarSensor2.  Specifically, I need to know where the bytes are located.  Specifically, I need to know what the byte is at index 2 and which byte is actually being changed (this will be between byte 3 and 13).  Below is what it looks like when you change the waterSensor1 and solarSensor1 calibration.
[image]
Perhaps when I know this I can interpolate the waterSensor3 & 4 and the solarSensor3 & 4.

george-basco, 2020-09-26:
For action 168 and a waterSensor2 calibration change, payloadByte(2) == 4 and calibration value is at index 7
For action 168 and a solarSensor2 calbiration change, payloadByte(2) == 5 and calibration value is at index 8

In an action 2 message, solar2 is located at index 17, right before air and solar1

Here is a packet capture of actions 168 and 2 ([packets.log](https://github.com/tagyoureit/nodejs-poolController/files/5286431/packets.log)).   The calibration changes made are:
1. waterSensor2++
2. waterSensor2++
3. waterSensor2--
4. waterSensor2--
5. solarSensor2++
6. solarSensor2++
7. solarSensor2--
8. solarSensor2--

I don't understand why the Solar2 temp in the action 2 packets falls and rises with the increase and decrease in calibration.  But will note that there is no sensor connected at this point.

george-basco, 2020-09-26:
Hooking up solar2 to an actual sensor, clears up the confusing behavior above.

[packets.log](https://github.com/tagyoureit/nodejs-poolController/files/5287437/packets.log)

[image]

rstrouse, 2020-09-27:
These should be calibrating correctly now.  Please grab njspc and -dashPanel.

george-basco, 2020-09-27:
In dashPanel, the "True Readout" calculation is not correct.  Need to subtract ``sensor.tempAdj``:
https://github.com/rstrouse/nodejs-poolController-dashPanel/blob/02d713b90b006d96d54a25352121b3b0e5d7b23b/scripts/config/general.js#L195

In this case the "True Readout" for Water Sensor 1 should be 78, not 82:
[image]

rstrouse, 2020-09-27:
Yeah, this was a bit more insidious than that.  The wierdness comes due to the fact that the whole time you are doing the adjustment the water temperature is changing.  This means the adjusted value is changing.  Originally, the True Readout used to bounce around but that was goofy as well.  

I like the way you are dealing with temperatures not offsets (IntelliTouch does this where you set +- degrees for the calibration) but the adjusted value is changing behind the scenes on you.  I originally thought that I could constantly keep track of the offset but that was weird as well.  So the interface now takes a snapshot when you load it then applies the proper adjustments to the server.  After the postback it checks to see if the offset took and only changes the values back if the new offset fails.  Maybe their Hart calculation fails with a rounding error on occasion due to the sensor accuracy and the fact that we are dealing with integers.

george-basco, 2020-09-29:
On the following two lines, ``state.temps.solar`` should be ``state.temps.solarSensor2``

https://github.com/tagyoureit/nodejs-poolController/blob/dd9abc561f35cfd9d385edade082b4bdb224b595/controller/boards/SystemBoard.ts#L911

https://github.com/tagyoureit/nodejs-poolController/blob/dd9abc561f35cfd9d385edade082b4bdb224b595/controller/boards/SystemBoard.ts#L928

rstrouse, 2020-09-29:
@george-basco - Thanks for this.  It would be much more difficult to get this ironed out if you didn't have the skills to see the references in the code.  I made the changes and posted them up.

george-basco, 2020-09-30:
Math for negative temp calibrations is missing parentheses around ``(byte & 0x007F)``.

https://github.com/tagyoureit/nodejs-poolController/blob/3b00da3cc436c5ed4dc085f7d181ad66b64a882d/controller/comms/messages/config/ExternalMessage.ts#L633

george-basco, 2020-10-01:
https://github.com/tagyoureit/nodejs-poolController/blob/348121cac82cf0a5e4266de1b54648cf2c2d60e1/controller/comms/messages/config/OptionsMessage.ts#L58-L66

In my i10D system, the ``water2`` and ``solar2`` calibration values are in bytes 6 and 7 respectively.  In the packets below, you can see the before and after in the 30 responses.   The first 168 request is changing water2 calibration from 0 to +5  The second 168 request is changing solar2 calibration from 0 to -5.  The other indices for water1, solar1 and air are all correct.

[image]

rstrouse, 2020-10-01:
Wow I had this shifted by -1 byte starting with waterSensor2 and assumed it matched up with the 30[13] payload order.  This was because the initial firmware mirrored the waterSensor1 in byte 2.  And also did this in the Action 2 like it did for IntelliTouch i10D this appears now to only be uninitialized data.  Maybe that assumption is incorrect as well, but as far as we know so far it goes body 1, body 3, body 2, then body 4.  I need someone with an XCP installed to cement those assumptions but I know the body 1 and 2 positions are correct.

This now makes much more sense as the Water & Solar Sensors are in pairs and appear to follow the OCP and XCP order.
3 = waterSensor1
4 = solarSensor1
5 = airSensor
6 = waterSensor2
7 = solarSensor2
8 = waterSensor3 ?
9 = solarSensor3 ?
10 = waterSensor4 ?
11 = solarSensor4 ?

rstrouse, 2020-10-03:
@george-basco - These all work correctly given the information I have.  Could you verify that they do what they are supposed to do for i10D.

george-basco, 2020-10-04:
yes, all looks god

rstrouse, 2020-10-04:
Awesome closing this one out

closed 2020-10-04 by rstrouse