tagyoureit/nodejs-poolController #215: Not detecting intelliclor

baudfather, 2020-09-23:
[2020-09-22_07-06-53.zip](https://github.com/tagyoureit/nodejs-poolController/files/5265285/2020-09-22_07-06-53.zip)
**Describe the bug**
Haven't been able to detect controller since 6.0.0

**To Reproduce**
Steps to reproduce the behavior:
Clean install of 6.1.0, mqtt enabled.  Wiped poolconfig/state json files; npm start

**Expected behavior**
IC-20 expected to be detected

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Packet Capture**
Attached
**Pool Equipment**
 - Controller: Easytouch 4p
 - Controller Firmware Version: unknown
 - Pump(s) manufacturer and model: VSF
 - Chlorinator:  IntelliChlor-20]
 - Heater(s): gas
 - Chemical controller: none
 - Valves: none

**Desktop (please complete the following information):**
 - OS: [chrome os, windows 10]
 - Browser chrome
 - Version latest

**Smartphone (please complete the following information):**
 - Device: Android Pixel XL
 - OS: Android 10
 - Browser [chrome beta]
 - Version latest

**Additional context**
Add any other context about the problem here.

baudfather, 2020-09-23:
Okay, so it's detecting only after I make a change at the OCP (opened an issue on this in the past), but I'm no longer able to control the IC-40 from the webclient.
[2020-09-22_20-11-34.zip](https://github.com/tagyoureit/nodejs-poolController/files/5265357/2020-09-22_20-11-34.zip)

baudfather, 2020-09-23:
Also just realized the pool company replaced my IC20 with an IC40.

tagyoureit, 2020-09-27:
I'm seeing that we discover the IC-40 with salt level 3300 but there aren't any packets in either log where the OCP/chlorinator are discussing the output %.  I also don't see where you are trying to control the IC-40.

Can you tell me more about what you mean by "it's detecting only after I make a change at the OCP"?  What change are you making?

Separately, I'm seeing numerous "AUX 1" circuits and the same schedule repeating over and over.  Is this what's set on your OCP?

[image]

baudfather, 2020-09-27:
Sorry, forgot that I wrote this in a hurry.  What's happening is upon startup (after a clean install), it's detecting everything except the IC-40 - that is, until I change the output %, then it picks it up and populates the status in njspc.  I'm not sure why there's so many AUX1 circuits - I tried removing unused circuits in a previous version and for some reason they all defaulted to AUX1 - something I'll have to go in and edit sometime.  I don't use the OCP schedules - originally just one was used just for the lights, but now that I can use MQTT to control it I've just disabled them entirely - I'm not sure if there's a way to remove each entry at the OCP, I'll have to read up on that.  Totally unrelated (hopefully) - ever since 6.1.0 the system seems to randomly turn off the pool body once a day - I'm hoping to track this down in the next week before I have to winterize - not sure what's triggering it since 6.1.0 also coincided with the IC40 replacement and I wasn't here when the tech installed it.

rstrouse, 2020-09-27:
Not to hijack this but I am willing to bet that your pool is shutting off at 5pm every day.   The default times are 8am-5pm for Pentair perhaps your MQTT code is not sending the id for the schedule and filling the next slot when you edit it.  I can't remember does *Touch allow you to create more than one schedule per circuit?  I seemed to remember that it did but don't remember how to clear a schedule.  Page 11 of the EasyTouch manual describes a mode selection that has new but perhaps it also has a delete.

Also, the newer versions of IntelliChlor don't seem to respond right away on startup.  njspc will pick up the first valid 16,2 message and start processing the chlorinator which will in turn make sure it exists in njspc.  I don't see anywhere where we detect whether the chlorinator is enabled for *Touch njspc won't detect the chlorinator until it actually responds.  Not sure where that byte is but I remember that in IntelliTouch you could enable and disable the chlorinator in the menus and it appears that option seems to also exist in EasyTouch but I don't think we process that byte.

baudfather, 2020-09-27:
@rstrouse - I wish it were the same time every day - it seems entirely random (usually overnight or morning) - I can usually find the last reported status via MQTT in Homeseer's timestamp (12:26, 08:33; 02:11 etc). I have to read up on the schedules when I have a chance, but from what I can tell there's no circuts assigned to any of them.

tagyoureit, 2020-09-28:
njsPC is usually pretty true to what is shown on the OCP.  We wouldn't show `pool` as being on a schedule so many times if it didn't exist on OCP.  2 things to check on your side - first, if the schedules are there for some reason I'd avoid anything going past midnight.  EG set a schedule to go to 11:59pm and then restart at 12:01am.  Second, I wonder if the egg timers are coming into play (even the default ones) and shutting off the pool after 12 or 24 hours?

To @rstrouse's comment:
> Also, the newer versions of IntelliChlor don't seem to respond right away on startup.

I do see the IC-40 responding very quickly.  It's just that the OCP isn't telling the Chlorinator to output at any % so we aren't picking it up.  

> I don't see anywhere where we detect whether the chlorinator is enabled for *Touch 

We pick it up at [ChlorinatorStateMessage.ts#L131](https://github.com/tagyoureit/nodejs-poolController/blob/master/controller/comms/messages/status/ChlorinatorStateMessage.ts#L131)

rstrouse, 2020-09-28:
@tagyoureit - In IntelliCenter there are two messages that are sent with the configuration options for the chlorinator.  These include action 30 byte(0) = 7 which is either asked for or broadcast on error.  The other is used to set the chlorinator settings.  It is action 168 byte(0) = 7.  It is sent by any controller that wishes to change the current settings for it.  The only real difference between the two is that 30 is a broadcast that identifies the current settings and 168 is an action to set them.  For IntelliCenter I grab both of them and scrape the data off it.  If it is rejected by the OCP then it will immediately follow with a 30 message containing the current values.  If it is successful, then there is sometimes an ACK(168) but it is not 1 for 1 when somebody is sending a bunch of updates.

baudfather, 2020-10-03:
Okay, so it seems I have a gremlin somewhere - ever since filing this issue the random pool shutdowns have stopped, so I guess that's a good thing?  

In regards to the chlorinator, it is enabled at the OCP, I've got it set pretty low (5%) since things are winding down and the cover is on most of the time.  As far as I know I'm not using any of the scedules - I've got to see if there's a way I can delete/wipe them, but they're not assigned to anything currently as far as I know.  Hopefully will have a chance to check this and clear this issue up before I need to winterize.  Thanks again for your help!

baudfather, 2020-10-09:
Unfortunatley never got a chance to look into this any further - had a short window of opportunity to close the pool and had to shut it down for the winter.  Feel free to close this, if I have any issues when spring rolls around I'll bring it up again.  Cheers :)

closed 2020-10-09 by tagyoureit