tagyoureit/nodejs-poolController #218: Setting the egg timer to "don't stop" on a circuit doesn't get broadcast out correctly

george-basco, 2020-09-27:
line 1297 needs to be moved before 1295 so that Hrs and Mins are properly updated.

https://github.com/tagyoureit/nodejs-poolController/blob/f249c6741b2e835b5c76b723db945e668676bb5d/controller/boards/IntelliCenterBoard.ts#L1295-L1297

rstrouse, 2020-09-27:
There was some weirdness with this in IntelliCenter but maybe there was some cleanup in 1.047.  If you sent 24 in the hours column it immediately sent 50+ action 30 messages as if it didn't quite like having both the don't stop byte set and the hours set to 24.  For groups the don't stop byte didn't exist but perhaps they normalized this.  It is weird that they limit this to 24 hours anyway as what's wrong with manually turning on a circuit for 2 or 3 days.

I'll have a look to see if the behavior changed.  Btw this was different between features and groups.

rstrouse, 2020-09-27:
Yep circuit groups still do not have a don't stop flag and the hours can now be set to 24 without causing the system to go busy sending information that is not required.  While it appears that the ICP still does do some synchronization it doesn't do it across all categories.  This used to crash the mobile app and it still does if you send too many changes in a row with the hours/minutes.  It looks like it doesn't debounce the values rather it will try to send rapid changes in a row which will crash it as all the panels clamor to get the poop in a group.  They optimized a bunch of this in 1.047 but it looks like they still have some chattiness to deal with.

I checked in the change to include Circuits, Features, Light Groups, and Circuit Groups.

rstrouse, 2020-10-06:
Closing after verifying with OCP output.

closed 2020-10-06 by rstrouse