tagyoureit/nodejs-poolController #220: Remove pumps that don't exist

miamijerry, 2020-09-28:
How do I remove the pumps showing in attached pictures?

There are no pumps
[image]

tagyoureit, 2020-09-28:
Have you pulled `next` with the updated OCP configuration?  This issue should resolve itself once the OCP is configured properly.

miamijerry, 2020-09-28:
This multiple pump showing issue is on a Aux 8 P/N 520657 motherboard.

miamijerry, 2020-09-28:
Earlier this RPI was connected to the 522523 Easytouch Lite board on work bench.

Then I moved the RPI outside and connected to Easytouch Aux 8

The issue with the pumps has not resolved itself...

miamijerry, 2020-09-28:
Resolved by replacing SD card image from before connection with 522523 Easytouch Lite PL4 board

closed 2020-09-28 by miamijerry