tagyoureit/nodejs-poolController #223: Intellicenter gas heaters are missing "Cooldown Delay" attribute

george-basco, 2020-09-29:
njspc doesn't seem to be aware of the "cooldown delay" attribute for gas heaters.

Here are the packets sent out when the iOS app changes the delay from 5 (the default) up to 8 and back down to 5:

[image]

rstrouse, 2020-09-29:
Another discrepancy from the web page.  There is no cooldown delay on a gas heater in the IC webpage.  It does appear however on the mobile app.  Btw, this option should not be turned on if you have a Pentair branded gas heater.  It air cools the heat exchanger.

george-basco, 2020-09-29:
The OCP also shows the option.  I have a Raypak heater and looks like a delay is recommended.

rstrouse, 2020-09-29:
That would be true for RayPack.  There should also be an override in the options for it but it looks like it has no bearing on whether the option appears.  I think it simply ignores it when you don't have it checked.

rstrouse, 2020-09-29:
I added cooldown delay support for gas heaters.  I looked through my documentation and it looks like they added another message to the heater output.  This moved the economy time to this new message offset and added the cooldown delay in the middle.  So as of 1.023 the heater configuration didn't even include this byte.  It could have been added in 1.040 though.

Anyhow grab njspc and dashpanel.  This acts like the mobile app in that the field is displayed regardless of whether you have that delay enabled.  I may go back and only display it when the delay is enabled in the future.

rstrouse, 2020-10-11:
Closing this after adding the cooldownDelay attribute which is now part of fw 1.047.

closed 2020-10-11 by rstrouse