tagyoureit/nodejs-poolController #224: Heat mode is not displayed correctly with Intellicenter running 1.047 firmware

george-basco, 2020-09-29:
[image]

GET /state/all returns:
 `` 'heatMode' => {
                                                       'desc' => 'Off',
                                                       'name' => 'off',
                                                       'val' => 2
                                                     }``

The ``name/desc`` is not correct for `` 'val' => 2``

However, GET /config/body/1/heatModes is correct:
 ``[{"val":1,"name":"off","desc":"Off"},{"val":2,"name":"heater","desc":"Heater"},{"val":3,"name":"solar","desc":"Solar Only"},{"val":4,"name":"solarpref","desc":"Solar Preferred"}]``

rstrouse, 2020-09-29:
Uggh.  This is because of the change between 1.040 and 1.047.  These identifiers have changed believe it or not 2 used to not be defined.  What used to be 0,3,5,21 changed to 1,2,3,4.  If you simply change it then change it back the pool state should update its text.

rstrouse, 2020-10-03:
Can you let me know if the aforementioned procedure resolves the issue.

george-basco, 2020-10-03:
Yes, once you make a change, things work fine.  Just an issue on clean restarts.

rstrouse, 2020-10-03:
Yeah, I hope its pretty rare that they change the underlying identifiers.  Thanks.

closed 2020-10-03 by rstrouse