tagyoureit/nodejs-poolController #227: Creating a new run-once schedule doesn't work with 1.047

george-basco, 2020-10-02:
Creating a run-once schedule seems to get rejected by the OCP.  Here is the iOS app creating such a schedule, followed by njspc attempting to create another with the same parameters:
[image]

The difference is the value of the runOnce byte:
https://github.com/tagyoureit/nodejs-poolController/blob/8d1b9e19b5abc6c9204ac05f8eea4746125f3f07/controller/boards/IntelliCenterBoard.ts#L2825-L2827

Don't understand the logic for runOnce, but changing it to 129 does get the schedule registered with the OCP.  No idea if this is new to 1.047..

rstrouse, 2020-10-02:
Yeah I caught the changes to the heatSource but missed the runOnce flag.  I think this was probably me coding around an error in 1.023 where the schedule type was mismatched between the action 30 messages and the 168.  In 168 bit 8 needed to be cleared for a run-once and bit 1 set.  In the 30 it only looked at bit 8 to determine run-once or not.  Like there was a misplaced bit shift in the code.

I think they are still struggling with this feature as the original fw had the ability to set offsets to sunrise/sunset (but it didn't work).  Then 1.040 came around and the heatSource was tragically broken.  No matter what you did it set the temperature to 78 on the body.  This appears to work correctly now.  If you look closely at that message there are two bytes at the end (78,100) that don't seem to have a home.  I suspect that the 100 is supposed to be a cooling setpoint for body schedules when they get around to coding it.  That leaves 2 missing bytes for sunrise/sunset offsets.  Perhaps they are encoded into start stop bytes but the way they set the sunrise/sunset is that every time the schedule ends it sends out a 168 to modify the schedule times to sunrise/sunset.

I am in the middle of reworking the 168 sequence for multi-message commands so that they allow for the reaction from other panels and don't get confused.  When I finish that up I will hit this up.

rstrouse, 2020-10-03:
I fixed this with the last update.  Please pull njspc and dashPanel.  Could you also please let me know if this resolves the issue for you?  Thanks.

george-basco, 2020-10-03:
Yes, works great now.  Thanks!

rstrouse, 2020-10-03:
Excellent!  Thanks

closed 2020-10-03 by rstrouse