tagyoureit/nodejs-poolController #228: PH and ORB valves to not update without reloading page - DashPanel

miamijerry, 2020-10-03:
Intelichem readings for PH and ORB do not update in dashPanel without a page reload. 

They do update in webClient without a page reload.

rstrouse, 2020-10-03:
I'll have a look.  Are you referring to the setpoints or the levels.  I do not have an IntelliChem so it is kinda difficult to test beyond mocking it up.  Btw did you used to have a Vera home automation system with Autelis?

rstrouse, 2020-10-03:
Pull a new dashPanel.  I think I may have found the issue.

miamijerry, 2020-10-03:
Yes, I have the Autelis unit also, I moved it from Vera to Homeseer.
[image]
The 7.1 and 685 numbers are what I was referring to.

rstrouse, 2020-10-03:
The death of Autelis led me here.  I wrote the Autelis integration with Vera and seemed to remember your id.  I also wrote an integration with njspc to Vera as well.

Did you pull an new dashPanel?

miamijerry, 2020-10-03:
I asked you to write an integration of njspc to Homeseer recently.
(I have one dead, and two living Autelis units)

Pulled a new dashPanel, and the problem still persists

rstrouse, 2020-10-04:
Just to be sure.  Hard refresh your browser.  I can’t seem to figure out how to not cache scripts in nodejs.

miamijerry, 2020-10-04:
Just realized that I did not reboot the RPI after the dashPanel pull.

[image]

Now there is no connection. shows "unknown model".
messageManager shows "ready" in top right corner.
webClient is working...

miamijerry, 2020-10-04:
Contains of themes folder

[image]

miamijerry, 2020-10-04:
This got a working copy up

[image]

miamijerry, 2020-10-04:
OK, I find it working this morning.

Hope all this makes sense to someone.

closed 2020-10-04 by miamijerry