tagyoureit/nodejs-poolController #229: Turning IC circuit groups on/off doesn't work on first attempt

george-basco, 2020-10-06:
If you turn on a circuit group from the OCP or iOS app, and then later attempt to turn it off via njsPC, it requires two attempts for the action to stick.   Will demonstrate with an example.

In this case there is a circuit group that consists of the single AUX1 circuit with a desiredState of 1.  No other circuits are turned on during this test.  Here's the capture:
[image]
packet 849 is the iOS app turning on the circuit group.  byte 14 shows the circuit group turned on, and byte 3 has AUX1 on
packet 879 is njsPC turning off the circuit group, and it leaves AUX1 alone
packet 885 is the OCP announcing that circuit group is still on
packet 914 is the second attempt from njsPC (via dashPanel user click) to turn off the circuit group
packet 917 shows that OCP accepts it this time and shows AUX1 turned off

If you turn the circuit group off from OCP, iOS app or intellicenter.com, they all work correctly.  But none of those are using 168 requests to turn it off.

Here's what the ICP does to get it to work with 168s:
[image]
packets 030 and 031 are the result of iOS turning on the circuit group.  Sometimes these show as a single packet like above
packet 124 is ICP turning circuit group off.  It also turns off AUX1 (different from njsPC, but don't think it's relevant).
packet 130 shows the OCP again announcing the group is still on, just like the njsPC case
packet 132 - immediately with no user interaction, ICP responds with a second turn-off packet
packet 134 - OCP accepts it this time

This behavior is very consistent.  Happy to provide more detail if this isn't easy to reproduce.

Seems like the solution would be to have njsPC respond like ICP does when OCP comes back with a "still on" 168.  Not pretty, but it's what ICP does.  A better solution might be to use the webSockets interface whenever possible.

rstrouse, 2020-10-07:
I'll have a look at this.  I have experienced the same weirdness with the wireless controller.   The key here is that 168 is on the broadcast source channel so that is most commonly an echo from the OCP.  This occurs randomly and only when the system sits idle for a while but not every time.  Most of the time I can submit a command and it takes right away.

If you still have this sequence up, can I get you to take a look at the timestamps and see the time difference between when the 130 message was received and the 132 was received.  There was a message sandwiched in between these.

There is also much more to this message than meets the eye.  If you look at that series of 255s these are supposed to be the circuit synchronization bytes.  I have yet to figure all of the bits out but these are bitmasks where 2 bit pairs represent specific values.  These come into play when you are synchronizing light groups for instance.

1.047 also added 4 additional bytes to this message.  These are the last 4 bytes.  They signify something I just do not know what but at times they will contain a 1 in the first second or 3rd position or all of them 0.  I have never witnessed them with either the 4th one set or more than one at a time set.  Also, please note that there were 5 messages bandied around the bus after the ACK was sent to 32 (ICP 0).  Perhaps 2 of them were 2 and 204 but that would be a span of 4 seconds.

If you look at your last 4 bytes they are all 0.  When I first ran the same test you ran mine was 0,1,0,0 then I cycled a few times with IntelliCenter.com and it switched to 0,1,1,0.  That is where it is now.

rstrouse, 2020-10-07:
And now I have experienced a different byte pattern on those bytes.  Only once and cannot replicate it.
[image]

george-basco, 2020-10-07:
In my case njsPC always fails to successfully change circuit group state if it was previously changed by iOS app.  If I turn it on with njsPC, then turning it off with njsPC works fine.

Any missing packets are most likely pump packets, otherwise 2/204's.

I've always never noticed anything in those last 4 packets, although hadn't really looked at them closely.  Looking at them now, I have only seen 0's.

Here is a complete picture of what happens.  Again, I can replicate this every time.

[image]

george-basco, 2020-10-07:
No idea if this is useful or accurate in my description, but at very least will help me keep track of what I've seen.

This first case is turning a circuit group on and then off from dashPanel:
[image]
In response to both of njsPC 168's, the OCP puts out its own 168, it seems because njsPC didn't turn on/off AUX1.

In this next case, dashPanel again turns the circuit group on, but this time ICP turns it off:
[image]
Again the OCP responds with a 168 to let everyone know that AUX1 has been turned on.
But this time when ICP turns the circuit group off (it also turns off AUX1) , the OCP just acks it and doesn't send out a 168, I suppose because everyone already knows AUX1 is off based on the ICP 168.

This third case is where the circuit group is turned on and then off by the ICP:
[image]
In this case the OCP just acks the two 168s and doesn't send out any of its own.  I again assume this is because ICP is taking care of the AUX1 state and the OCP has no need to update any state.  So if njsPC were to also turn on/off AUX1 in its 168 packets, we likewise wouldn't see any OCP 168s to complete the on/off process.

This fourth case is where ICP turns the circuit group on and njsPC turns it off:
[image]
When njsPC attempts to turn off the circuit group, ICP responds with a 168 saying turn it back on.  I think what happened is that OCP turned off the circuit group (hence the ack) and was going to turn off AUX1 and send out a 168, but ICP was quick enough to send out the 168 turning the circuit group back on.  In this case AUX1 never actually was turned off.  Verified by listening for the relay to flip.

This fifth case is the same as the previous one.  This is a less common outcome:
[image]
In this case the ICP is too slow and the OCP is able to turn off AUX1 and send out its 168 in response to njsPC turning off the circuit group.  The ICP still sends out its 168 saying, wait turn it back on.  In this case you hear AUX1 getting switched off and then back on.  Then ICP then asks for some info.  We end up in the same final state as the previous case where the circuit group and AUX1 are still on after the njsPC request to turn it off.

To summarize, both ICP and OCP don't like it when someone else turns off a circuit they had turned on themselves.  If you ask them a second time, they seem to acquiesce.  They accept the initial turn-off without complaint as long as they didn't turn it on.

george-basco, 2020-10-07:
Based on the fourth and fifth case in previous post, I wanted to see if turning off AUX1 from njsPC would change anything.  I don't think it did, but saw something different than what I normally see.

In this first case, I replayed a njsPC turn-off 168 edited to turn off AUX1 as well.
[image]
Surprisingly the ICP has nothing to say about this.  This is the first time I've seen this behavior while paying attention.  As I've said before, when njsPC tries to turn off something it didn't turn on, it almost always fails (I had thought it was always).

I did the exact same thing again:
[image]
This time the result was more expected with the ICP saying not so fast, turn things back on.  I thought maybe AUX1 would actually turn off momentarily as the OCP ack'ed it, but it did not.  Seems that the ICP 168 was quick enough to prevent OCP from cycling AUX1 off and back on.

This of course just goes to show how inconsistent this stuff is, which I know you know too well.  But I am interested in pursuing this a bit because how reliably njsPC actions fail when turning things off that someone else turned on.  At least on my system.

george-basco, 2020-10-07:
Realize I didn't answer your question above about the "time difference between when the 130 message was received and the 132 was received"

Here is the same scenario where iOS turns on circuit group.  Then turn it off on ICP, where ICP tries once, OCP rebuffs, ICP retries and is successful:
[image]
The time delta between packets 453 (rebuff) and 456 (retry) is 0.224 seconds.

george-basco, 2020-10-07:
Continuing with the example of turning on/off a circuit group.  This time between the OCP and ICP.

First case is OCP turning it on, then off:
[image]

Now OCP turns it on, and ICP turns it off:
[image]

ICP turns it on and off:
[image]

ICP turns it on, OCP turns it off:
[image]

What's interesting is that both panels object if the other panel turns off what they had turned on.  It objects by turning it back on, but then the turning-off panel tries again and is successful in its second attempt to turn it off.

george-basco, 2020-10-07:
All the examples I've been showing here have been one UI turning something on, followed by the same or different UI turning it off.   If you flip that and first turn something off, then turn it back on, you see the same behavior as documented in these test cases.  Which of course makes sense, just wanted to clarify.

george-basco, 2020-10-08:
Thought this behavior was only the case for circuit groups.  I had tried feature circuits and didn't see the problem.  However, plain circuits like AUX1 have the same behavior where OCP and ICP both restore the state when njsPC tries to change the state of something previously set by the OCP or ICP.  So in my case both circuits and circuit groups both consistently show this behavior.

rstrouse, 2020-10-12:
I think we may have uncovered a bug in the firmware.  The reason I say this is that I can get my wireless controller (WCP) and indoor controller (ICP) to fight over the state.  The ICP seems to try again if the WCP sasses it but the WCP requires two requests and it sometimes takes several seconds for the button to go back out again.  It seems to depend on how long ago you made the change on the ICP.

I may just mimic the behavior of the ICP.  It looks like it simply waits to see if another controller resets the 168[15] message.  If  168[15] contains an appropriate bit after they set it then it sets it back and kills the listener.  If you do it again the listener isn't there so it doesn't set it back.  Smells like a buggy rat to me.

rstrouse, 2020-10-14:
More information on this.  It appears that this can be delayed for up to 2 seconds.  It doesn't matter if it is njspc, WCP, or ICP.  The controller that originally turned on the circuit will attempt to turn it back on again if you turn it off.  

This is definitely new with 1.047.   This has to be some sort of mechanism to ensure the state gets set to what the target is.  Since the entire process is synchronous, I suspect that they wanted the reset to expire at some point but it appears to do that only after an extended period of time.  Only speculation but perhaps they set their timer in seconds but assumed it was milliseconds or microseconds.  

The difference between 500 milliseconds and 500 seconds is an extraordinary difference.  Half a second vs 8 minutes.  I could hold my breath for both but the latter I would only do once.

tagyoureit, 2020-11-02:
Closing for now.  Please re-open if this continues to be an issue.

closed 2020-11-02 by tagyoureit