tagyoureit/nodejs-poolController #231: Settings for rs485Port in different situations?

miamijerry, 2020-10-12:
In config.json on a Raspberry PI
the rs485 port is configured as;

{
  "controller": {
    "comms": {
      "rs485Port": **"/dev/ttyUSB0",**
      "mockPort": false,
      "netConnect": false,
      "netHost": **"raspberrypi",**
      "netPort": 9801,
      "inactivityRetry": 10,
      "portSettings": {
        "baudRate": 9600,
        "dataBits": 8,
        "parity": "none",
        "stopBits": 1,
        "flowControl": false,
        "autoOpen": false,
        "lock": false
      }
    }
  },

How would the config.json be set for a JBtek CH340 USB connected to a Windows computer ;
[image]

Also this device USR-TCP232-304;
[image]

What would the config.json entry be?

[image]

And what would I use for;
Local port ?
Remote port ?
Work mode ?
I assume that "Remote Server Address" is going to be computer running poolController.

rstrouse, 2020-10-12:
I haven't used that adapter but if it communicates over a socket then it may work to set netConnect to true, the address for the rs485Port as 192.168.0.201 and the netPort as 8234.  The socket on the other end is typically socat so I don't know if it will work.

rstrouse, 2020-10-12:
Oh and for the JBTek that setting is ```"rs485Port": COM4``` or ```"rs485Port":"\\\\.\\COM4"``` depending on the root settings for the port with ```"netConnect": false```

miamijerry, 2020-10-14:
Was able to use CH340 serial adaptor on a windows 10 computer with config.json set to "com10".

Also was able to use a RS485 to RJ45 adapter as a server at the pool equipment, and setup a virtual com port as a client on the windows 10 computer, Working fine!
https://www.troublefreepool.com/threads/pentair-intellicenter-pool-control-dashboard-instructional-guide.218514/page-4#post-1927837

closed 2020-10-14 by miamijerry