tagyoureit/nodejs-poolController #232: IC-40 Chemistry Section Missing

tamu20, 2020-10-13:
**Describe the bug**
Dashpanel Chemistry section is not displayed & MQTT is not reporting salt level.  poolState and API list the salt level

**To Reproduce**
Steps to reproduce the behavior:
1. Go to dashpanel on port 5150

**Expected behavior**
Chemistry section to be displayed and MQTT report salt level

**Screenshots**
N/A

**Packet Capture**
[2020-10-12_19-14-03.zip](https://github.com/tagyoureit/nodejs-poolController/files/5371947/2020-10-12_19-14-03.zip)

**Pool Equipment**
 - Controller: IntelliCenter
 - Controller Model: IntelliCenter i5PS
 - Controller Firmware Version: 1.047
 - Pump(s) manufacturer and model: Intelliflow VSF

 - Chlorinator: IntelliChlor-40
 - Heater(s): gas
 - Chemical controller: N/A
 - Valves: Intellivalve
 - Any other relevant equipment:

**Desktop (please complete the following information):**
 - OS: OS X
 - Browser chrome

tagyoureit, 2020-10-15:
MQTT bindings are [here](https://github.com/tagyoureit/nodejs-poolController/blob/master/web/bindings/mqtt.json#L161).

rstrouse, 2020-10-15:
I added salt level to the mqtt.json bindings.  I followed the existing format.

closed 2020-10-30 by tagyoureit