tagyoureit/nodejs-poolController #233: npm start, not working on windows 10 installation.

miamijerry, 2020-10-13:
Get the following when attempting to start poolController on a windows 10 computer;

C:\Users\NUC8i7HVK\nodejs-poolController>npm start&

> nodejs-poolcontroller@6.1.0 start C:\Users\NUC8i7HVK\nodejs-poolController
> npm run build && node dist/app.js

> nodejs-poolcontroller@6.1.0 build C:\Users\NUC8i7HVK\nodejs-poolController
> tsc

Error reading configuration information.  Aborting startup: SyntaxError: Unexpected token c in JSON at position 54

C:\Users\NUC8i7HVK\nodejs-poolController\config\Config.ts:33
            this._cfg = fs.existsSync(this.cfgPath) ? JSON.parse(fs.readFileSync(this.cfgPath, "utf8")) : {};
                                                           ^
SyntaxError: Unexpected token c in JSON at position 54
    at JSON.parse (<anonymous>)
    at new Config (C:\Users\NUC8i7HVK\nodejs-poolController\config\Config.ts:33:60)
    at Object.<anonymous> (C:\Users\NUC8i7HVK\nodejs-poolController\config\Config.ts:127:31)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! nodejs-poolcontroller@6.1.0 start: `npm run build && node dist/app.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the nodejs-poolcontroller@6.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\NUC8i7HVK\AppData\Roaming\npm-cache\_logs\2020-10-13T22_02_27_395Z-debug.log

C:\Users\NUC8i7HVK\nodejs-poolController>

Contains of debug.log;

0 info it worked if it ends with ok
1 verbose cli [ 'C:\\Program Files\\nodejs\\node.exe',
1 verbose cli   'C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js',
1 verbose cli   'start' ]
2 info using npm@5.6.0
3 info using node@v8.11.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle nodejs-poolcontroller@6.1.0~prestart: nodejs-poolcontroller@6.1.0
6 info lifecycle nodejs-poolcontroller@6.1.0~start: nodejs-poolcontroller@6.1.0
7 verbose lifecycle nodejs-poolcontroller@6.1.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle nodejs-poolcontroller@6.1.0~start: PATH: C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin;C:\Users\NUC8i7HVK\nodejs-poolController\node_modules\.bin;C:\Program Files\Git\cmd;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\nodejs\;C:\Users\NUC8i7HVK\AppData\Local\Microsoft\WindowsApps;C:\Users\NUC8i7HVK\AppData\Roaming\npm
9 verbose lifecycle nodejs-poolcontroller@6.1.0~start: CWD: C:\Users\NUC8i7HVK\nodejs-poolController
10 silly lifecycle nodejs-poolcontroller@6.1.0~start: Args: [ '/d /s /c', 'npm run build && node dist/app.js' ]
11 silly lifecycle nodejs-poolcontroller@6.1.0~start: Returned: code: 1  signal: null
12 info lifecycle nodejs-poolcontroller@6.1.0~start: Failed to exec start script
13 verbose stack Error: nodejs-poolcontroller@6.1.0 start: `npm run build && node dist/app.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\index.js:285:16)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at EventEmitter.emit (events.js:214:7)
13 verbose stack     at ChildProcess.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\lib\spawn.js:55:14)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at ChildProcess.emit (events.js:214:7)
13 verbose stack     at maybeClose (internal/child_process.js:925:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
14 verbose pkgid nodejs-poolcontroller@6.1.0
15 verbose cwd C:\Users\NUC8i7HVK\nodejs-poolController
16 verbose Windows_NT 10.0.18363
17 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "start"
18 verbose node v8.11.3
19 verbose npm  v5.6.0
20 error code ELIFECYCLE
21 error errno 1
22 error nodejs-poolcontroller@6.1.0 start: `npm run build && node dist/app.js`
22 error Exit status 1
23 error Failed at the nodejs-poolcontroller@6.1.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

rstrouse, 2020-10-13:
@miamijerry -- you have an error (most likely a misplaced comma or quote) in your config.json.  Copy the contents and paste it into this website and it will highlight the issue.  https://jsonlint.com/

miamijerry, 2020-10-14:
OK needed "com10" instead of com10

Still have issues;
poolController

C:\Users\NUC8i7HVK\nodejs-poolController>npm start&

> nodejs-poolcontroller@6.1.0 start C:\Users\NUC8i7HVK\nodejs-poolController
> npm run build && node dist/app.js

> nodejs-poolcontroller@6.1.0 build C:\Users\NUC8i7HVK\nodejs-poolController
> tsc

Init state for Pool Controller
(node:9212) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'isProxy' of undefined
    at Object.get (C:\Users\NUC8i7HVK\nodejs-poolController\controller\State.ts:43:36)
    at new EqState (C:\Users\NUC8i7HVK\nodejs-poolController\controller\State.ts:439:29)
    at new EquipmentState (C:\Users\NUC8i7HVK\nodejs-poolController\controller\State.ts:585:1)
    at State.init (C:\Users\NUC8i7HVK\nodejs-poolController\controller\State.ts:325:26)
    at C:\Users\NUC8i7HVK\nodejs-poolController\app.ts:34:34
    at <anonymous>
    at process._tickCallback (internal/process/next_tick.js:188:7)
    at Function.Module.runMain (module.js:695:11)
    at startup (bootstrap_node.js:191:16)
    at bootstrap_node.js:612:3
(node:9212) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:9212) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[2020-10-13 20:24:46] info: Serial port: com10 opened
[2020-10-13 20:24:46] info: Serial port: com10 request to open succeeded without error

C:\Users\NUC8i7HVK\nodejs-poolController\controller\State.ts:43
                    if (util.types.isProxy(val))
                                   ^
TypeError: Cannot read property 'isProxy' of undefined
    at Object.get (C:\Users\NUC8i7HVK\nodejs-poolController\controller\State.ts:43:36)
    at State.get isInitialized [as isInitialized] (C:\Users\NUC8i7HVK\nodejs-poolController\controller\State.ts:302:68)
    at Function.process (C:\Users\NUC8i7HVK\nodejs-poolController\controller\comms\messages\status\EquipmentStateMessage.ts:336:20)
    at Inbound.processBroadcast (C:\Users\NUC8i7HVK\nodejs-poolController\controller\comms\messages\Messages.ts:467:39)
    at Inbound.process (C:\Users\NUC8i7HVK\nodejs-poolController\controller\comms\messages\Messages.ts:582:22)
    at SendRecieveBuffer.processInbound (C:\Users\NUC8i7HVK\nodejs-poolController\controller\comms\Comms.ts:403:29)
    at SendRecieveBuffer.processPackets (C:\Users\NUC8i7HVK\nodejs-poolController\controller\comms\Comms.ts:219:21)
    at Timeout.timers_1.setTimeout [as _onTimeout] (C:\Users\NUC8i7HVK\nodejs-poolController\controller\comms\Comms.ts:192:109)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! nodejs-poolcontroller@6.1.0 start: `npm run build && node dist/app.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the nodejs-poolcontroller@6.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\NUC8i7HVK\AppData\Roaming\npm-cache\_logs\2020-10-14T00_24_47_880Z-debug.log

debug.log

0 info it worked if it ends with ok
1 verbose cli [ 'C:\\Program Files\\nodejs\\node.exe',
1 verbose cli   'C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js',
1 verbose cli   'start' ]
2 info using npm@5.6.0
3 info using node@v8.11.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle nodejs-poolcontroller@6.1.0~prestart: nodejs-poolcontroller@6.1.0
6 info lifecycle nodejs-poolcontroller@6.1.0~start: nodejs-poolcontroller@6.1.0
7 verbose lifecycle nodejs-poolcontroller@6.1.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle nodejs-poolcontroller@6.1.0~start: PATH: C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin;C:\Users\NUC8i7HVK\nodejs-poolController\node_modules\.bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\nodejs\;C:\Users\NUC8i7HVK\AppData\Local\Microsoft\WindowsApps;C:\Users\NUC8i7HVK\AppData\Roaming\npm
9 verbose lifecycle nodejs-poolcontroller@6.1.0~start: CWD: C:\Users\NUC8i7HVK\nodejs-poolController
[... 21 lines cut ...]
22 error nodejs-poolcontroller@6.1.0 start: `npm run build && node dist/app.js`
22 error Exit status 1
23 error Failed at the nodejs-poolcontroller@6.1.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

dashPanel

TypeError: f.isFile is not a function
    at Function.getBackgrounds (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\dist\server\upload\upload.js:193:19)
    at app.get (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\dist\server\Server.js:167:60)
    at Layer.handle [as handle_request] (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\layer.js:95:5)
    at next (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\route.js:137:13)
[... cut ...]
    at trim_prefix (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\index.js:317:13)
    at C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\index.js:284:7
    at Function.process_params (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\index.js:335:12)
    at next (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\index.js:275:10)
    at SendStream.error (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\serve-static\index.js:121:7)

rstrouse, 2020-10-14:
In the directory for njspc and dashPanel type ```npm install```.  It appears you do not have the supporting libraries installed.

miamijerry, 2020-10-14:
TypeError: f.isFile is not a function
    at Function.getBackgrounds (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\dist\server\upload\upload.js:193:19)
    at app.get (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\dist\server\Server.js:167:60)
    at Layer.handle [as handle_request] (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\layer.js:95:5)
    at next (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\route.js:137:13)
    at Route.dispatch (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\route.js:112:3)
    at Layer.handle [as handle_request] (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\layer.js:95:5)
    at C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\index.js:281:22
    at Function.process_params (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\index.js:335:12)
    at next (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\index.js:275:10)
    at jsonParser (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\body-parser\lib\types\json.js:110:7)
    at Layer.handle [as handle_request] (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\layer.js:95:5)
    at trim_prefix (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\index.js:317:13)
    at C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\index.js:284:7
    at Function.process_params (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\index.js:335:12)
    at next (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\express\lib\router\index.js:275:10)
    at SendStream.error (C:\Users\NUC8i7HVK\nodejs-poolController-dashPanel\node_modules\serve-static\index.js:121:7)

rstrouse, 2020-10-14:
type ```node --version```.  It looks like you are running an out of date node version.  The current version is v12.x.x.

miamijerry, 2020-10-14:
Yes, was at 8.x Installed 12.x, had to do "npm rebuild"

Thanks for the incredible assistance !

rstrouse, 2020-10-14:
Your welcome.  Been down that rabbit hole before.

closed 2020-10-14 by rstrouse