tagyoureit/nodejs-poolController #236: Lost serialport Comm after version upgrade

jgaro, 2020-10-15:
Windows 10, Node 12, next version 6.1.1. Upgraded from next 6.0. Lost communication to pump

`[10/15/2020, 10:57:59 AM] info: Starting Replay Capture.
Init state for Pool Controller
[10/15/2020, 10:57:59 AM] info: Server is now listening on 0.0.0.0:4200
[10/15/2020, 10:57:59 AM] info: Serial port: \\.\COM10 request to open succeeded without error
[10/15/2020, 10:58:00 AM] info: [10:58:00 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:00 AM] info: [10:58:00 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:00 AM] info: [10:58:00 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:00 AM] info: [10:58:00 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:00 AM] info: [10:58:00 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:01 AM] info: [10:58:01 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:01 AM] info: [10:58:01 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:02 AM] info: [10:58:02 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:02 AM] info: [10:58:02 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:03 AM] info: [10:58:03 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:03 AM] info: [10:58:03 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:04 AM] info: [10:58:04 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:04 AM] info: [10:58:04 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:05 AM] info: [10:58:05 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:06 AM] info: [10:58:06 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:07 AM] info: [10:58:07 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 10:58:08 AM] info: [10:58:08 AM] 192.168.1.100 GET /state/all/ {}`

rstrouse, 2020-10-15:
Do a git pull on 6.1.1.  I rearranged the reconnect code a bit.  I was also able to recreate your issue by moving my adapter to a higher interrupt and disconnecting IntelliCenter so that is a good sign.

jgaro, 2020-10-15:
after the first time running this, I saw: 
`[10/15/2020,` 11:49:47 AM] info: Found pump at 1 address 96
[10/15/2020, 11:49:48 AM] info: [11:49:48 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 11:49:49 AM] info: [11:49:49 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 11:49:50 AM] info: [11:49:50 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 11:49:51 AM] info: [11:49:51 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 11:49:52 AM] info: [11:49:52 AM] 192.168.1.100 GET /state/all/ {}
[10/15/2020, 11:49:53 AM] info: [11:49:53 AM] 192.168.1.100 GET /state/all/ {}`

But on subsequent terminate/restarts, I'm not seeing that (and no "DISPLAY" on the pump LCD

rstrouse, 2020-10-15:
Huh.  From the listing above it is communicating.  Can you please perform this operation so I can look at the rs485 traffic?

https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution

rstrouse, 2020-10-15:
The process above should have created a replay.zip file.  Can you upload that here?

jgaro, 2020-10-15:
I have other issues with the webclient (see: https://github.com/tagyoureit/nodejs-poolController-webClient/issues/8) so I was running this from windows cmd. I think this is the one though? 
[2020-10-15_12-47-54.zip](https://github.com/tagyoureit/nodejs-poolController/files/5386867/2020-10-15_12-47-54.zip)

rstrouse, 2020-10-15:
Perfect!  It looks like our comms are doing what they should.  What is confusing the pump configuration is that it doesn't know what kind of pump is on the other end.

Did you set up the pump in webClient?
https://github.com/tagyoureit/nodejs-poolController/wiki/Virtual-Pump-Controller---v6
Perhaps this configuration was lost between the upgrade.  But it clearly doesn't know what kind of pump is on the other end.  This is important as the messaging structure is fundamentally different.

jgaro, 2020-10-15:
1. On the first run of the poolController, it does seem to work normally. However, if I terminate and restart, the server does not appear to attempt to connect to the pump again. Perhaps this is normal?
2. I've previously just used the poolController via API and only set up the webClient to troubleshoot it. But after the upgrade, I have this error when trying to run it: `'env' is not recognized as an internal or external command,
operable program or batch file.`

rstrouse, 2020-10-15:
You know @tagyoureit may likely not see your comment on closed issues.  Issue #8 has been closed for some time.  You should submit another issue but I believe that fsevents module is supposed to be an optional dependency as it only operates on MacOS.  Funny how it claims to be cross-platform when it clearly isn't < 15% of all users.

It is normal that njspc will not attempt to get the existence of pumps after it has them already.  It also searches for other equipment out there the first time it runs.   However, could you open a command prompt and type ```node --version```.  You need to be running > 10.x.x.x but I would suggest you run the latest LTS which is 12.x.x.x.

As far as sending an api call this is ```PUT config/pump {id: 1, type: 128, circuits:[{ id:1 circuit:6, speed: 2700, units:0}]}```
This will set the type to IntelliFlo VS and set the pool circuit to 2700.  To get the configuration options for pumps ```GET config/options/pumps```

jgaro, 2020-10-15:
yep! running the latest LTS, v12.19.0

jgaro, 2020-10-15:
maybe this is some clue about pump communication? `[10/15/2020, 2:44:28 PM] info: [2:44:28 PM] 192.168.1.100 GET /config/options/pumps {}
[10/15/2020, 2:44:28 PM] error: Cannot read property 'hasHeatSource' of undefined`

rstrouse, 2020-10-15:
Ok so this is more than a clue it points directly at the problem.  The type for the pool circuit is incorrect.  It must be 2.  I will be posting up a change in a few minutes but it will require you to delete poolState.json and poolConfig.json then restart.

rstrouse, 2020-10-15:
Ok do a git pull on njspc.  This type had been incorrect for a long time but the type was just recently being used to determine whether the overall circuit may be affected by a heater.

jgaro, 2020-10-15:
works perfectly now, thanks so much!

closed 2020-10-15 by jgaro

rstrouse, 2020-10-15:
You're welcome.  That was a round-about but it did allow me to do some cleanup on the serial port code.