tagyoureit/nodejs-poolController #243: chemController 0 or 1 ??

miamijerry, 2020-10-30:
I can not get poolController communicating with my Intellichem.

I see this in terminal;
[10/30/2020, 3:32:16 PM] info: Stopping chemController 0.
[10/30/2020, 3:32:16 PM] info: Starting Virtual Chem Controller: Chem id:1/address:144

Is that correct??

I had the communication working for a few minutes, a couple of days ago. I thought it might have been the fact that I was using a two wire RS485 device, tried today with 3 wire RS485, still not working to Intellichem. 

[10/28/2020, 4:17:39 AM] warn: No response from IntelliChem id:1/address:144

Seeing extra 00000 in the setpoint requests, but some do come thru as 7.2;

[10/30/2020, 3:56:53 PM] info: [3:56:53 PM] 192.168.0.106 PUT /state/chemController {"id":1,"pHSetpoint":7.200000000000001,"orpSetpoint":700,"alkalinity":25,"calciumHardness":25,"cyanuricAcid":0,"saturationIndex":-0.99,"acidTankLevel":0,"orpTankLevel":0}

Also in webClient the settings for chemController appear stuck on "Standalone" and I was able to switch to "Connected to OCP'' a few days ago.

tagyoureit, 2020-10-31:
Please attach logs per the instructions/wiki so I can go through them.  I really can't diagnose any issues on the comments that things are stuck and they worked a few days ago.  https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution

miamijerry, 2020-10-31:
Re-installed webClient, was able set intellichem as "Connected to OCP". started capture per your instructions.

Intelichem communication is working very solid today. 

Only issue remaining is lost of webClient ip:8080 page, which I believe is occurring after multiple reboots...

If I lose intellichem communication i will post logs.

Thanks

closed 2020-10-31 by miamijerry