tagyoureit/nodejs-poolController #244: Spillway on easytouch2 4

emlowe, 2020-11-01:
Spillway circuit doesn't seem to be detected. I also have a schedule to run the spillway and this appears as "unnamed"

[image]
[image]

[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/5471526/replay.zip)

tagyoureit, 2020-11-02:
Hi, unfortunately the replay.zip file is corrupt and I can't open it.  Are you able to try again?  If you have problems with the automatic download you can manually zip up the files in the /logs and /data directory in the njsPC app.  thx.

emlowe, 2020-11-02:
[log.zip](https://github.com/tagyoureit/nodejs-poolController/files/5476837/log.zip)

This one seems ok (chrome/safari on macos seem to mangle it) - did this using curl

rstrouse, 2020-11-02:
@tagyoureit is it possible that the invalid ids array should not include 5 as an invalid id?  That would make 1-6 for circuits with 1 being spa and 6 being pool.  Ultimately the panel would have AUX1 - AUX4 starting on circuit id 2 and ending on circuit id 5.  So EasyTouch2 4 would be a panel with 4 aux circuits + shared pool/spa.  

This would also match up with IntelliTouch i5 and IntelliCenter i5 panels.  The difference with these two and this ET2 4 is that circuit id 1 and 6 can be assigned to any combination of pool or spa + 3 macro circuits.

emlowe, 2020-11-02:
[image]
 Does this help at all

rstrouse, 2020-11-02:
Not really, the 18 circuits referenced there are including both relay and feature circuits.  I assume that the Aux 4 means that it is an aux circuit (short for aux relay circuit) but I have never owned an EasyTouch.  I have had IntelliTouch and currently have IntelliCenter.  

In terms of your circuit map I see
1 = Spa
2 = Cleaner
3 = Pool Light
4 = Spa Light
5 = This should be Spillway but we are skipping it.
6 = Pool
-- This ends all the circuits that are attached to a relay (Pool/Spa are attached to the filter relay)
11 = Feature 1
12 = Feature 2
13 = Feature 3
14 = Feature 4
...
20 = Aux Extra (Typically used for solar)

emlowe, 2020-11-02:
I'm pretty sure it's a 4 - here is the sticker
[image]

tagyoureit, 2020-11-02:
@rstrouse entirely possible that 5 should not be excluded.

@emlowe Pull the `next` branch.  I just made a quick change your model.  If it looks good, I'll also adjust for all the other ET4 models.

johnny2678, 2020-11-03:
Not sure if this is related, but I have some feature circuits missing on the latest `next` branch (1st screenshot) that are present on my easytouch and still show in `5.3.x` (2nd screenshot)

[image]
[image]

Let me know what info would help here.  Packet capture? or did I just misconfigure somehow?

tagyoureit, 2020-11-03:
Sure... a packet capture would be great.  @johnny2678 do you have two circuits both named "Pool Low"?  And did you pull the latest `next` branch today and try again?  And which model EasyTouch do you have?

johnny2678, 2020-11-03:
Apparently I do.  Here's the screenlogic setup for my easytouch:
[image]
[image]

I can play with the settings and try and remove one of the `Pool Low`s

Still missing the `Pool Higher` and `Pool High`

Not sure if I ran the packet capture long enough.  Let me know if you need more.
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/5482119/replay.zip)

tagyoureit, 2020-11-03:
Unfortunately your zip is corrupt, too.  It has something to do with the browser/platform combination.  But I see what the issue is and can look at the replay from @emlowe (but feel free to attach a working copy of yours as well).

@rstrouse - I was correct that the ET4 doesn't have an Aux4.  Is this still the same as the Intellicenter i5?  The spillway can be a feature and doesn't need to be a circuit.  I'll take a look now.

johnny2678, 2020-11-03:
ok, I need @emlowe 's curl cmd to get the packet capture.  Tried again with firefox and that zip is corrupt too.

tagyoureit, 2020-11-03:
So, um, errr...
[image]

The fix for adding (not skipping) Circuit 5 works for @emlowe's ET4 but there is no Circuit 5 (Aux 4) for @johnny2678's ET4.  Thoughts?

> ok, I need @emlowe 's curl cmd to get the packet capture. Tried again with firefox and that zip is corrupt too.
The log files will live and be zipped up in the njsPC/logs directory.  You should be able to just copy/upload them.  If you have Mac you can use `rsync source dest` eg `rsync -avzr pi@ipaddress:/home/pi/nodejs-poolController/logs ./local-directory` (make appropriate substitutions).

johnny2678, 2020-11-03:
Ok, here's a zip of the logs folder 
[logs.zip](https://github.com/tagyoureit/nodejs-poolController/files/5482478/logs.zip)

tagyoureit, 2020-11-03:
Pull the latest `next`.  I eased up on the restrictions skipping the extra feature circuits.  I swear there was a brochure that I used to guide me through this but of course now I can't find it.

@johnny2678 There is one schedule that says "Unknown Name" which looks like it was referencing Circuit 11 (Pool Low) which is now 'Not Used' (meaning we don't store it).  Is that schedule still active on your side?  If not, I can add some logic to skip schedules that reference these 'not used' circuits.

[image]

rstrouse, 2020-11-03:
> @rstrouse - I was correct that the ET4 doesn't have an Aux4. Is this still the same as the Intellicenter i5? The spillway can be a feature and doesn't need to be a circuit. I'll take a look now.

No that is not the same as the i5s.  The numbering represents **5 total relays** 4 aux circuits + filter relay.  What I find odd in the replay is that starting at 11 for the first feature they are ordered as below in the poolConfig.json.
12 = Pool Lower = Feature 2
13 = Pool Lowest = Feature 3
14 = Pool Low = Feature 4

And in the screenshot above they appear as
Pool Low = Feature 1
Pool Lower = Feature 2
Pool Lowest = Feature 3
Pool Low = Feature 4
Pool High = Feature 5
Pool Higher = Feature 6

Notice Feature 1 & 4 have the same name (I assume the same name id).  If you are finding the circuit by name id then 4 would overwrite feature 1 as feature 4.  Perhaps it gets confused at that point and stops processing features 5 & 6.  I assume all of these come in on the action 11 message is that correct?

tagyoureit, 2020-11-03:
This is from the manual, FWIW:

> Select the circuit number 1/18. Press the Up button two times to select AUX 1 to choose this circuit for a light circuit On/Off button. The generic circuits names are: Spa, Pool, Aux 1-7 (EasyTouch 8), Aux 1-3 (EasyTouch 4), FEATURE 1-8 (Ft. 1-8), AuxX (Aux Extra). The circuit number (1/18) corresponds to its assigned circuit name. 

Notice the "Aux 1-3 (EasyTouch 4)".  (Yes, that's still bugging me.)

I think the replay that @johnny2678 sent he disabled Feature 1 (circuit id 11) because it had the same name as Feature 4.  He can confirm that, but I think we are on the right track now with my latest push.  These are the features I'm detecting:

```
[
{
"id": 12,
"showInFeatures": true,
"isOn": false,
"name": "Pool Lower",
"type": {
"val": 0,
"name": "generic",
"desc": "Generic"
},
"nameId": 206,
"equipmentType": "feature"
},
{
"id": 13,
"showInFeatures": true,
"isOn": false,
"name": "Pool Lowest",
"type": {
"val": 0,
"name": "generic",
"desc": "Generic"
},
"nameId": 200,
"equipmentType": "feature"
},
{
"id": 14,
"showInFeatures": true,
"isOn": false,
"name": "Pool Low",
"type": {
"val": 0,
"name": "generic",
"desc": "Generic"
},
"nameId": 64,
"equipmentType": "feature"
},
{
"id": 15,
"showInFeatures": true,
"isOn": false,
"name": "Pool High",
"type": {
"val": 0,
"name": "generic",
"desc": "Generic"
},
"nameId": 202,
"equipmentType": "feature"
},
{
"id": 16,
"showInFeatures": true,
"isOn": false,
"name": "Pool Higher",
"type": {
"val": 0,
"name": "generic",
"desc": "Generic"
},
"nameId": 203,
"equipmentType": "feature"
},
{
"id": 20,
"showInFeatures": true,
"isOn": false,
"name": "AUX EXTRA",
"type": {
"val": 0,
"name": "generic",
"desc": "Generic"
},
"nameId": 93,
"equipmentType": "feature"
}
]
```

rstrouse, 2020-11-03:
> Notice the "Aux 1-3 (EasyTouch 4)". (Yes, that's still bugging me.)
Technically I guess that is 4 relays.  Filter + Aux 1-3.  That is consistent with i5 where there is Filter + Aux 1-4 in IntelliTouch and IntelliCenter.

That config for sure looks better.

emlowe, 2020-11-03:
Pulled latest code as requested - things look good now - spillway detected and appears in schedule. Attached is packet capture

[dump.zip](https://github.com/tagyoureit/nodejs-poolController/files/5482945/dump.zip)

johnny2678, 2020-11-03:
> I think the replay that @johnny2678 sent he disabled Feature 1 (circuit id 11) because it had the same name as Feature 4. 

Yes, was trying to remove one of the Pool Lows and I removed the wrong one.

The latest did it - seeing `Pool High` and `Pool Higher` in the features list now.  
[image]

closed 2021-02-04 by tagyoureit