tagyoureit/nodejs-poolController #245: Where is lat / long suppose to be entered?

miamijerry, 2020-11-02:
Seeing this in terminal;
[11/2/2020, 10:19:21 AM] warn: dt:Mon Nov 02 2020 09:59:20 GMT-0500 (Eastern Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.

tagyoureit, 2020-11-02:
It's not by address, it's latitude and longitude.  Address is just for the "pool owner" information but sunrise/sunset are calculated locally via lat/long so we don't need to make any calls out to the internet to interpolate the address.

closed 2020-11-02 by tagyoureit

miamijerry, 2020-11-02:
The question is "Where do I enter Lat / Long ??"

rstrouse, 2020-11-02:
Close njspc then edit the poolConfig.json file in the data directory.  You want to enter the latitude and longitude in the pool.location node.  Then restart njspc.
```"pool": {
    "options": {
      "clockMode": 12,
      "clockSource": "internet",
      "units": 0,
      "adjustDST": true,
      "pumpDelay": false,
      "cooldownDelay": false,
      "manualPriority": false,
      "manualHeat": false
    },
    "alias": "Pool",
    "owner": {
      "name": "Mr. President",
      "phone": "5555555555",
      "phone2": "",
      "email": "",
      "email2": "mail2@home.com"
    },
    "location": {
      "zip": "20500",
      "latitude": 38.89,
      "address": "1600 Pennsylvania Avenue NW",
      "longitude": -77.036,
      "timeZone": 131,
      "country": "United States",
      "city": "Washington",
      "state": "DC"
    }
```
In the future I will add this to dashPanel so you can add it from there.

tagyoureit, 2020-11-02:
How did you enter the address?  The lat/long is right below that.

[image]

[image]

tagyoureit, 2021-01-08:
This also works in Dashpanel now.

[image]

gambler2841, 2021-05-11:
I apologize in advance, I'm not extremely experienced with Linux or the Raspberry Pi. I'm encountering a problem with connecting to the pool controller. I am getting hung up on the following:

[image]

Prior to this I was getting the lat/long error, and after adding per the instructions above, now if is just stuck.

Any help is greatly appreciated!

davthomaspilot, 2021-06-01:
I just installed the latest Pool Control Dashboard.    I don't see the lat and lon entry boxes in the Personal Information dialog:

[image]

and I'm getting the warning:

Jun 01 10:18:20 alexa-2 node[2281]: [6/1/2021, 10:18:20 AM] warn: dt:Tue Jun 01 2021 10:18:17 GMT-0400 (Eastern Daylight Time) lat:undefined lon:undefined Not enough information to c

tagyoureit, 2021-06-01:
It looks like you are on an old version of dashPanel.  Please run `git pull` and `npm i` assuming you have already done a `git clone https://github.com/rstrouse/nodejs-poolController-dashPanel.git` to initially load the app.
If you haven't updated the original app you should also update njsPC app itself.

davthomaspilot, 2021-06-02:
> Oh, and if you did pull the latest dashPanel make sure you do a hard refresh (Ctrl-F5) and that will load the latest versions of the files.

I couldn't do a git the first time--I was too lazy to recall how to set up a public key.
But, I did do a download of the zip file.  Is that the same?
when you say "njsPC", you mane nodejs-poolController, right?  (versus a tag for a release or something like that)

davthomaspilot, 2021-06-02:
I got errors many minutes after doing the npm start.  So, I made a public get and did a git clone:

git clone ***@***.***:tagyoureit/nodejs-poolController.git

Will that give me the latest pool controller?
I'll do the same for dashPanel, once I confirm last night's fix in poolController works.

davthomaspilot, 2021-06-03:
doing a git pull resolved the issue.  I didn't do the hard refresh.