tagyoureit/nodejs-poolController #247: Temp / Body Events, InfluxDB JSON

arrmo, 2020-11-23:
**Describe the bug**
A few steps here, let me try to explain what I have seen - and some of this may come back to config (vs. code), by all means yell if there are things I should try! 
1) In pool mode - water circulates through pool and spa (so temps are really the same for both). From /state/all, I do see ```"waterSensor1": 61 ``` ... OK, and the pool (body) temp is also the same, but no temperature reported for the spa. Add an option to "copy" the pool to the spa in "pool mode"?

2) Turned the spa on, heater kicks in, temperature starts to go up. I do see ```"waterSensor1": 64``` ... but it still reports as a waterSensor, and as pool temp, not spa. Does this sensor not need to be "copied" to the spa, in "spa mode"? Or really, copy it to one, the other or all (option?), as a function of the operating mode?

Tied in to the two observations above, I do notice a couple other things,
a) HomeKit (Homebridge) does show a "Water Sensor" ... that makes sense, and it matches to the waterSensor above. But Spa temp always reports 32F ... I think because the Spa body never provides a temp (even in Spa mode).

b) In web/bindings/influxDB.json, there are body temperatures being captured in two events - temps and body. The temps event seems to add points as follows (just grabbing a single temps point, easier to explain that way 😄),
```
> select * from ambientTemps ORDER BY DESC LIMIT 10
name: ambientTemps
time                          airTemp sourceApp sourceIP     units
----                          ------- --------- --------     -----
2020-11-23T13:17:19.19-06:00  60      njspc     192.168.2.77 Fahrenheit

> select * from bodyTemps ORDER BY DESC LIMIT 10
name: bodyTemps
time                          Pool PoolSetpoint SpaSetpoint body heatMode heatStatus sourceApp sourceIP     units
----                          ---- ------------ ----------- ---- -------- ---------- --------- --------     -----
2020-11-23T13:18:05.851-06:00 64   78                       Pool Off      Off        njspc     192.168.2.77
2020-11-23T13:17:53.528-06:00                   98          Spa  Heater   Off        njspc     192.168.2.77
2020-11-23T13:17:19.192-06:00 61                            Pool Off      Off        njspc     192.168.2.77 Fahrenheit
```
This data shows that temps triggers both, but some items are not output correctly - no biggie, easy to fix in the json ... but, what is missing is that the body (spa on in this case) is reporting the spa temp as pool. This matches to the notes above.

c) I do see that body events happen as well - perhaps just use that, but make sure the code always sends a body event after temps? And have the code cycle through each body, sending an event for each? This would allow the json to work then, independent of the number of bodies.

**To Reproduce**
Tried to explain above, in the description - but yell if more info is needed.

**Expected behavior**
Temperatures "align" with the body in use (and option to copy pool to spa temp perhaps, when in pool mode?). Hopefully the description above makes sense.

**Screenshots**
N/A

**Packet Capture**
N/A (I think 🤣).

**Pool Equipment**
 - Controller: IntelliTouch, 7+3
 - Pump(s) manufacturer and model: no "smart" pumps
 - Heater(s): gas
 - Valves: Jandy

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser: N/A
 - Version: N/A

**Smartphone (please complete the following information):**
N/A

**Additional context**
Yell if there are things you want me to try. Thanks!

PS: Is there a way (API?) to force certain events (like temps, or body)? Just to make debugging easier / quicker.

rstrouse, 2020-11-24:
Here is the way this should be working.
The data should follow the running body in a shared system.  If body1 is on then waterSensor1 will be reflected in body id:1.  If it is off (meaning the circuit relay associated with the body is off) then the temperature displayed in body id:1 will be the last temperature recorded when it was running.  If body id:2 is turned on then the temperature in waterSensor1 is reflected into body id:2.  In a shared system the bodies are mutually exclusive.  If neither body is on, then the temperature on either body should reflect the last time it was running and watersensor1 will continue to rise and fall as it is read.

Is that not what is occurring?  Now there are two events where you will see the body state.  This includes temps where the body is included with the event and body where the body state is actually changing.  

The body event occurs when a body actually changes state and the temps event occurs every time there is a change with one of the sensors.  

From what I glean on your setup there are only three conditions where you would get a body event.
1. The heater status changes 
2. The heat mode changes
3. The state of the body toggles between on and off.
These changes can occur as the result of a schedule, one of the controllers (including njspc) makes a body change, or as part of a grouped macro circuit.

Note the last time my spa ran it heated to 83F and the pool water is currently a balmy 55F.  When the pool turns off the temperature of the water in the pipe is really not all that relevant.
[image]

njspc does cache this data and it isn't reported again until the particular body is turned on again.  In fact on the IntelliTouch controllers this value is hidden if the body isn't running.  No water temp is displayed anywhere when a body is not on.

You might be able to force a temps message if you PUT: /state/temps {solarSensor4: <randomNumber>}

rstrouse, 2020-11-24:
Also double check your circuit functions in IntelliTouch.  Circuit 1 should have a function of spa.  In a i7+3 not i7+3S model, circuit 1 is the default spa circuit.

rstrouse, 2020-11-24:
I just checked in some files to next that may fix the issue when switching bodies.  I don't have a Touch controller, I am running IntelliCenter but other versions of the Touch controllers repeated waterSensor1 and waterSensor2 in the byte mapping so the second body was actually looking at that byte and it just worked.  However, I am willing to bet that your Touch controller operates more like IntelliCenter where that byte is not duplicated.  As a result, the second body wasn't getting the temperature sensor data.  Pull next and give it a try.  

It should fix your problem.

arrmo, 2020-11-25:
Thanks for the thoughts (and updates!). A few comments - do let me know what you think.

> Now there are two events where you will see the body state. This includes temps where the body is included with the event and body where the body state is actually changing.

Yep, agreed! We're saying the same thing. And as temps also seems to emit (cause an event on) body, just use body for influxdb, agreed?

> The body event occurs when a body actually changes state and the temps event occurs every time there is a change with one of the sensors.

Yes, that makes sense - and looks right. So whenever this triggers, only one body is sent per event, right? If so, I can see how to update influxdb.json.

> You might be able to force a temps message if you PUT: /state/temps {solarSensor4: [randomNumber]}

Yep, seems to work - perfect (for debug)!

> Also double check your circuit functions in IntelliTouch. Circuit 1 should have a function of spa. In a i7+3 not i7+3S model, circuit 1 is the default spa circuit.

Looks right. ID is not 1, but circuit is. That's OK, agreed?

> Pull next and give it a try.

I do see two watersensors now ... not sure what waterSensor1 is (it's not matching the pool or spa temp), but waterSensor2 is matching the spa. Make sense? => Actually, the mismatch may be because the pool is not running. Just a sec ... 😆. OK, they match when running (waterSensor1 and pool).

Thanks again!

arrmo, 2020-11-25:
FYI, I do have influxdb.json updated here (locally for a moment), seems to be working. Any concerns on the questions above? Just let me know, and I can adapt. But pushing to influxdb only based on body events ... which seems to work, and I think matches your thoughts ... right?

arrmo, 2020-11-25:
OK, and one more 🤣. Your PUT works great, to trigger. I used /state/body/setPoint/ => data shows up as below,
```
> select * from ambientTemps ORDER BY DESC LIMIT 10 tz('America/Chicago')
name: ambientTemps
time                          airTemp sourceApp sourceIP     units
----                          ------- --------- --------     -----
2020-11-24T20:37:10.288-06:00 61      njspc     192.168.2.77 Fahrenheit
```
```
> select * from bodyTemps ORDER BY DESC LIMIT 10 tz('America/Chicago')
name: bodyTemps
time                          Pool Pool.setPoint body heatMode heatStatus sourceApp sourceIP
----                          ---- ------------- ---- -------- ---------- --------- --------
2020-11-24T20:37:19.298-06:00 63   78            Pool Off      Off        njspc     192.168.2.77
2020-11-24T20:37:15.366-06:00 63   78            Pool Off      Off        njspc     192.168.2.77
```

But also ... huh? I have checked, don't see a circuit 192 or Intellibrite. Why is this existing (never mind being triggered by /state/body/setPoint/)?
```
> select * from circuits ORDER BY DESC LIMIT 10 tz('America/Chicago')
name: circuits
time                          id  isOn  isOnVal name         sourceApp sourceIP     type
----                          --  ----  ------- ----         --------- --------     ----
2020-11-24T20:37:52.745-06:00 192 false 0       Intellibrite njspc     192.168.2.77 IntelliBrite
2020-11-24T20:37:51.745-06:00 192 true          Intellibrite njspc     192.168.2.77 IntelliBrite
```

Thanks!

arrmo, 2020-11-25:
FYI, just a bit more data to share, spa on then back off. And yes, I did try changing the Spa setpoint also 😄.
```
> select * from bodyTemps ORDER BY DESC LIMIT 20 tz('America/Chicago')
name: bodyTemps
time                          Pool Pool.setPoint Spa Spa.setPoint body heatMode heatStatus sourceApp sourceIP
----                          ---- ------------- --- ------------ ---- -------- ---------- --------- --------
2020-11-24T20:55:05.926-06:00 63   78                             Pool Off      Off        njspc     192.168.2.77
2020-11-24T20:54:56.107-06:00 64   78                             Pool Off      Off        njspc     192.168.2.77
2020-11-24T20:54:46.279-06:00 65   78                             Pool Off      Off        njspc     192.168.2.77
2020-11-24T20:54:26.606-06:00 66   78                             Pool Off      Off        njspc     192.168.2.77
2020-11-24T20:54:16.784-06:00 67   78                             Pool Off      Off        njspc     192.168.2.77
2020-11-24T20:54:04.978-06:00                    66  99           Spa  Heater   Off        njspc     192.168.2.77
2020-11-24T20:54:04.975-06:00 66   78                             Pool Off      Off        njspc     192.168.2.77
2020-11-24T20:53:21.37-06:00                     66  99           Spa  Heater   Off        njspc     192.168.2.77
2020-11-24T20:53:05.642-06:00                    66  101          Spa  Heater   Off        njspc     192.168.2.77
2020-11-24T20:52:47.948-06:00                    66  101          Spa  Heater   Off        njspc     192.168.2.77
2020-11-24T20:51:39.135-06:00                    65  101          Spa  Heater   Off        njspc     192.168.2.77
2020-11-24T20:50:30.324-06:00                    64  101          Spa  Heater   Off        njspc     192.168.2.77
2020-11-24T20:47:58.94-06:00                     63  101          Spa  Heater   Off        njspc     192.168.2.77
2020-11-24T20:47:58.935-06:00 63   78                             Pool Off      Off        njspc     192.168.2.77
2020-11-24T20:37:19.298-06:00 63   78                             Pool Off      Off        njspc     192.168.2.77
2020-11-24T20:37:15.366-06:00 63   78                             Pool Off      Off        njspc     192.168.2.77
```

tagyoureit, 2020-11-25:
Is the last set of data you posted correctly or is there some concern there?

Also, if you can upload a replay capture I'll take a look at the Intellibrite.  I'm not inclined to completely reprogram my pool at the moment.  :)

arrmo, 2020-11-25:
> Is the last set of data you posted correctly or is there some concern there?

Nope, it's working great - just wanted to share the info. Of course, if you see issues, do yell.

> upload a replay capture I'll take a look at the Intellibrite

Sure. Let me see how to capture that, it's a new one to me ... LOL.

tagyoureit, 2020-11-25:
There's a wiki for that!  :). https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution#packet-capture---60

arrmo, 2020-11-26:
That was easy 🤣. Thanks!

OK, attached. I had to just let it run for a bit (sorry) - as I'm not quite sure what is triggering it. Just yell if you want me to try something.

[2020-11-24_20-15-22.zip](https://github.com/tagyoureit/nodejs-poolController/files/5600181/2020-11-24_20-15-22.zip)

arrmo, 2020-11-28:
OK, influxdb.json PR in place - just yell if you see any issues, I can address as needed. Thanks!

On the Intellibrite, let me know if you see anything I need to try. Appreciate the help!

tagyoureit, 2020-11-28:
I accepted the PR.  Thanks.

I also pushed up a couple filter checks at the emit level (eg, circuit, feature, etc) and also the point level.  If you want to get more granular in skipping data, let me know.

For the lightGroup, I would disable it.  The way *Touch sends the Intellibrite info I literally need to capture all of it, compare it to what I previously had, and then remove anything that isn't currently installed.  In the `influxDB.json`, add `enabled: false` to the lightGroup like this:

```
      "name": "lightGroup",
      "description": "Bind lightGroup emit",
      "enabled": false,
```

arrmo, 2020-11-30:
Thanks! Pulled the code update, modified `influxDB.json` - will let you know.

arrmo, 2020-12-01:
Seems to be working, thanks!

Do you want me to submit a PR for that (enabled default) change?

tagyoureit, 2021-02-04:
FYI, we found a bug that njsPC wasn't pulling the air/solar temp as often as it should be.  

Closing issue is this is fixed.

closed 2021-02-04 by tagyoureit