tagyoureit/nodejs-poolController #250: IntelliTouch i10+3D with Expansion - Upgrade to Intellicenter i10-PS with (2) Expansion

ckurlinski, 2020-12-07:
Here's the first log, After firmware update on both boards to 1.047
[intellicenter_startup.zip](https://github.com/tagyoureit/nodejs-poolController/files/5654950/intellicenter_startup.zip)

rstrouse, 2020-12-07:
For this dump.  Exactly what was installed?
I assume there was only one expansion installed is that correct or was both of them installed?

ckurlinski, 2020-12-07:
Correct, Need to wire the second expansion.
There was nothing but the (2) personality kits and njsp on the bus

rstrouse, 2020-12-07:
Is the expansion kit an i10x or i5x?

rstrouse, 2020-12-07:
Oh wait you said only the base personality board?  That would mean i5x.

ckurlinski, 2020-12-07:
It's a i5x with 5x expansion added

ckurlinski, 2020-12-07:
The reseller said Pentair doesn't make a 10x, just a 5x with an expansion module added

rstrouse, 2020-12-07:
So one expansion with the base board and an the 5x expansion personality board.  I guess that is what makes up an i10x.

ckurlinski, 2020-12-07:
Correct

rstrouse, 2020-12-07:
That almost makes a fartload of sense but I assumed it was encoded differently.  You are currently missing 5 circuits that I will dig up for you.  We detected the expansion module but didn't realize the base board was assumed based upon the existence of the expansion panel.

That does fall in line with the layout on the master panel now.

ckurlinski, 2020-12-07:
They look almost identical....
First on is the Main Board
[image]
[image]

ckurlinski, 2020-12-07:
See the model numbers on the expansion boards

rstrouse, 2020-12-07:
So let me see if I have this straight.  The differences are the following.  Bear in mind the card that I am seeing on the main panel is an i10ps not an i10d.

|     |   i5p   |   i5x   | i10x | i10ps |
| --- | ------- | ------- | ----- | --- |
| circuits | 5 | 5 | 5 | 5 |
| valves | 2 | 0 | 0 | 2 |
| bodies | 1 | 0 | 0 | 1 |
| sensors | 3 | 0 | 0 | 0 |

No wait a minute the Pentair rep is not correct.  The expansion is the i10x card.  There are some things that I see missing there though.  Where are the temp sensor inputs for the third body?  It also appears there is no gas or solar heater capability for that said body.  The valve a & b are the omron relays on the i5p.  Given this, where do the solar and water 3 temperature feeds come from.  You do know that the i10ps is a shared body system.  It is much different than the i10d.  Are you sure that is what you want?

ckurlinski, 2020-12-07:
I thought Ordered A 10D....
Damn, didn't catch that.

rstrouse, 2020-12-07:
You were throwing me for a loop.  I don't think the i10ps will work for you.  It makes body 1 & 2 mutually exclusive and combines the water through a spillway if set up that way.

ckurlinski, 2020-12-07:
Explains the Spa label on the expansion card in the main board

rstrouse, 2020-12-07:
Are you planning on more than 2 valves?  If so you need a valve expansion or 2.  1 valve expansion per load/expansion center.

> Explains the Spa label on the expansion card in the main board

Yes it does.

rstrouse, 2020-12-07:
The good news is that I see the i10ps with an expansion panel installed in the logs.  Not much good for you though.

ckurlinski, 2020-12-07:
I need all of these valves working.....

[image]
[image]

ckurlinski, 2020-12-07:
I can play around with the routing of the relays, move Spa stuff to spa...

rstrouse, 2020-12-07:
The good news is you can max out at 26 valves.  The only problem is that the bodies are mutually exclusive with a ps.  You cannot run spa and pool at the same time.  It also is reliant on an intake/return valve setup.  Maybe you have this but it is not the same as the d configuration.  Not even remotely.

rstrouse, 2020-12-07:
You can also control up to 16 of those pumps.

ckurlinski, 2020-12-07:
Yeah, just hit that wall.
I don't really have a Spa, per se anyway, (1) filter pump with chlorinator for Infinity style
Then, valves open and close, lowers the spa for heating, UV turns on.
Then there's vacation mode, valves are open, just filtering.
Then spa jet mode lowers more
Then spa Party Mode, hookers and blow show and the thing becomes like a Studio 54.....
So I can work around (1) relay turning off....

ckurlinski, 2020-12-07:
Well, this sucks....
Anything on the Main Personality Expansion Card is triggered as the Spa, killing the Main Pool, and vice versa.....

rstrouse, 2020-12-07:
What is really cool here is that you can have 2 sets of schedules.  One for when there will be bather load (occupied) and way.  However, I think you will have issues with that ps vs the d if that is the way you are plumbed.

That is correct.  The shared body means that the equipment for the main pump is shared and the bodies are connected.  When the spa is on the intake is drawn from the spa and returned to the spa.  If the pool is on the intake is drawn from the pool and returned from the pool.  If the spillway feature is engaged the intake is drawn from the pool and returned to the spa.  Bear in mind it is different pumbing.

The other thing that comes into play here is that there is only 1 water temp sensor and the gas heater relay is set up to either heat the spa or the pool.  There are not 2.  As well there is only one solar connection.

rstrouse, 2020-12-07:
If it is a champagne spa where the returns into the spa are managed to keep the heat in then I believe it is designed to handle that but it must be plumbed to accommodate it.  What this does is circulate the water exclusively into the spa.  The pool is not getting water from the filter system.   Then the lowering of the water in the main body is a function of the vanishing edge spillway.  Again it has to be plumbed that way.

rstrouse, 2020-12-07:
All of the water features and other circuits are not part of this shared function.  This is only for the main filter.

rstrouse, 2020-12-07:
By chance if you still have this hooked up.  Can I get you to run a capture with restart?  I don't know when we will see an i10ps with expansion panels.

ckurlinski, 2020-12-07:
I will have to do it tomorrow.
But yeah, need to do a work around, no time to change the thing to i10D.
Owner will be here soon.

rstrouse, 2020-12-07:
You know it is a single board.  Perhaps you can convince Pentair to ship you one in the interim.  Sorry it turned sideways.  The thing that is so messed up is that there really isn't a definitive source regarding what each of the models are capable of.  It borders on idiotic.

tagyoureit, 2020-12-07:
Our friends at TFP probably have the best descriptions.  https://www.troublefreepool.com/threads/intellicenter-quick-reference-information-guide.198056/

The IntelliTouch [brochure](https://www.pentair.com/content/dam/extranet/pentair-pool/residential/brochures/pool-automation/IntelliTouch_Pool_and_Spa_Control_Systems_English.pdf) (the most official source) is the only comprehensive table from Pentair that was half digestible.

rstrouse, 2020-12-07:
Yeah there is nothing for IntelliCenter.  A list like the IntelliTouch brochure would be nice but it would still leave out important points.  Like where are the temp sensors for the remaining bodies.

ckurlinski, 2020-12-07:
Is there a way. to send info to you guys privately.
Just found something I don't want to say publicly.

rstrouse, 2020-12-07:
Over on the gitter chat you can PM.

ckurlinski, 2020-12-07:
Understood

rstrouse, 2020-12-25:
The valve diagram is below and the relay in question is https://www.amazon.com/Electronics-Salon-control-Pluggable-Module-G2R-1/dp/B00M1L75YU/ref=sr_1_3?dchild=1&keywords=24vac+relay+board&qid=1608855042&sr=8-3

You do not need a DPDT and in fact it would complicate the matter since your transformer will be energizing the relay for when it is diverted or not diverted.  The valve will always have power from the transformer.  The valve diagram shows 2 of the valves hooked up but you can probably figure out the other 3 from the diagram.
[image]

ckurlinski, 2020-12-29:
This is a capture / reset with all (3) intellicenters up.

[three_boxen.zip](https://github.com/tagyoureit/nodejs-poolController/files/5751952/three_boxen.zip)

ckurlinski, 2020-12-29:
Thank you for the info, ordering it now.

rstrouse, 2020-12-29:
I'll have a look at the log outputs and decipher the bits.  I'm sure we probably have some adjustment on the code to accommodate the expansion panels.  Thanks for the dump (you never want to say that to somebody leaving a bathroom).

I hope my diagram made sense!

ckurlinski, 2020-12-29:
Perfect sense.
Essentially it's exactly what I was going to do.
Just nice to confirm it is not unusual.
Getting the relay bank Jan 4th
BTW, what did you do the layout in?

rstrouse, 2020-12-29:
I used Visio Professional.

rstrouse, 2020-12-30:
@ckurlinski - Just to verify what I am looking at with the data.
1 Load center with i10D card and 1 valve expansion module.
3 Power centers with an i10x card installed and not other expansions.

So there is a total of three i10x cards, one i10D card, and one valve expansion installed at the time of this capture.  Is that correct?

tagyoureit, 2021-02-04:
Closing due to lack of activity.

closed 2021-02-04 by tagyoureit