tagyoureit/nodejs-poolController #253: Invalid body assignment, test branch, orp and ph

miamijerry, 2020-12-30:
[image]
[image]

I only get Intellichem showing in dashPanel when I am on "Test" branch. This was working fine at some point changing PH and ORP setpoints. Currently I get this error message about "Invalid body assignment" and can not change setpoints.

rstrouse, 2020-12-30:
I'll have a look.  We have been adding quite a bit to the chemistry control lately so it is possible we got the body assignments twisted up.  If I remember, which OCP are you running?

miamijerry, 2020-12-30:
Easytouch2 8 aux

rstrouse, 2020-12-30:
Thanks.  I assume with the Easytouch 8 you don't get to specify whether it is for the pool, spa, or both.  Is that correct?

rstrouse, 2020-12-30:
@miamijerry -- I removed the references for the body when attached to an EasyTouch system.  Pull next and dashPanel.  You should also now be able to set your tank level by clicking on the chem tank.  There are lots of changes and enhancements to the display for the dosing to include when the controller is dosing, what it is dosing at the time, and even how much is left but neither I or @tagyoureit have an IntelliChem to test this with so when you get a chance it would be great to see the output displayed when IntelliChem is dosing, mixing, and monitoring.
[image]

miamijerry, 2020-12-31:
Yes on the Easytouch 8 there is no selection for whether the PH and ORP setpoints are attached to a body.

Thanks, I will provide feedback

miamijerry, 2020-12-31:
Working fine, Thank You
[image]

closed 2020-12-31 by miamijerry