tagyoureit/nodejs-poolController #254: Trying to talk to my VSC Pump.  Pi3 loaded with the lastest.

n0xd, 2020-12-31:
**Expected behavior**

A clear and concise description of what you expected to happen.
**All I really need is pump control for now. A few commands.** 
**Using Pi3b, 64 gb card, JBtek USB to RS485 Converter Adapter ch340T chip 64-bit Window 7 Supported** 

I will do the Packet Capture shortly.  Thanks for helping. I'm a ham op, retired from the PO as a Electronic Tech. Love projects like this.

**Packet Capture**
Follow the instructions to complete a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) and attach the resulting zip/log files.

[image]

[image]

**Pool Equipment** - 
 - Pump(s) manufacturer and model: . Pentair Intelliflo  VSC older one
 - Chlorinator:   none
 - Heater(s):   none
 - Chemical controller:  none
 - Valves:  none
 

**Desktop (please complete the following information):**
 - OS: WIN10 Pro
 - Browser Firefox
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: iPhone 10
 - OS: 
 - Browser, safari
 - Version

rstrouse, 2020-12-31:
> I'm a ham op, retired from the PO as a Electronic Tech. Love projects like this.

You know there are those that operate but the rest of us are just the hind quarter of a pig.

Ok lets start with getting an actual part number for the pump.  I haven't heard of the VSC pumps.  I have heard of the VST pumps in which there are 2 models.  There are some differences with each model in how they are addressed.

So from the screenshot it appears you have figured out the messageManager and got to a point where you are sending a message.  You almost have the right message to send, but you included the preamble, header, and checksum in the payload.  I will give you a message to send out on the bus but this is a screenshot of a normal pump sequence 
NOTE: There are 2 pumps in this sequence.  
1. We send an action 4 which tells the pump to disable its control panel and the pump should echo that 4 back to us
2. We send an action 6 which tells the pump to enter into a drive state of 10 and the pump should echo that 6 back to us.
3. At this point we can then command the pump to set a flow rate.  If we get here and get some responses from the pump we are in business.
[image]

Now the message that I want you to send is an action 4.  If we get a response there the pump is communicating.  If it doesn't respond we need to make sure that it is at the proper address.  There are 4, 8, or 16 possible addresses depending on the pump model.  To send to the next address change the address to 97 which will actually be the second pump (ignore 95 as that is an incorrect address in the dropdown). Pump addresses range from 96-111.

[image]

n0xd, 2020-12-31:
> <nodejs-poolController@noreply.github.com>
> Pump. Pi3 loaded with the lastest. (#254)

Am I talking to the pump?   Also I have the Penatir var speed control. 
The early version if that helps.  With the daughter home from college 
and holidays I trying to get this running in my spare time. Thanks again 
for the help.

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

rstrouse, 2020-12-31:
Yes if you send the message I described in the screenshot above.  What determines whether it is for the pump is the address as every piece of equipment is attached to the same 2 wires.  You don't by chance have that variable speed controller hooked up do you?  If you do, it may take over as the master controller and nothing else will be able to control that pump.

n0xd, 2020-12-31:
I'm still working on what you gave me. Haven't seen pool as a source 
yet. But still playing. Are you a ham?
This is a pic of my operation. :)

Sent: 12/31/2020 5:16:27 PM

https://www.avast.com/antivirus

rstrouse, 2020-12-31:
> Are you a ham?

If you ask my wife I am.  But I am more like the green eggs.

I think however since you are responding via the e-mail link any of your attachments are not coming through.  You should respond by clicking on this link.  It will allow you to drag and drop images and the like.
https://github.com/tagyoureit/nodejs-poolController/issues/254

The sources are simply a number and believe it or not pool is not a source for RS485 comms.  The OCP (Outdoor Control Panel) is the primary source address for the master and when you are running a virtual controller you will emulating that.  This source address is 16.  The destination address is 96 for pump 1.   Most of the time I find that when a pump isn't communicating it is because somebody changed the programming on the pump to be an address other than 1.  In that case you need to either go back into the control panel mounted to the pump and set it to 1 or change the address of the message above to the same address that is set on the pump.
1 = 96
2 = 97
3 = 98
4 = 99
...
16 = 111

n0xd, 2020-12-31:
[image]

[image]

Here is a pic of my operation.  Also when I mention ham it's my hobby. Amateur radio. You prob knew that. Thanks again for the help.

rstrouse, 2021-01-01:
Well for sure we are not getting a response from that pump on address 96.  What we can try is to first delete the top message from your queue by pressing on the trash can.  Then edit the remaining message by clicking on the pencil and paper.  Change the address to 97 and send the queue if it responds then it will be an echo of the message but the source will be the pump and the dest will be the OCP.

You can try this with the remaining 16 addresses.  If we don't get a response then either our wiring is suspect or the pump is not liking life.  From the pics that looks like this Sta-Rite pump.  
https://www.horizonpoolsupply.com/_uls/resources/IntelliPro_Manual.pdf

And from the pic it also looks like water got under the lcd screen.  I remember reading somewhere a method for lifting up the screen and drying it out.

rstrouse, 2021-01-01:
Here is a link to a method for fixing the display.
https://www.youtube.com/watch?v=K95jkQH-nyA

n0xd, 2021-01-01:
The first thing I tried. Didn't work on this one. I need to find 
another. Do you know a cheap place to find them. I see them on ebay for 
150-200.
My neighbor just updated to a new one so I grabbed this one. Much better 
than my old one.

Sent: 12/31/2020 8:01:12 PM

https://www.avast.com/antivirus

rstrouse, 2021-01-01:
The cheapest place I know of to get one is in your neighbors back yard at around 3am.  Other than that I don't have an angle.

n0xd, 2021-01-01:
I never though of that.  But I don't need any bad karma :)   I'll keep 
plugging away. I know what to send and what to expect from the pump.  
Thanks very much and have a Better New Year.
In sw florida here.

Sent: 12/31/2020 8:22:49 PM

https://www.avast.com/antivirus

rstrouse, 2021-01-01:
I'm on the other coast so you get to start having a better year earlier than me.  This state has lost its mind.

n0xd, 2021-01-02:
[image]

This is what I am getting out of the rs485 on the raspberry pi  when I send cmd 4 to pump 2(1)    As monitored on my netbook with a seperate rs485 .  What is your thought? Does it look close?

rstrouse, 2021-01-02:
So in decimal what you are seeing it the command you sent and it is the correct message to send when the pump address is set to 1 on the pump.  I suspect that the comms are good because this is an echo of the command you sent.  The pump has sent nothing back.  You can try each pump address as there is a chance that the pump was changed from the default of 1 and you have no way of checking to see what address it is set to because the screen is torched.  To do this change the 96 value to 97 for address 2, 98 for address 3... etc up to 111 for address 16.  This will rule this out as the pump won't respond to another pump's address.
```
The first 4 bytes are simply noise left over from another message
[1, 255, 2, 25][255, 0, 255][165, 0, 96, 16, 4, 1][255][2, 25]
```
The exact message that the pump will send back is
```
[255, 0, 255][165, 0, 16, 96, 4, 1][255][2, 25]
FF 00 FF A5 00 10 60 04 01 FF 02 19 -- In hex
```

rstrouse, 2021-01-02:
If you use messageManager to send these messages it will calculate the correct checksum and apply the appropriate preamble and headers.  Otherwise sending a message without it you will need to calculate the last 2 bytes into a checksum if you try to send from the terminal program.

tagyoureit, 2021-02-04:
Closing due to lack of activity.

closed 2021-02-04 by tagyoureit