tagyoureit/nodejs-poolController #255: Black color palette icon disappeared, can not access intellibrite control

miamijerry, 2021-01-01:
The intellibrite black icon which provides access to colors and other controls disappeared. This is on both master and next branches.

[image]

rstrouse, 2021-01-01:
When you are running the purple theme it changes to white and I see it in your screenshot.  If you are running EasyTouch or IntelliTouch, each light doesn't have the ability to set individual IntelliBrite themes so the icon gets moved to the header because you can only have one IntelliBrite control in those systems which control all lights.

Below is the view from IntelliCenter.  The palette button exists on each light because each light can have its own theme and color.
[image]

miamijerry, 2021-01-01:
The white icon is not clickable. There was both a white and black icon before, the black icon was just below the white icon. Only the black icon has been clickable for me.

May have changed when the "invalid assignment issue" got fixed ??

rstrouse, 2021-01-02:
EasyTouch2 8 right?  It looks like there were changes to how the IntelliBrite light groups were managed.  Can you upload your poolState.json and poolConfig.json so I can see how they differ from IntelliCenter.

miamijerry, 2021-01-02:
[poolConfig.zip](https://github.com/tagyoureit/nodejs-poolController/files/5759708/poolConfig.zip)
[poolState.zip](https://github.com/tagyoureit/nodejs-poolController/files/5759709/poolState.zip)

rstrouse, 2021-01-02:
@miamijerry - pull njspc and dashPanel.  I made changes to support the new way light themes are being handled.  Don't forget to hard refresh your browser in dashPanel.

miamijerry, 2021-01-02:
Thanks, Working again!
[image]

miamijerry, 2021-01-04:
[image]
When I go to select a color, I get an error
Invalid Equipment ID
Easytouch2 8

rstrouse, 2021-01-06:
Please grab the latest njspc and dashPanel.

miamijerry, 2021-01-06:
Pulled the latest and the issue is the same.

[poolConfig.zip](https://github.com/tagyoureit/nodejs-poolController/files/5778320/poolConfig.zip)
[poolState.zip](https://github.com/tagyoureit/nodejs-poolController/files/5778321/poolState.zip)

rstrouse, 2021-01-07:
When you say select color do you mean a lighting theme or are you referring to individual colors for sAM modes?
The interesting thing is that your IntelliBrite group contains a light circuit that is not defined.
```
    {
      "id": 192,
      "master": 0,
      "circuits": [
        {
          "id": 0,
          "position": 0,
          "master": 0,
          "isActive": false
        },
        {
          "id": 2,
          "circuit": 4,
          "position": 1,
          "color": 4,
          "swimDelay": 5,
          "master": 0,
          "isActive": true
        }
      ],
      "isActive": true,
      "name": "Intellibrite",
      "type": 3,
      "lightingTheme": 194
    }
```
That circuit 0 is not valid.  I only see 1 IntelliBrite light in your circuits.

miamijerry, 2021-01-07:
Party, Romance, Caribbean, American, Sunset, Royal, Blue, Green, Red, White, and Magenta, are all working.

Can the undefined entries be deleted?

rstrouse, 2021-01-07:
The color section is really for sAM modes and these are typically related to multiple light circuits.  I will have a look at the Touch code to see what I see in relation to this.  In the meantime perform the following steps. 
1. please stop njspc.
2. Remove the circuit with id:0 from the JSON array including the comma.
3. Change the id for the circuit identified as 2 to 1.

In the end your IntelliBrite light group should look like the following.  After you have completed the edit save the file and restart njspc.
```
    {
      "id": 192,
      "master": 0,
      "circuits": [
        {
          "id": 1,
          "circuit": 4,
          "position": 1,
          "color": 4,
          "swimDelay": 5,
          "master": 0,
          "isActive": true
        }
      ],
      "isActive": true,
      "name": "Intellibrite",
      "type": 3,
      "lightingTheme": 194
    }
```

miamijerry, 2021-01-07:
Performed edit as detailed.

Operates perfectly

Thanks

closed 2021-01-07 by miamijerry