tagyoureit/nodejs-poolController #258: poolController MQTT interface question

miamijerry, 2021-01-07:
Working on mcsMQTT homseer integration of poolController.

When mcsMQTT broker is taken offline, and then later enabled, mcsMQTT receives about a thousand messages for each minute that mcsMQTT was offline.

So after 50 minutes, mcsMQTT has 50,000 messages to process.

Question, Could the poolController MQTT interface be retaining these messages until the mcsMQTT broker comes back online ??

rstrouse, 2021-01-07:
This probably has something to do with the qos settings.  Either on the njspc side (set in the config.json) or on the mcsMQTT side (dunno where that is set).

https://www.hivemq.com/blog/mqtt-essentials-part-6-mqtt-quality-of-service-levels/

tagyoureit, 2021-01-08:
You can also change the QOS settings in `/web/bindings/mqtt.json` for individual topics.  See the [wiki#mqtt](https://github.com/tagyoureit/nodejs-poolController/wiki/Bindings-Integrations-in-2.0#mqtt).

miamijerry, 2021-01-08:
I have a qos setting of "0" which is fire and forget, but somehow, 50,000 messages are being retained (Somewhere?), when mcsMQTT is offline, and then they come thru when mcsMQTT comes back online.

Michael McSharry, modified mcsMQTT to only process the last 100 messages, so this appears to be working.

miamijerry, 2021-01-08:
Can I use  "/web/bindings/mqtt.json" to stop most of the intellichem communication?

rstrouse, 2021-01-08:
Yes simply set enabled:false on the topic.  You can also make a copy of the mqtt.json so it won't get overwritten when you pull and point the mqtt interface to the new file.

miamijerry, 2021-01-08:
Can you please provide me an example of editing the file, 

I only want these six items; /alarms/ph, /alarms/orp, phlevel, orplevel, phsetpoint, orpsetpoint, 

the remaining items are too much ...

[mqtt.zip](https://github.com/tagyoureit/nodejs-poolController/files/5788198/mqtt.zip)

rstrouse, 2021-01-08:
Step 1: Stop njspc
Step 2: Make a copy of the mqtt.json file in the web/bindings directory and name it something else.  perhaps miamimqtt.json.
Step 3: Modify the config.json and change the filename to your new filename.
```
      "mqtt": {
        "name": "MQTT",
        "type": "mqtt",
        "enabled": true,
        "fileName": "miamimqtt.json",
        "globals": {},
        "options": {
          "protocol": "mqtt://",
          "host": "pi4",
          "port": 1883,
          "username": "",
          "password": "",
          "rootTopic": "@bind=(state.equipment.model).replace(' ','-').replace('/','').toLowerCase();",
          "retain": true,
          "qos": 0
        },
        "uuid": "2091c969-beee-4767-a9ca-11f6b6208830"
      },

```
Step 4: Modify each topic that you don't want to receive by adding "enabled": false to that topic.  Thake care to make sure you are putting all the correct quotes and commas into the file.  A good practice it to validate it at https://jsonlint.com/ when you are done.
```
        {
          "topic": "config/chemControllers/@bind=data.id;/@bind=data.name;/alkalinity",
          "message": "{\"alkalinity\":@bind=data.alkalinity;}",
          "enabled": false
        },

```
Step 5: Start njspc
Step 6: From here on out it will monitor the miamimqtt.json file so changes you save to it will immediately take effect so you can disable or enable any of the messages on the fly.

miamijerry, 2021-01-08:
Wow, thank you that is very helpful

miamijerry, 2021-01-09:
OK, I was able to edit mqtt.json

But I noticed that one mqtt topic contains most of the items I was trying to stop getting information on...

Does the attached look normal, this much information on the topic of "ORP Alarm"

[image]

tagyoureit, 2021-01-09:
That looks like all the info from chemControllers.orp.  You want chemControllers.alarms.orp.  Put `http://server:4200/state/chemControllers` in a browser to see what comes back.

miamijerry, 2021-01-09:
That has more information.
[http_chemcontrollers.txt](https://github.com/tagyoureit/nodejs-poolController/files/5791596/http_chemcontrollers.txt)

tagyoureit, 2021-01-09:
Yes, exactly as I was saying above - data.alarms.orp.  What is your custom binding that you have in miamimqtt.json?

rstrouse, 2021-01-09:
Yeah those bindings in the mqtt.json are not correct for the alarms.  If I get some time this weekend I'll fix 'em.  For instance the message for alarms/orp should be defined as...
```
          "message": "{\"orp\":@bind=data.alarms.orp;}"

```

miamijerry, 2021-01-10:
See attached my binding
[mqtt.zip](https://github.com/tagyoureit/nodejs-poolController/files/5792041/mqtt.zip)

tagyoureit, 2021-01-10:
D'oh!  Yes, of course they are wrong.  They are from the "old" data format before njsPC got supercharged with the new chemController data structure.   The correct data format is in influxDB.json.  Most of it can just be copied over.  It is now broken up into chemControllersGen (general), chemControllersPh and chemControllersORP because the data structures are so huge.  If @rstrouse doesn't get to it first, I'll fix it up later.

miamijerry, 2021-01-10:
Would be be possible to enable mqtt communication for:
Chemcontroller PH setpoint
Chemcontroller ORP setpoint
Chlorinator poolsetpoint
Chlorinator spasetpoint
Superchlorinate hours
Superchlorinate On/Off
????

iotarho, 2021-01-14:
Following up on a Gitter post for a feature request to have MQTT able to control the pool heater on/off state (and temperature setpoints as well).  Thank you!

miamijerry, 2021-01-14:
> Following up on a Gitter post for a feature request to have MQTT able to control the pool heater on/off state (and temperature setpoints as well). Thank you!

I have found these to already be working.
Pool Heatmode
"YourRootTopic"/state/body/heatMode
{"id":1,"heatMode":$$VALUE:}
Value 0=off 1=enable

Spa Heatmode
"YourRootTopic"/state/body/heatMode
{"id":2,"heatMode":$$VALUE:}
Value 0=off 1=enable

PoolSetPoint
"YourRootTopic"/state/body/setPoint
{"id":1,"setPoint":$$VALUE:}
Value is the degrees XX

SpaSetPoint
"YourRootTopic"/state/body/setPoint
{"id":2,"setPoint":$$VALUE:}
Value is the degrees XX

iotarho, 2021-01-16:
Thank you Miamijerry--your suggestions worked!

miamijerry, 2021-01-17:
[PUT and MQTT commands.docx](https://github.com/tagyoureit/nodejs-poolController/files/5825376/PUT.and.MQTT.commands.docx)

Looking for the missing commands marked with ??? in the attached list

rstrouse, 2021-01-23:
Funny, I didn't put together jerryt and miami.  I thought I had seen that document before.  I answered you on tfp but I will paste that here for posterity.

// Super Chlorinate.
topic: pool/state/chlorinator
message: {id:1, superChlor: $$VGP, superChlorHours: $$VALUE}

// Pool Setpoint
topic: pool/state/chlorinator
message: {id:1, poolSetpoint: $$VALUE}

// Spa Setpoint
topic: pool/state/chlorinator
message: {id: 1, spaSetpoint: $$VALUE}

The endpoint for the chem controller is the same but only the following values are settable. This endpoint is only available on the next repo posted as of 1/22.
orp.setpoint = integer
orp.tank.level = decimal
ph.setpoint = decimal
ph.tank.level = decimal
calciumHardness = integer
alkalinity = integer
cyanuricAcid = integer
borates = integer <-- This value is ignored for IntelliChem
// Include only the attributes you would like to set so for instance to set the orp setpoint and the tank level do
topic: pool/state/chemController
message: {id:1, orp: { setpoint: 730, tank: { level: 4} } }

// To set the chemical levels
topic: pool/state/chemController
message: {id: 1, calciumHardness: 100, alklinity: 100, cyanuricAcid: 25, borates: 5}

miamijerry, 2021-01-23:
Njspc 6.1.1 Next Build 911
MQTT Feedback
pool/state/chemController
{"id":1,"ph":{"setpoint":$$VALUE:}
Nothing (I am blind, in that I do not know how to find feedback on the command, either it works or not)

pool/state/chemController
{"id":1,"orp":{"setpoint":$$VALUE:}
Nothing

pool/state/chlorinator
{"id":1,"spaSetpoint":$$VALUE:}
and
{"id":1,"poolSetpoint":$$VALUE:}
Backwards - poolSetpoint changes spa setpoint, and spaSetpoint changes pool setpoint.

pool/state/chlorinator
{"id":1,"superChlorinate":$$VALUE:}
Rock solid on-off control of superchlorinator

pool/state/chlorinator
{"id":1,"superChlorHours":$$VALUE:}
Tried "superChlorHours", "superChlor" and "hours" with $$Value or $$VGP
Nothing worked....

rstrouse, 2021-01-23:
I don't have an intelliChem or EasyTouch so I am blinder than you.  However, I will add some messaging to the MQTT output so we can verify that it is making it to the subscription.  It looks like the chemController event isn't being triggered at all.

The backwards setpoints tell me that I need to take a look at the EasyTouch code for the chlorinator.  All the chlorinator settings work fine on IntelliCenter.  I'll take a look at the chlorinator stuff for EasyTouch and add some messaging for mqtt chemController event and let you know when to pull it.

rstrouse, 2021-01-23:
Pull a new njspc.  I found the condition where the chemController subscription jumped the shark with IntelliChem.  I don't have an EasyTouch but all of the endpoints for the chlorinator work great in IntelliCenter.  I have a question out to @tagyoureit to see if the setpoints and hours work on EasyTouch from the dashPanel display.   If they don't then there is an issue with the RS485 comms.

I know I have asked you this before but I assume you have an EasyTouch or IntelliTouch.  Is that correct?

miamijerry, 2021-01-23:
Easytouch2 8

Dashpanel commands work setting superChlorinate hours and start, except I have to have at least 2 hours set, one hour does not work.

rstrouse, 2021-01-23:
Yeah I just talked to @tagyoureit he needs to take a look at the reversed setpoints as well as the superChlorinate settings.  He was aware that there was some weirdness with it.

Did the chemController functions do their thing?

miamijerry, 2021-01-23:
PH and ORP are working PERFECTLY !!!!

5 out of the 6
Backwards setpoints are not a big issue, easy to swap commands.

SuperChlorHours remaining...

tagyoureit, 2021-01-24:
I fixed the backwards setpoints on *Touch.   Also added chlor endpoint for MQTT (/config/chlorinator or /state/chlorinator).

miamijerry, 2021-01-24:
Spa and Pool setpoints are correct, Thank You !

Still can not able to get a working SuperChlorHours mqtt command
Tried these six combinations...

pool/state/chlorinator
{"id":1,"superChlorHours":$$VALUE:}
{"id":1,"superChlor":$$VALUE:}
{"id":1,"hours":$$VALUE:}
pool/config/chlorinator
{"id":1,"superChlorHours":$$VALUE:}
{"id":1,"superChlor":$$VALUE:}
{"id":1,"hours":$$VALUE:}

rstrouse, 2021-01-25:
@tagyoureit this does work properly on IntelliCenter with {id:1, "superChlorHours": $$VALUE;}

tagyoureit, 2021-01-25:
There's a slight difference in how I've implemented superChlorinate on *Touch because it only reports down to the hour whereas IntelliCenter reports down to the minute.  I'm discussing with @rstrouse on the right approach but at least for now with *Touch you need to explicitly enable superChlor.
```
{"id":1,
"superChlorHours": 2,
"superChlorinate": true}
```
Sending superChlorHours will save the value in poolConfig.json but not actually kick off the super chlorination.

miamijerry, 2021-01-25:
I am having no luck with superChlorHours

I have two separate mqtt items, one to control superChlorinate, On/Off
This is rock solid

config/chlorinator
or
state/chlorinator
with payload
{"id":1,"superChlorHours":$$VALUE:}

Does not appear to be writing to poolconfig.json
I changed file permissions on poolConfig.json to anyone, still not writing.
I can manually edit poolConfig, and that works correctly

miamijerry, 2021-02-01:
state/chlorinator
{"id":1,"superChlorHours":$$VALUE:}

Does not appear to be writing to poolconfig.json

rstrouse, 2021-02-03:
Does this work when changing from dashPanel?  There are differences between EasyTouch and IntelliCenter here and it works in IntelliCenter so I need to know if you can click on the chlorinator in dashPanel and change the superChlor hours.  You don't have to start a cycle, simply change the number.

tagyoureit, 2021-02-03:
I tested this in both dashpanel and mqtt and it works.  You are pre-pending with your pool name, correct?  `pool/config/chlorinator`?

miamijerry, 2021-02-03:
Yes, editing super chlorinate hours in dashPanel does change the hours on Easytouch. This is a PUT command

MQTT command
pool/state/chlorinator
{"id":1,"superChlorHours":$$VALUE:}

Does not appear to anything

tagyoureit, 2021-02-03:
I cut out about 5s in the middle... 
https://user-images.githubusercontent.com/1646840/106799779-7f890f80-6614-11eb-846e-4c491de0be88.mp4

Have you pulled the latest `next` branch code recently?

miamijerry, 2021-02-03:
Yes, I am on 150ea19

Would a replay help?

tagyoureit, 2021-02-04:
Sure... are you sure that $$VALUE is sending the right value?

miamijerry, 2021-02-04:
Well this was the correct question to ask. I moved these
MQTT commands
pool/state/chlorinator or pool/config/chlorinator
{"id":1,"superChlorHours":$$VALUE:}
to a different topic, and both commands work fine.

Thank You

So the issue must be how this one topic handles the command, I will refer to the Homeseer mcsMQTT plugin author. I know this one topic was different in that the intend was to send the payload of hours with the "On" command.

Thanks again.

closed 2021-02-04 by miamijerry