tagyoureit/nodejs-poolController #260: Super Chlor can not be cancelled

miamijerry, 2021-01-09:
Easytouch2 8

Can start "Super Chlor" function just fine from dashPanel, but for the time period set for super chlor, I can not turn off pool. The pool is locked out by the super chlor function.

dashPanel does not have a Cancel button for Super Chlor

I can sent put command to poolController to cancel the Super Chlor, from mcsMQTT;
[1/9/2021, 10:02:13 AM] info: [10:02:13 AM] 192.168.0.103 PUT /state/chlorinator/superChlorinate {"id":1,"superChlorinate":false}
but poolController or Easytouch is not cancelling the Super Chlor.

Autelis system is able to cancel the Super Chlor without issue.

rstrouse, 2021-01-09:
Actually, that is not true.  If the superChlor is engaged then the button changes to Cancel.  There might be an issue with the Touch settings.  Are you not seeing this?

[image]

tagyoureit, 2021-01-09:
Per the [MQTT documentation](https://github.com/tagyoureit/nodejs-poolController/wiki/Bindings-Integrations-in-2.0#set-values-with-mqtt), /state/chlorinator/superChlorinate is not implemented.

miamijerry, 2021-01-10:
We are sending PUT commands the same as dashPanel
[image]
Easytouch2 8, no cancel button is displayed

miamijerry, 2021-01-17:
Cancel button now working fine on Easytouch2 8 with poolcontroller 6.1.1 build 909

closed 2021-01-17 by miamijerry