tagyoureit/nodejs-poolController #262: Feature request: Ability to turn off all Features via MQTT

MaxVonEvil, 2021-01-16:
As discussed on Gitter. I would like to request the ability to off multiple (all) Features at once.
(this may be generally applicable for other scenarios, turning on/off  multiple circuits, lights etc, - however I'll describe my specific case for simplicity)

The use-case is to provide a single-select interface to choose pump speed (currently this is a drop-down entity in Home Assistant), yet I'm building towards emulating some nice radio buttons). 

In njsPC I have mapped 6 pump speeds to a corresponding set of Features, named 750RPM, 1200RPM and so on.
Like some others, I do not have any Intelli* gear, i.e. I do not have access to the macros facility they offer.

For now I've currently worked around the current limitation by implementing a flow in Node Red which stores the selected speed value from the HA dashboard, then publishes an 'off' to the setstate topic for all speed selection Features, then recalls the selected value and sends that to the proper circuit.

This feature would likely help others to build intuitive pump interfaces, as the way things work today (pump speeds -> circuits -> highest speed wins) leaves room for improvement.

miamijerry, 2021-01-17:
Can't you just send eight commands?

"yourRootTopic"/state/features/11/feature1:isOn
"YourRootTopic"/state/features/setState 
{"id":11,"isOn":"$$LABEL:"}
Label - 0=Off, 1= On
feature 2 =12, to feature 8 =18

MaxVonEvil, 2021-01-18:
Hi Jerry - that's actually what I'm doing in Node Red at the moment. What I'm suggesting is have a one-off command on the njsPC side which would reduce the inherent complexity.

tagyoureit, 2022-05-23:
I'm going to close this since it's been a while... but I think the Circuit Groups might be able to solve this for you. You can assign the desired on/off status to any number of circuits.  Do this for each of your speed settings and you have built-in macro capabilities

[image]

closed 2022-05-23 by tagyoureit