tagyoureit/nodejs-poolController #263: Next build 909 mqtt.json, alt and original block chemcontroller data

miamijerry, 2021-01-18:
Updated to next build 909, and lost chemcontroller data by mqtt,

Tried all three mqtt.json files; (mqttAlt.json, and mqttoriginal.json) and these each block the chemcontroller data.

Used the modified mqtt.json I had from earlier build (attached), and the chemconroller communication works again.
[mqtt.zip](https://github.com/tagyoureit/nodejs-poolController/files/5831032/mqtt.zip)

tagyoureit, 2021-01-18:
Did you upgrade from `master` to `next`?

miamijerry, 2021-01-18:
Was previously on working "Next" version dated 1-7-2021, before experiencing this issue.
But every time I re-install, I start with Master.

Then "git branch -r" and "git checkout next"

tagyoureit, 2021-01-18:
Sounds like you might not be on the latest version.  You don't need to reinstall everything.  If you are on the `next` branch you only need to do `git pull` followed by `npm run build`.  The bundled mqtt.json should work with the currently published branch.  Because you went back to an older version and it worked it makes me believe you aren't on the latest...

miamijerry, 2021-01-18:
Same result only 191 mqtt items found.

git pull says "Up to date".
git reflog show install at be35d9a

[image]

tagyoureit, 2021-01-19:
Are there actual items missing?  We moved a lot of data/topics around and so if you had the old mqtt.json + the new mqtt.json you would have old data mixed in with new data.  What's missing that you are expecting to be there?

miamijerry, 2021-01-19:
348 - 191 = 157 missing mqtt items, 

most missed are ph level, ph setpooint, orp level, and orp setpoint.

maybe I can create a list??

miamijerry, 2021-01-19:
I am clearing the database between runs of poolController build 909. 
One with the normal mqtt.json found in 909 and then again with the mqtt.json from build 905

[Build 905 mqtt json in build 909 MQTT Topics with chemcontroller items.txt](https://github.com/tagyoureit/nodejs-poolController/files/5836358/Build.905.mqtt.json.in.build.909.MQTT.Topics.with.chemcontroller.items.txt)

rstrouse, 2021-01-19:
I assume you mean you are pruning the tree in mqtt explorer.  I don't have IntelliChem but I do have a REM Chem device defined that uses the same chemistry controller structures.  Here are the steps I took to try to duplicate your issue.  I have both the mqttAlt.json and mqtt.json interfaces defined in the config.json.
1. Stopped njspc then pruned both topics from mqtt explorer.
[image]
2. I then restarted njspc and gave it some time to finish initializing.  Two new topics showed up in MQTT explorer with all the relevant items attached to the pool
[image]
3.  Drilling down on the topic from mqtt.json here is the structure.
[image]
4.  Drilling down on the topic from mqttAlt.json here is the structure.
[image]

miamijerry, 2021-01-19:
I am using mcsMQTT a homeseer automation MQTT broker plugin.

I clear the database of previously received topics in mcsMQTT, upgrade poolController Next to the latest, and then copy the list of topics found from the SQlite database to a txt file for you.

Let me try MQTT Explorer...

miamijerry, 2021-01-19:
OK resolved the issue.

Deleted contents of /home/pi/nodejs-poolController folder.

Downloaded next build 909 zip, wrote contents into
/home/pi/nodejs-poolController
ran npm i
Edited /home/pi/nodejs-poolController/config.json for specifics of mqtt broker

Working properly with 909 mqtt.json

rstrouse, 2021-01-20:
That is so very strange.  There aren't even units on the temps object.  Do you see any errors in the njspc console?

miamijerry, 2021-01-20:
[build 905 mqtt json console silly.txt](https://github.com/tagyoureit/nodejs-poolController/files/5845115/build.905.mqtt.json.console.silly.txt)
[build 909 mqtt json console silly.txt](https://github.com/tagyoureit/nodejs-poolController/files/5845116/build.909.mqtt.json.console.silly.txt)

miamijerry, 2021-01-21:
I was able to get the 909 build mqtt.json to provide orp and ph data by making the attached mqtt edits...

[Edits to 909 mqtt json to get PH and ORP working.docx](https://github.com/tagyoureit/nodejs-poolController/files/5850868/Edits.to.909.mqtt.json.to.get.PH.and.ORP.working.docx)

rstrouse, 2021-01-21:
This is insightful.  Can I get you to run a replay on your njspc.  The issue is likely because of structural differences between IntelliChem and REMChem.  If what I suspect is true this will be a pretty short fix.  There is a difference with the REM devices as the probes report at a higher frequency regardless of whether a dose is being calculated.  The difference between data.pHLevel and data.ph.level is exactly that difference.  

For IntelliChem I assume there should be no differentiation since I believe it only reports the probe data when it is eligible to dose.  This is when there are no lockouts, no alerts, and it has a reported flow.

miamijerry, 2021-01-21:
dumb it down for me...

Replay instructions?
Does it matter which mqtt.json I use?

rstrouse, 2021-01-21:
Doesn't matter which json file you use.  If you are using dashPanel simply click on the burger menu on the upper right, choose the capture replay button.  A screen will come up with an option to capture the reload of the configuration and simply press the begin capture and it will interrogate your system.
[image]

After the system goes back to ready in the upper right then press the cancel capture button and it will download a zip file to your browser.  Upload that zip file here.
[image]

There is a similar function in webClient.

miamijerry, 2021-01-21:
[replay (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/5851449/replay.1.zip)

rstrouse, 2021-01-22:
Ok so I went through the MQTT publish code and cleaned a few things up.  Before I post this up can I get a copy of all of the mqtt json files you are using.  There is clearly a publish in at least one of them where the alarms data is getting written incorrectly.

miamijerry, 2021-01-22:
[905mqtt.zip](https://github.com/tagyoureit/nodejs-poolController/files/5857811/905mqtt.zip)
[909.zip](https://github.com/tagyoureit/nodejs-poolController/files/5857817/909.zip)

rstrouse, 2021-01-23:
Grab a new njspc.  I made several mods to the mqtt processing.

miamijerry, 2021-01-23:
Solid, thank you

closed 2021-01-23 by miamijerry