tagyoureit/nodejs-poolController #265: Solar temperature not being captured and displayed

antamy, 2021-02-03:
I have an Intellitouch i5+3S system with solar water heating panels, with a valve that is actuated based on solar temperature.  The Intellitouch display (Menu/Setup/Equipment/Solar) has Water Solar Present = Yes and Solar Is A Heatpump = No).  The Temp Diff screen from this menu displays a value for the current Solar Temperature.

Solar temperature is displayed as 0 in the UIs and also from /state/temp.
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/5920575/replay.zip)

rstrouse, 2021-02-04:
By chance do you know what the solar temps should have been at the time you captured the replay?

Perhaps:
Air Temp = 58F
Solar Temp = 60F

antamy, 2021-02-04:
Yes, those are correct!

rstrouse, 2021-02-04:
Ok so I modified the code to pull the air temp and solar temp in IntelliTouch.  You will need to switch to the next branch.

Navigate to your nodejs-poolController directory and type the following commands.
```sudo git checkout next```
```sudo git pull```

Then restart njspc.

antamy, 2021-02-05:
> 
> 
> Ok so I modified the code to pull the air temp and solar temp in IntelliTouch. You will need to switch to the next branch.
> 
> Navigate to your nodejs-poolController directory and type the following commands.
> `sudo git checkout next`
> `sudo git pull`
> 
> Then restart njspc.

Not sure if my email to you went through, but this is now all working!  Thanks!

tagyoureit, 2021-02-06:
Closing this to continue with replays on #266.

closed 2021-02-17 by tagyoureit