tagyoureit/nodejs-poolController #266: Solar heat status not being displayed

antamy, 2021-02-05:
I have an Intellitouch i5+3S system with solar water heating panels, with a valve that is actuated based on solar temperature. The Intellitouch display (Menu/Setup/Equipment/Solar) has Water Solar Present = Yes and Solar Is A Heatpump = No).  The system has a heater (a heatpump) that is controlled directly from the Gas Heater terminal.

The Pentair wall display has two icons to indicate heating - one pulses when solar is active, the other pulses when the heater is active.  The heat mode for the pool is currently set to Solar Only and the Pentair wall display shows that the solar is active.  The dash panel reports Heat Mode = Solar Only, Heater Status = Off for the pool.   Shouldn't there also be a Solar Status indicator indicating that solar is active?

I am currently running the next branch of the software with the recent fix for Solar Temp, and the master branch of dashpanel

[heaters.zip](https://github.com/tagyoureit/nodejs-poolController/files/5934472/heaters.zip)
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/5934482/replay.zip)

tagyoureit, 2021-02-06:
In the Gitter chat :point_up: [January 31, 2021 11:43 AM](https://gitter.im/nodejs-poolController/Lobby?at=601708529fa6765ef8ec3d7d) you mentioned you have a "heat pump".  Before I dig into this, do you mean that you have an UltraTemp Heat Pump or do you have a Solar System with Heat Pump Option?  Sounds like you have the former (UltraTemp Heat pump), correct?

When you go to your ICP (indoor control panel) what heat modes do you have?  Is it off/heater/solar/solar pref or off/heater/heat pump/heat pump pref?  

Also, when you go view the heat status in dashpanel it will show you the _current_ status.  If you click on that it should also bring up the options for which you can set the heat status like so:
[image]

antamy, 2021-02-06:
I do have a heat pump, an Aquacal.  But ignore that for this.  It is connected to the system via the gas heater pins on the system board, so it just looks like a dumb heater.  There is no data exchanged between them.

I see the pop up dialog too and that seems to work correctly.  What is not working correctly is status.   I don't know if Heater Status has a On | Off | Solar - I only see Off when just the solar is activated.    And then when I run the spa, which automatically turns on the heater, the Status is Off.

I can play around with this over the next few days, so hopefully we can track this down.

tagyoureit, 2021-02-07:
First, so we are on the same page...
Heat Mode = off/heater/solar only/solar pref.  These are the modes for which you can set the body.
Heat Status = off/heater/cooling/solar.  It shows the current active status of the heating system.

> The dash panel reports Heat Mode = Solar Only, Heater Status = Off for the pool. Shouldn't there also be a Solar Status indicator indicating that solar is active?

If these truly are the modes, then no... you can have the body set to "solar only" (heat mode) but it is the middle of the night (or the pool temp is > set point) so the solar (heat status) isn't active at the moment.  

There seems to be a mismatch between what you are saying is happening above and what I see in the replay.  I don't see dashPanel reporting Solar Only.  It shows the heat mode as off and also the heat status as off.  

[image]

Given this, it's hard to track down anything.  Can you try to send another packet capture of longer duration when you change the heat mode (either/both from the wall panel or dash panel) and let me know where it doesn't match?  Feel free to change over to the spa and change the heat modes and include those with the packet capture.

antamy, 2021-02-07:
I have attached some captures.  I'd suggest you change the text on the UI to be clearer - currently it says "Heater" status but from what you are saying, this seems to be used for "Heating Status" or "Heat Status", not the actual status of the Heater equipment.  Maybe borrow the icons from Pentair (a sun for solar, a flame for heater, both of them pulse / animate when that heat source is active)?

The captures are:

replay-1

- Pool=On, Mode=Solar Only, Status=Off
- ICP and dashPanel are consistent
- change on ICP to Mode=Heater after 15s.  
- After about 10s, ICP displays the pulsing flame icon.  
- dashPanel shows Heat Mode=Heater, Heater Status=Off (ie incorrect, the heater is on)

replay-2

- Pool=On, Mode=Solar Only, Status=Off; Spa=Off, Mode=Heater, Status=Off
- ICP and dashPanel are consistent
- on ICP, turn Spa on after 15s
- dashPanel has green circles to show Spa is on, Spa Heat Mode=Heater, Heater Status=Off. ICP shows pulsing flame next to Spa

dashPanel UI also screwed up after stopping the logging and returning to the Home screen.  The Temperature section disappeared.  On refreshing the browser, it restores but with no Set Points displayed.  It remains like this until I restart the poolController.

[image]

I will try to get a capture with the solar on later today - the weather may not co-operate, though!

Thanks

[replay-1.zip](https://github.com/tagyoureit/nodejs-poolController/files/5939331/replay-1.zip)

antamy, 2021-02-07:
Forgot to upload replay-2.zip
[replay-2.zip](https://github.com/tagyoureit/nodejs-poolController/files/5939384/replay-2.zip)

rstrouse, 2021-02-07:
Guys, I shot a screenshot of my IntelliCenter on dashPanel with the heater running on the spa.  Obviously, given the current temps outside I cannot kick on the solar.

When solar is on there will be a glimmering sun in the position where the flickering flame is located.  Also, the heater status will show Solar.
[image]

antamy, 2021-02-07:
Is that flame icon in the current build?  I've never seen it!   

Also, if you have an Intelliflo do you see anything other than 0 for gpm? (sorry for changing the topic!)

rstrouse, 2021-02-07:
Yes it is and yes I do.
[image]

tagyoureit, 2021-02-07:
I think you might have some comms issues in your console log.  I see a lot of the "2" packets (which are the longest ones and contain the overall system status) are corrupt.

```
[2/7/2021, 3:46:14 PM] warn: {"id":72701,"valid":false,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255],[255,0,255],[165,1,15,16,2,29],[15,46,40,0,0,0,0,0,0,0,71,0,0,0,71,71,32,73,72,73,0,0,5,4,53,189,127,215,203],[252,255]],"ts": "2021-02-07T15:46:14.150+0000"}
[2/7/2021, 3:46:14 PM] debug: Packet not processed: 215,203,252,255,0,255,165,1,18,16,110,2,0,0,1,56
```

This can leads to all sorts of problems.  Can you try rebooting and see if these same issue and invalid messages disappear?  And make sure nothing else is using the same comm port?  If they do persist, we can give you a small script that will hard-reset your rs485 port.

I wonder if this is the cause of this.  I've never seen an Intellitouch have different packet definitions from another Intellitouch.

Also, @rstrouse, any idea why replay-2 would suddenly start to look for virtual controllers in the middle of the config queue?
```
[2/7/2021, 3:56:22 PM] verbose: Config Queue Completed... 92% (14 remaining)
[2/7/2021, 3:56:22 PM] verbose: Config Queue Completed... 92% (14 remaining)
[2/7/2021, 3:56:23 PM] info: {"id":73812,"valid":true,"dir":"out","proto":"broadcast","pkt":[[],[255,0,255],[165,1,16,33,216,1],[6],[1,182]],"ts":"2021-02-07T15:56:23.028+0000"}
[2/7/2021, 3:56:23 PM] info: {"id":73812,"valid":true,"dir":"out","proto":"broadcast","pkt":[[],[255,0,255],[165,1,16,33,216,1],[6],[1,182]],"ts":"2021-02-07T15:56:23.028+0000"}
[2/7/2021, 3:56:23 PM] verbose: Wrote packet [255,0,255,165,1,16,33,216,1,6,1,182].  Retries remaining: 3
[2/7/2021, 3:56:23 PM] verbose: Wrote packet [255,0,255,165,1,16,33,216,1,6,1,182].  Retries remaining: 3
[2/7/2021, 3:56:23 PM] info: Searching chlorinators, pumps and chem controllers
[2/7/2021, 3:56:23 PM] info: Searching chlorinators, pumps and chem controllers
[2/7/2021, 3:56:23 PM] info: Searching for a pump at address... 96
[2/7/2021, 3:56:23 PM] info: Searching for a pump at address... 97
[2/7/2021, 3:56:23 PM] info: Searching for a pump at address... 96
[2/7/2021, 3:56:23 PM] info: Searching for a pump at address... 97
```

rstrouse, 2021-02-07:
Once it ascertains the OCP it shouldn't look for those.  However, I do believe the replay clears the OCP before it starts looking again because it constructs a new PoolSystem.  However, if this were occurring I would have expected to see RESETTING DATA before it starts.

The interesting thing here is that it looks like it isn't processing the Action 2 message correctly.  In this case it looks like the byte signature changed.

antamy, 2021-02-07:
I powered down the Pi, checked the wires on the RS485 USB stick and powered back on.  I tried the replay-1 scenario again and got the same results.  I do see some of those warn messages in the log.  I have my cable running from the ICP to the Pi.  There is nothing else using the Pi - I just rebuilt it a few days ago.

I can try the reset script if you want to send it along.  Does that reset on the Intellitouch itself?

Thanks
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/5939568/replay.zip)

tagyoureit, 2021-02-07:
[FixSP.zip](https://github.com/tagyoureit/nodejs-poolController/files/5939608/FixSP.zip)
Try this.  You'll need to unzip it to it's own directory and then run `npm i` to install dependencies.  After you do that, run `node captureChanges --port /dev/ttyUSB0`. You should see it trying to parse some messages.  Sometimes I have luck leaving this open in the background when running njsPC, other times I can close it and the port works, and yet other times I need to run it and reboot and it seems OK.

This script hard resets the registers on the RS485 chip so it can process the packets correctly.

rstrouse, 2021-02-07:
@tagyoureit -- Do you have heaterOptions populated on your temps.bodies[x] object in poolState?  The list is smaller in Touch but it should be there.

```
       "heaterOptions": {
          "total": 1,
          "gas": 0,
          "solar": 1,
          "heatpump": 0,
          "ultratemp": 0,
          "hybrid": 0,
          "maxetherm": 0,
          "mastertemp": 0
        },
```

tagyoureit, 2021-02-07:
Yes.
```
        "heaterOptions": {
          "total": 2,
          "gas": 1,
          "solar": 1,
          "heatpump": 0,
          "ultratemp": 0,
          "hybrid": 0,
          "maxetherm": 0,
          "mastertemp": 0
        }
```

rstrouse, 2021-02-07:
This is missing on @antamy's poolState.  It appears it gets set on the Action 34 or 162 message.

tagyoureit, 2021-02-07:
I see that packet in replay-1 but not replay-2.
`{"id":72364,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255],[255,0,255],[165,1,15,16,34,3],[7,0,70],[1,55]],"ts": "2021-02-07T15:45:32.555+0000"}`

tagyoureit, 2021-02-07:
After running through the replay-1, his poolState does have the heaterOptions
[poolState copy.txt](https://github.com/tagyoureit/nodejs-poolController/files/5939679/poolState.copy.txt)

antamy, 2021-02-07:
ok, I've been running this.  I installed it, stopped njsp and ran it.  I see lots of Valid 2 messages, which I assume is positive.   Then I start njsp.  dashPanel shows the Loading xx% message and it takes quite some time to get to 100%.  During that time, the packet logger is showing lots of Invalid 2 messages.  Once it gets to 100%, that seems to settle down, but I periodically see Invalid messages, eg:

```
Invalid 2
INVALID
{"type": "Broadcast", "direction": "inbound", "packet": [255, 0, 255, 165, 1, 15, 16, 2, 29, 12, 55, 32, 0, 0, 0, 0, 0, 0, 0, 67, 0, 0, 0, 72, 72, 255, 0, 255, 165, 0, 96, 16, 4, 1, 255, 2, 25, 255, 0, 255], "message": "29", "timestamp": "2021-02-07T17:58:29.131Z"}

INVALID
{"type": "Chlorinator", "direction": "inbound", "packet": [16, 2, 0, 3, 0, 73, 110, 116, 101, 108, 108, 105, 99, 104, 108, 111, 114, 45, 45, 52, 48, 188, 16, 3], "message": "19", "timestamp": "2021-02-07T18:01:19.173Z"}
```

No change in UI behavior when I turn the heater on

tagyoureit, 2021-02-07:
Don't worry too much about the output of the captureChanges script.  It doesn't really process all the packets correctly.  It is mostly used to set the serial port comms correctly.

rstrouse, 2021-02-07:
I do not believe Replay 2 finished.  There is only one body and it doesn't have a controller type.  This tells me that replay-2 had bad comms at the time it was captured.  Replay-1 does acquire the i5+3.  

Just to be sure this is an i5+3 and not an i5+3S or i7 or i9?

antamy, 2021-02-07:
> Just to be sure this is an i5+3 and not an i5+3S or i7 or i9?

I just checked on the physical panel.  It's an i5+3.

rstrouse, 2021-02-07:
Ultimately it looks like the problem is simply this.  The heaterOptions object is not being populated.

Everything else is acquired.
tempSensors --- Check
bodies --- Check
heaters --- Check

The only thing that is missing is heaterOptions.

antamy, 2021-02-07:
I checked the wiring at the ICP and took another capture.   This time, I had the heat mode set to heater and the heater was on when I started the capture.  The solar temp is now 78F, so warm enough for it to kick in.  20 seconds after starting the capture, I switch to Solar Only.  The UI updates the mode, but still no update to the actual status. ICP shows that solar heating is active.

[replay(1).zip](https://github.com/tagyoureit/nodejs-poolController/files/5939717/replay.1.zip)

tagyoureit, 2021-02-07:
These packet captures are only 1 minute long.  Can you send a longer one?  One thing to keep in mind is that during the initialization phase (which ended 52s into the 59s packet capture) is that you may be trying to set modes the pool doesn't know about yet.  This will be indicated by the upper right in dashpanel changing from 'initializing' with a % to "Ready".  

Once you reach the ready state, the change a mode somewhere.  Wait 30s or more.  Then change to another mode.  

As @rstrouse is saying, if we don't know that your heatermodes have been discovered yet then we won't set it as it isn't a valid mode.  And when you start a packet capture with the reset option it wipes our your config and starts from scratch.

antamy, 2021-02-07:
ok, I'll try that tomorrow.  Thanks for all your help on this.

antamy, 2021-02-08:
This trace is for about 3.5 minutes.  Started with Heat Mode=Solar, Status (ICP and dashP)=Off.  Took about 50s for the Loading to reach 100%.  90s from the start I switched via the ICP to Heat Mode=Heater.  ICP and dashP both displayed that mode, ICP Status is On, dashP is Off.

[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/5944908/replay.zip)

tagyoureit, 2021-02-09:
I'm seeing some different packets in your logs than I would expect for similar systems.  This is making me wonder if you have additional bodies on your system (possibly with an expansion card??).

I'd like to see if you can send another capture with the initialization that Screenlogic initializes.  If you can start a packet capture like your normally done (wait for njsPC to finish) and then change the pool or spa setpoints 2 or 3 times.  This will usually make SL initiate a full refresh.  Or you can go to your ICP and request the configuration from the OCP.    

Also, do you have the computer SL software?  If so, I'd be curious what appears on the loading screen (here is an example).
[image]

antamy, 2021-02-09:
I don't have Screenlogic.  I can't really see the screenshot you attached, however my system reports firmware v 1.17.

It is entirely possible that there may be some odd configuration on this system. It is only 5 years old and I haven't seen anything in the panel that looks like an expansion card.  However, I do have a menu item for Aux4 that I have no idea why it is there or what it does!  My equipment is just an i5+3 board, an Intelliflo pump, a Pentair chlorinator, one string of Intellibrite lights and a set of lights on my boat dock.

Based on this, let me know if you want an additional capture.

Thanks

[image]

antamy, 2021-02-09:
Ok, I think I understood what you meant abut the configuration.  I also confirmed that my Aux3 is for the dock lights and Aux 4, although wired on the board, connects to a relay that does nothing.

I captured a trace as follows:

- pool pump on, heat=solar, but solar not heating
- njsPC fully initialized after 50s
- after 60s used ICP to request the configuration from the OCP
- after 120s, change pool heat mode to Heater
- after 210s, changed pool heat mode to Solar Pref
- after 270s, turned on Spa

All of these actions changed dashPanel as expected, except Heater Status remained Off.  

Also gpm on the pump is always 0, which is not surprising since the pump only reports rpm and watts (3,000 rpm, 1,600 watts when the Spa is running)

[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/5954658/replay.zip)

rstrouse, 2021-02-10:
In looking at the message capture with regard to the pump flow.  The pump appears to not be reporting the flow.  Do you see this in your indoor control panel for the pump?  In fact the message signature from the pump appears to be a VS pump not a VF pump.  Are you sure of your pump model?  Only the VF or VSF pumps report GPM.

The message below is your pump reporting its status.  The byte that is circled would normally indicate the flow (gpm).  It is however reporting a speed of 3,000rpm which are the two previous bytes.
[image]

The message sequence below from your capture is IntelliTouch setting the pump to 3000rpm and the pump responding that it set itself to 3000rpm.  If this were a flow based pump the last byte on the payload would be 0 and it would be setting 11gpm.
[image]

antamy, 2021-02-10:
That's a good catch!  My pump is Variable Speed, so the UI doesn't match it.
 On the ICP, under setup/equipment/intelliflo/status, I see:

[image]

rstrouse, 2021-02-10:
Do you by chance know the model number for this pump.  We are getting a signature of 134 under normal circumstances a VS pump is 128.  Is this an IntelliFlo, IntelliPro, or SuperFlo pump?
[image]

antamy, 2021-02-10:
I'll have a closer look in the daylight, but I see 354605.E on the label.  It's an Intelliflo Variable Speed Ultra Energy Efficient Pump.  Maybe there's more info in the menu on the display.

rstrouse, 2021-02-10:
Interesting the Pentair VS pump part number is typically 011028 or thereabouts depending on the year it was made.  By chance is this a Sta-Rite pump or a VSOM or XF or do you see a part number 011056?

antamy, 2021-02-10:
There is just this label on the housing and part# 353251 on the drive unit.  

[image]

rstrouse, 2021-02-10:
The pump appears to simply be a 011028.  It probably has some certification or tweaks to its power curve that gives it a different designation.  I hope to have this resolved today after I get some more information from @tagyoureit on the way VF versions are processed.

rstrouse, 2021-02-10:
If you pull njspc the pump woes should be accommodated.  There is still some weirdness with your heaterStatus so we may need a very specific sequence recorded and uploaded.  This will give us a better picture of the differences between your status message and the one we have typically seen in the wild.

antamy, 2021-02-10:
Ok, I'll take a look. I had to head back to the frozen north for a week, but I can play around with this remotely

tagyoureit, 2021-02-10:
We're still trying to understand why your heat status' come through differently.  Can you run through all the combinations of bodies/heat modes?   Again, feel free to leave ~30 seconds between each step (unless you see it successful in njsPC).

1. All bodies set to no heater.
2. Set Pool to Solar
3. Set Heat Setpoint to 90 (to turn it on)
4. Set Heat Setpoint to 50 (to turn it off)
5. Set Pool to Solar Pref, repeat steps 3/4
6. Set Pool to Heater, repeat steps 3/4
7. Set Pool to no heater
8. Turn on Spa
9. Repeat steps 2-7 but for the spa instead of the pool.

Bonus points if can have a video of your ICP while you are doing this.  :)

Appreciate your patience/help trying to figure this out.

antamy, 2021-02-12:
ok, here you go.  I couldn't record the ICP since I'm 1,200 miles away from it :-(    But I recorded the dashPanel I was using.

Not exactly 30 seconds between each change, but there was no loading showing on the dashPanel.  Also note there is something odd going on with the temps - I thought I set to 90, but it set to 100.  And when I reduced the spa temp I couldn't get below 100.  Ultimately, refreshing the dashPanel let me set that.

[2021-02-12 15-45-55.zip](https://github.com/tagyoureit/nodejs-poolController/files/5974458/2021-02-12.15-45-55.zip)
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/5974459/replay.zip)

rstrouse, 2021-02-15:
We have been going through your message output and have noticed something that neither of us see.  By chance do you have a cooldown delay set for your heater?  This is required for some heaters (not MasterTemp) but I am unsure of AquaCal heat pumps which I assume you are operating just like a gas heater and the heat pump itself determines whether it is more efficient to use the heat pump or to fire the heater.  The solar side of this is actually connected to solar panels I assume.

antamy, 2021-02-15:
No, there’s nothing like that.  The heater is just connected to the Gas Heater pins on the board.  I do not think there is a cooldown delay, but I can’t check that right now.  There is no communication between the heater and the Intellitouch.  The solar side is just a valve (Valve A) that diverts water to tubes on the roof.  There is no intelligence in the system as to whether the Aquacal is more efficient.  The only intelligence is the heat differential between the solar temp and the pool temp – on Solar Pref, the heater turns off once the differential is exceed (mine is set at 6 degrees).

rstrouse, 2021-02-15:
Out of curiosity please check on your OCP for the following settings.
Spa Manual Heat - This is located under setup -> equipment -> manual heat and will be yes or no.
Master Delay - This is located under setup -> equipment -> filter delay and will be yes or no.

rstrouse, 2021-02-16:
Pull next.  I believe we fixed the issue with the heat status.

antamy, 2021-02-16:
Well, this is progress!  I now see the flickering flame for the heater.  But the heat mode is incorrect, even though the pop up menu displays the mode correctly.  This screen shot is immediately after njspc has started and the dash has refreshed.  If I change the heat mode in the popup, the main panel changes too.

[image]

There is also a UI issue when Solar Preferred is selected:

[image]

Lastly, although I'm not at the house to check, the Intellichlor is always on when the pump is on.  It is currently showing off and no output %.  The Clean Cell label is correct and the ppm seems correct.

Thanks!

rstrouse, 2021-02-16:
I'll work backwards from your list.  IntelliChlor does not start reporting current output right away sometimes when the body is activated.  We see this in the message returns from the cell itself.  It's just a theory but I think when it samples conductivity to calculate the salt level it gets a bit tardy.  I haven't put a pin in when it does this exactly but folks have reported that it does this every 24 hour cycle but I am pretty sure that isn't right because I have witnessed it sooner.  Give it a minute or two and it will start reporting its status.

The words solar preferred used to say solar pref.  Because it is now spelled out it pushes the flame off the panel.  I have moved the flame back into the panel when it is in this mode.  You will need to pull dashPanel and hard refresh your browser.

With the initial heat modes there is a bit of a chicken and egg thing going on with it.  When we are first acquiring the panel data it doesn't have all the information on the bodies but by this point it has already reported the heat mode.  @tagyoureit made some changes to njspc early this morning that allows the setpoints and heat mode to be acquired while it is waiting on the remaining body information.  Pull njspc and try again.  NOTE: this should only happen when njspc is getting the initial data from the OCP.

tagyoureit, 2021-02-16:
I generally find that I have to refresh the browser after the first initial load from empty config files.  There are a few small things that don't sync/update automatically or refresh properly.  We can try to work on those, but if you hit refresh and it all looks good you shouldn't encounter these small issues ever again (unless you reset your configuration files).

antamy, 2021-02-16:
I don't see a newer version of dashPanel right now.  I'll keep an eye out for it.  Seems that everything is working, the Intellichlor info refreshed itself.  Thanks for all your work figuring this one out!

rstrouse, 2021-02-16:
Yeah that's because some putz that uses my name forgot to hit the commit button.  It is there now.

rstrouse, 2021-02-16:
Just a quick tip for you if you uncheck Show as Feature on the pool and spa circuits it will remove them from the features list and you can click them from the bodies section.   If you do the same with Dock lights and Pool Light it will move them to the lights section.

antamy, 2021-02-16:
The current dashPanel is displaying correctly.  As an added bonus, it seems to display better on my phone too, but maybe that's just me.  The settings page with all the tabs does not display correctly, but that happened previously.

Thanks for the tip on the features.  Any idea why Pool Light and Intellibrite show, when they are both the same thing?  I have noticed a couple of other glitches which I'll open up issues for if I can reproduce them.  One was around automatically setting TOD, the other was in settings and caused my system to lock up!  I'll see if I can track that down next time I'm on site.

Thanks!

tagyoureit, 2021-02-16:
Intellibrite function contains the commands for the colors and can be a single light or a group of lights.  It makes a bit more sense if you have >1 Intellibrite light.  

I just noticed a couple of small things with Intellibrite, too (doesn't show the synchronizing message; and it isn't turning the circuit on/off).

rstrouse, 2021-02-16:
Yeah the settings page was never meant to be operated from a screen the size of a phone.  I had thought about disabling it on a phone but there are some minor things that can be done with it at this point and it does scroll around.  We may go back in at some point and add some reactive css to it but that is pretty low priority at this point.

tagyoureit, 2021-02-17:
Since the initial issue was resolved, I'm going to close out this case.  Please open a new case for any subsequent issues.  Thx!

closed 2021-02-17 by tagyoureit