tagyoureit/nodejs-poolController #267: Pentair IntelliPH?

brentratliff, 2021-02-11:
Any idea it this will work with IntelliPH? In the market to add acid and chlorine pumps for automation.

brentratliff, 2021-02-13:
Not sure if others want this answered, but I went with an Intellichem which I know works. Thanks for the good work on this.

closed 2021-02-13 by brentratliff

tagyoureit, 2021-02-13:
See https://github.com/tagyoureit/nodejs-poolController/issues/124.  IntelliPh doesn't send packets on the RS485 bus, it seems to have a listen-only mode.

johnny2678, 2021-02-13:
Yep, I recently removed the IntelliPH control board because I couldn't interact with it over the bus.  I kept the tank/pump and wired up my own automation.  Now, whenever pH readings get too high, the pool gets a quick dose from the acid pump via a solid state relay powered by a raspberry pi.

tagyoureit, 2021-02-13:
That's exactly what we built REM (relay equipment manager) for.  It's a companion for 6.5.0 and controls gpio, spi and i2c devices.

[image]

[image]