tagyoureit/nodejs-poolController #268: SWG setPoint changes immediately revert

johnny2678, 2021-02-21:
**To Reproduce**
In dashPanel:

- change pool setPoint using the +/- (or manual entry)
- it will revert to original value

**Expected behavior**
new setPoint will be committed

[image]

**Packet Capture**
attached...
[replay (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/6018118/replay.1.zip)

**Pool Equipment**
 - Controller: EasyTouch 2 4
 - Controller Firmware Version: not sure
 - Pump(s) manufacturer and model: Pentair VS
 - Chlorinator: IntelliChlor-40
 - Heater(s): Pentair Mastertemp
 - Chemical controller: homegrown

**Additional context**

looks like SWG packets aren't completing
```
3|poolController  | [2/21/2021, 11:23:59 AM] info: [11:23:59 AM] 192.168.5.31 PUT /state/chlorinator {"id":1,"poolSetpoint":75}
3|poolController  | [2/21/2021, 11:23:59 AM] verbose: Wrote packet [255,0,255,165,56,16,33,153,10,1,15,0,0,0,0,0,0,0,0,1,193].  Retries remaining: 3
3|poolController  | [2/21/2021, 11:23:59 AM] verbose: Wrote packet [255,0,255,165,56,16,33,217,1,0,1,232].  Retries remaining: 3
3|poolController  | [2/21/2021, 11:24:08 AM] debug: Packet not processed: 255,255,0,255,165,56,16,34,215,1,1,1,232
3|poolController  | [2/21/2021, 11:24:08 AM] debug: Packet not processed: 255,255,255,255,255,255,255,255,0,255,165,56,15,16,23,16,1,0,0,0,1,245,8,52,0,0,0,0,0,1,0,128,2,215
3|poolController  | RS485 Stats:{"bytesReceived":385536,"success":13282,"failed":191,"bytesSent":10858,"collisions":0}
3|poolController  | [2/21/2021, 11:24:12 AM] debug: Chlorinator message marked as invalid after not finding 16,3 in payload after 26 bytes
3|poolController  | [2/21/2021, 11:24:12 AM] debug: Chlorinator message marked as invalid after not finding 16,3 in payload after 26 bytes
3|poolController  | [2/21/2021, 11:24:14 AM] info: [11:24:14 AM] 192.168.5.31 GET /state/circuits {}
3|poolController  | [2/21/2021, 11:24:14 AM] info: [11:24:14 AM] 192.168.5.31 GET /state/circuits {}
3|poolController  | [2/21/2021, 11:24:15 AM] info: [11:24:15 AM] 192.168.5.31 GET /state/pumps {}
3|poolController  | [2/21/2021, 11:24:15 AM] info: [11:24:15 AM] 192.168.5.31 GET /state/pumps {}
3|poolController  | [2/21/2021, 11:24:15 AM] info: [11:24:15 AM] 192.168.5.31 GET /state/chlorinators {}
3|poolController  | [2/21/2021, 11:24:15 AM] info: [11:24:15 AM] 192.168.5.31 PUT /state/chlorinator {"id":1,"poolSetpoint":75}
3|poolController  | [2/21/2021, 11:24:15 AM] verbose: Wrote packet [255,0,255,165,56,16,33,153,10,1,15,0,0,0,0,0,0,0,0,1,193].  Retries remaining: 3
3|poolController  | [2/21/2021, 11:24:15 AM] verbose: Wrote packet [255,0,255,165,56,16,33,217,1,0,1,232].  Retries remaining: 3
3|poolController  | [2/21/2021, 11:24:16 AM] silly: Retrying outbound message after 1.007secs with 3 attempt(s) left. - 165,56,16,33,217,1,0,1,232
3|poolController  | [2/21/2021, 11:24:16 AM] verbose: Wrote packet [255,0,255,165,56,16,33,217,1,0,1,232].  Retries remaining: 2
3|poolController  | [2/21/2021, 11:24:17 AM] silly: Retrying outbound message after 1.08secs with 2 attempt(s) left. - 165,56,16,33,217,1,0,1,232
3|poolController  | [2/21/2021, 11:24:17 AM] verbose: Wrote packet [255,0,255,165,56,16,33,217,1,0,1,232].  Retries remaining: 1
3|poolController  | [2/21/2021, 11:24:32 AM] info: [11:24:32 AM] 192.168.5.31 GET /state/circuits {}
3|poolController  | [2/21/2021, 11:24:32 AM] info: [11:24:32 AM] 192.168.5.31 GET /state/circuits {}
3|poolController  | [2/21/2021, 11:24:33 AM] info: [11:24:33 AM] 192.168.5.31 GET /state/pumps {}

rstrouse
@rstrouse
11:31
```

Happy to provide more information - just let me know what you need.

rstrouse, 2021-02-21:
Alright so right off the bat it looks like you are sending a setpoint from dashPanel 20 and njspc is sending 15 no matter what.
[image]

I have narrowed down the code to the way Touch is sending the new values to the OCP.

rstrouse, 2021-02-21:
Alright, so I have reworked the chlorinator setpoint values to reflect the correct incoming values.

From the nodejs-poolController directory run the following commands.  This will move you to next.
`git checkout next`
`git pull`

I you are running on a pi you may need to use sudo in front of those commands.   Btw this will also make the display function work on the schedules for you.

rstrouse, 2021-02-21:
Also, tell me a little bit about your homegrown chem controller.  I am in the midst of setting mine up.  Below is the interface that is shown in dashPanel.   It controls the dosing amounts by calculating a demand then engaging peristaltic pumps for dosing, interfaces with pH and ORP feeds, and it works with chlorinators as well.  It maintains dosing and mixing schedules as well as emitting the data to Influx for graphing and history to include:
1. Acid Demand
2. Saturation Index
3. Chem tank levels (these are calculated no need for a sensor since they are calibrated to the pump)
4. Flow Sensor (If installed)
5. Dosing lockouts
6. Temperature compensation
[image]

johnny2678, 2021-02-21:
**that was it.**  setPoints are holding now on the latest `next` branch.  Thanks for the quick attention to this.  I will observe and reopen if there are other issues but I think **you can consider this closed.**

As to my homegrown chem controller, it does some of what you've described. I don't have tank level sensors or a flow sensor.  But it does dose acid and log to influx.  I didn't get too fancy with it.  Just a 10 second dose every 5 minutes pH reads below the threshold I set.

[image]

the gap in reading you see around 10am is when the spa runs for 15 minutes every day and the gap around 1700 is when I took down npcjs yesterday to upgrade to 6.5 ;)

Using a canabilized IntelliPH that I could never get a bus reading on.  So I junked the dumb Pentair pH dosing timer and hijacked 24v from the SWG Powercenter and put it on a solid state relay controlled by rPi GPIO. 

For SWG, my thought was it really only works in 2 states - on/off.  You control how much FCL is produced by the % setting.  So I just run the SWG at a level that is slightly inadequate for my environment to maintain FCL > 1-2, around 15%, and then boost that to 75% when ORP crosses the threshold.

I am tracking a weird issue where SWG dosing actually drives down ORP so my super unscientific fix was to invert my logic so SWG only get's boosted to 75% when ORP rises above the threshold I set.  hey, it's keeping my water clear so as far as I'm concerned, it works 🤷‍♂️🤪

[image]

Happy to post more about it - took me a couple of years of bouncing ideas around on Tag's gitter before it came together.

(btw, I did download relayEquipmentManager but I'm going to need some more time with it ;)

rstrouse, 2021-02-21:
Everything you need to set this up is included in your current build with the exception of RelayEquipmentManager.  This uses RelayEquipmentManager (REM) to manage the hardware.  Ultimately, @tagyoureit and @gw8674 are live on this controller.

For my acid dosing I bought a Pentair Acid Tank and pump (same one you are using) and @gw8674 converted his IntelliPh.  @tagyoureit is using a different setup but it is humming along with Stenner pumps and Tanks.  I purchased Altas scientific industrial probes for pH and ORP as well as a Hayward flow cell with flow sensor.

I don't have the system up and running as I put myself in the grunt chair to get it all coded.  I just got some more parts yesterday but I tend to go all the way when I do things so I am jackhammering out a section of the pad to consolidate my fire control mechanisms and get prepared for future DMX lighting on the pool.

Btw if you have REM downloaded you can set it up and play around with it.  In the configuration screens under the chemistry tab you have the ability to add a REM Chem controller to your system and the connection to REM is located in the config.json.  As an aside you can play around with it without sensors but it really does need them to actually calculate a dose.  Oh and one other thing ORP is a strange duck as it is affected by UV and flow in the cell.
[image]

johnny2678, 2021-02-21:
your parts list sounds very similar to mine ;) I went with the same hayward flow cell and the cheaper Atlas consumer probes,  so it will be good to have a comparison data point on your industrial probes.

I'll have to ping you later on getting REM set up.  Pulled up the UI but didn't really know what I was looking at yet.  In addition to the relay and probes, I also have three ds18b20 therm sensors that I'm reading with the rPi gpio.  Would be great throw away my 1000+ lines of poorly coded python for this. 

for the atlas sensors, did you use the i2c tentacle hat?  I have that but not sure what to do in REM to get it to interface.  Currently i2c tab looks like this... 
[image]

rstrouse, 2021-02-21:
Yes I used the tentacle hat.  For temp support, I think @tagyoureit added the appropriate device drivers for ds18b20, there is a highly accurate SPI control that will let you use any 10k temp sensor, and/or the I2C Atlas EZO-RTD.  Any one of these will work and can be directly fed to the pH sensor.  Alternatively, you can also feed your Pentair temp sensor back into REM for the pH sensor.

If something else is using the I2C bus it will be locked out.  run i2cdetect 1 -y to see if there are any sensed probes on the bus.  Also if you run REM on a PC it will mock up some addresses for you.
Here is what mine looks like for my test pi at the moment.  The probe one of the pro versions sitting in storage solution.
[image]

johnny2678, 2021-02-21:
hmm not sure what I'm missing - probes are being read by python every 10 seconds on i2c address 98,99 (orp/ph) but I don't see what you see on your screen.

i2cdetect output looks like this:
```
pi@PiPool:~/PiPool $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- 62 63 -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --         

                
pi@PiPool:~/PiPool $ i2cdetect -l
i2c-1	i2c       	bcm2835 I2C adapter             	I2C adapter
```

wife's making me do a thing so I'll have to check in later.

rstrouse, 2021-02-21:
Yeah mine has me doing a thing to.  Projects, projects, projects!  I'm starting to feel like Jan Brady.
Make sure you have the latest relayEquipmentManager and check to see if the bus acquired an accessible path.  You may need to run REM with sudo if you didn't go in and assign rights to the pi user. `sudo npm start`
[image]

johnny2678, 2021-02-22:
ok, REM works with I2c now... sort of.  sudo couldn't find npm so had to run as root for it to see i2c.

Configured my pH and ORP probes...

how do I get to this screen?

[image]

Also very curious about what flow sensors u guys are using?  or is everyone just using a VSF pump?  I only have a VS pump so not sure how to calculate flow and Atlas doesn't sell a 2" flow sensor, nor have I found another flow meter I can interface with.

rstrouse, 2021-02-22:
Click on the gears at the top right of dashPanel then click on the chemistry tab.  Click add controller and choose REM Chem

rstrouse, 2021-02-22:
I have a flow sensor attached to my flow cell.  It fits in the leftmost port and is attached to Gpio.  The other guys are using pressure sensors.  @gw8674 was talking that he was going to put an IntelliChlor sensor in from an old cell.

johnny2678, 2021-02-22:
ahh... didn't think about using an old salt flow cell.  Do you have a link for the flow sensor you plugged into the heyward flow cell?  That would only read flow on/off, right?  not flow rate?

And sorry, I'm super dense or the tequila is kicking in.  I found this screen where you configure probes - but where do you actually point it back to the REM ip:port?  probe connection/device dropdowns are showing empty for me even tho they are configured in REM

rstrouse, 2021-02-22:
This is the one that fits in your flow cell.
https://www.poolweb.com/products/flow-switch-for-globe-flow-cell?variant=34990386053258&msclkid=4dafac6919e617ebf80acc5a0b9c0475&utm_source=bing&utm_medium=cpc&utm_campaign=**LP%20-%20Shop%20-%20*Catch-All&utm_term=4583176783037341&utm_content=All%20Products

rstrouse, 2021-02-22:
You need to set up the REM connection in your config.json. Then it will show in the dropdown and return all the connected devices.

johnny2678, 2021-02-22:
wow, guys... just have to say you have really outdone yourselves.  I'm still digging through all the options but I was able to get this far.

[image]

Was wiring all this together myself using NodeRed:
[image]

To be honest, this was working for me, but knowing you guys were simultaneously working on bring REM into the fold... I mean seriously GET OUT OF MY HEAD!! ;)

rstrouse, 2021-02-22:
Yep fix your connection to the pH pump and click on the tank to add some Acid and it will start seeking the setpoint.

johnny2678, 2021-02-22:
So, if I wanted to switch over to REM acid dosing, how could I configure it to send an influx packet like this every time it doses?

context.set("amt",flow.get("pH_dosing_duration") * flow.get("pH_dosing_rate_oz_per_second"))

```
var newmsg = { payload:[{
    measurement: "PoolStats",
    fields: {
        acidDosed_oz:context.get("amt")
    },
    tags:{
        chemistry: msg.doseType,
        mode: flow.get("pool_mode")
    },
    timestamp: new Date()
          
}]}
```

the `chemistry` and `mode` tags I can hard code but is there a way to send the amount of acid dosed to MQTT?

want to switch over but I use grafana as my monitoring dashboard and I don't want to have to reconfigure if possible.

btw, if there is a better place for the conversation - (gitter, another issue, etc), please let me know.

rstrouse, 2021-02-22:
Yes there it is part of the njspc Influx bindings.  These are totally customizable and will write directly to your Influx DB.  You can add your own bindings and format and connect to the bindings or use the ones that are already defined.  You will find these in the `web/bindings/influxDB.json` file.  If you want to send this data to MQTT there are three binding options.  mqtt.json and mqttAlt.json publish data to your broker in a predefined format.  However, you can create your own and link it in the config.json to have any format of the data you would like.

There is also a very interesting number that you may like.  It is called dailyVolumeDosed (in mL).  This will show you the total volume over a rolling 24hour period that is dosed.  This beauty of this value is that it is a very good indicator of acid demand.  But if you want the straight amount dosed, this is in the volumeDosed value.

johnny2678, 2021-02-22:
any idea how to clear the `ORP Daily Limit Reached` warning?  Guess that means ORP won't dose?

[image]

I set max ORP time to 6 hours thinking that would clear it, but no dice
[image]

johnny2678, 2021-02-22:
sorry, more questions - you are totally welcome to not answer

how do I test ORP dosing?  My current ORP level is around 765, I have the SWG at 0%, and the ORP is set to 800 - looks like that is the max.  Is there a way to trigger the dosing logic?
[image]

rstrouse, 2021-02-22:
At this point there is no way to force a dose using a chlorinator.  In fact we are still working out the IntelliChlor control so that part may be a bit rocky.  Our focus was on tank based dosing originally so pH and Chlorine tanks and peristaltic pumps could be accommodated.  Over the next couple of weeks we will be working out some of the loopiness associated with overriding the Chlorinator beyond the management during pH dosing.  This function works pretty well to ensure the chlorinator is not engaged when you are dosing acid.  You can enable/disable this with pH Dose Priority.

This model is much more straightforward than calculating a pH or Chlorine dose and delivering it but it stills has some warts.  It might be best if you let the OCP Control the chlorinator for now and I would greatly appreciate your feedback in working this out on the Touch controllers.  I have an IntelliCenter and the chlorinator functions are similar but not exact.

It appears you managed to figure out how to send feeds for the chem sensors (ORP and pH) to njspc from REM as I see you have values.

johnny2678, 2021-02-22:
Sure, happy to help.  Just let me know what you need.

I do have pH Dose Priority turned on.  Conceptually, that's exactly how pentair doses acid from it's controllers (IntelliPH for sure and I assume Intellichem).  Kill power to the SWG, dose the acid, turn the SWG back on.  Will take a walk out to the pad next time there's an acid dose and see if the SWG Cell lights turn off and back on.

I'm still absorbing all of this and trying to make it work with my existing Grafana monitoring setup. Caught a quick issue while browsing MQTT   #269 

Here's a graph showing this morning's pH automation from my home grown (red) and then REM pH automation (purple) this afternoon.

So I'm slowly figuring it out.  Of course you used mL and my dosing measurements were in oz  😂

[image]

johnny2678, 2021-02-22:
also, for the flow sensor setup under Hardware - would it make sense to make the SWG flow cell an option?  Pentair SWGs report low flow as a status and otherwise report status as `OK`.  Could that be an option here?

[image]

rstrouse, 2021-02-22:
Unfortunately, the SWG reporting on RS485 is terrible.  The reporting is sometimes delayed for almost a minute as the SWG does its thing and it only reports when the cell has a request to turn on which is antithetical to our purpose.  However, there are a number of flow switches that you can hook up to REM as most of the ones in this list (including the Pentair) are dry contact switches.
https://www.amazon.com/s?k=hayward+flow+switch&crid=9P044Z0F0B19&sprefix=hayward+flow+%2Caps%2C246&ref=nb_sb_ss_ts-doa-p_1_13

Bear in mind that even when there is no sensor the system, REM Chem will not dose unless the body circuit is engaged and the startup time is expired.  The body that triggers this is the body you have selected.  The flow sensor makes sure there is actual flow and something like a breaker on the pump being flipped, a misplaced valve, or in the case of a flow cell, stale water in the cell reporting readings, won't trigger a dump of chemical into the pipe.

Looks like our fingers outgrew our gloves on some of the mqtt bindings.   I'll check the stock mqtt and the mqttAlt bindings for enlarged finger mistakes.

Yeah we originally started using oz but depending on the flow rate for the pump, the units got harder to follow 29.5mL = 1oz so there were a whole lot of fractional values floating around and by the time everything is calculated, rounding errors over time started to grow.  As well, pump ratings and calibration of such ratings were easier to understand because the quantities at any given moment could be very small.

johnny2678, 2021-02-22:
Ha, no worries on the Oz/mL - I'm doing the conversion in my influx queries.

I'm not concerned at all about the flow switch since my acid pump is powered off of the Pentair Powercenter, which only gets power when the pump circuit is on.  Yes, there are situations where the pump could be on and flow could still be low, but it's worked so far.

I'm working my way through the influx schema right now - which is much more comprehensive than what I was using.  I'll have some questions but I'll wait to post until I've had some more time to poke around.  

Tmr morning @ 8am ET should be fun - acid demand always builds up overnight so I'll get to see how the morning dosing goes.  Turned off my nodered acid dosing flow so acid dosing is completely under REM + njspc now,

rstrouse, 2021-02-22:
Sounds like you came a long way in a short period of time with this.  This has a whole lot of tweak-ability including how the pH and ORP sensor feeds.  I think the guys have theirs doing some sampling (this is a setting on the feeds for the sensors).  What did you figure out for the temp compensation on the probe.  Are you feeding the temperature back into the probe from EasyTouch?

tagyoureit, 2021-02-22:
FWIW, I hooked up a pressure inducer to my filter and that's feeding the flow sensor.  

We also haven't been able to do too much testing on the REM chlor settings because the water temp has been colder than what the chlorinator can produce at.  When it warms up I'll be able to test it out a bit more thoroughly.  

Glad you got on board!!

johnny2678, 2021-02-22:
> Are you feeding the temperature back into the probe from EasyTouch?

Ooo, Good point.  My homegrown script issues the RT,[tempC] command when getting a reading from the probe.  I haven't figured out how to read the ds18b20's that I have tapped into the plumbing with REM yet.  Suppose I could just use the pentair probe reading but whole # temp reading seems so imprecise.

> I hooked up a pressure inducer to my filter and that's feeding the flow sensor.

very cool, good idea. Let me know if you have time to post pics and a part #.  I have an analog sensor (dial w/ no wires) in glycerin solution  doing this right now.  I could probably swap it out.

johnny2678, 2021-02-22:
btw, quick post showing the performance of my pH dosing automation prior to switching to REM today...
[image]

Area 1 - pH being controlled by Pentair's dumb % dosing setting.  Did OK keeping pH level, but still some fluctuations. Plus I had to adjust the % output from time to time. Usually every week.

Area 2 - the circuit board burnt out on the pentair pH controller. The only way I had to adjust pH levels down was to walk out to the pool and pour acid in like a caveman .  pH got really out of whack

Area 3 - I put in my pH auto-dosing solution.  Smooth and level with no intervention 😉

ps. except for the downward spike where a bug in my nodered script dumped my entire acid tank into the pool 😂.  We're not swimming now and the bug was easy to squash so it returned to normal in a couple of days.  Was glad to see the guardrails you put around dosing in REM.  

HINT: hopefully you close all dosing relays when REM starts up.  That was my mistake as I restarted my nodered flow in the middle of dosing so it never received an off command.

rstrouse, 2021-02-23:
I bought an EZO-PRS but I don't have it hooked up yet.  @tagyoureit did a pressure transducer tied that he is reading in REM.  I also think he added a driver for ds components into REM but it evades me right now.

REM uses a latching mechanism that will trigger the relay off if no signal is received from njspc within 3 seconds.  So if a dose starts and njspc is stopped it will kill the relay on REM.  As well if the dose is going on and any of the equipment stops reporting it will shut down the dose.  It also does an integrity check every 10 seconds to ensure the scheduled demand has not been met.

Depending on the pool volume and the characteristics of your pool the mixing time will help to lessen any overshoot.  Each time a dose is calculated it calculates the demand based upon the body volume (I hope you set this up in the dashPanel under bodies).  This demand is used to determine how many seconds the pump should run to deliver the appropriate dose up to the maximum dose volume or time.  Then the mix time is used to allow the dose to change the pH level.  

If the pump is calibrated to the output, the body volume is correct, and the TA, CH, CYA, and Borate values are close it should give you a really accurate demand number and corresponding dose.  Hopefully a little bit better than that caveman you hired to dose for you.

tagyoureit, 2021-02-23:
I haven't programmed anything for the ds18b20.  I was 90% of the way done programming the dht22 when @rstrouse directed me in another direction.  But the ds looks really easy to add ([example](https://github.com/thisdavej/ds18b20-raspi/blob/master/index.js) with not other dependencies).  We would need to add an extra section for 1-wire devices as we don't have any programmed yet.

johnny2678, 2021-02-23:
yea, I've got the dosing w/ a 5 minute mixing time set up.  You guys really went for the platinum package with this feature.  All I was doing was dosing for a set time when the pH was reading below the threshold.  Calculating acid demand and dosing to that exact setpoint is... wow.

@tagyoureit let me know your thoughts on getting DS temp readings into REM.  I've got three hooked up in serial through one-wire on GPIO17.

I did notice my CSI calcs aren't matching yours.  Need to figure out how to send my readings for TA/CH/CYA/CSI over MQTT.  I've been tracking that through my homegrown script also

[image]

rstrouse, 2021-02-23:
> I did notice my CSI calcs aren't matching yours. Need to figure out how to send my readings for TA/CH/CYA/CSI over MQTT. I've been tracking that through my homegrown script also

If you pull next for njspc I fixed the mqtt anomalies you mentioned earlier and added the TA/CH/CYA values to the MQTT emit.  The CSI was already there names saturationIndex.

johnny2678, 2021-02-23:
will do @rstrouse 

can you set the chem values (TA/CH etc) via MQTT?

rstrouse, 2021-02-23:
> can you set the chem values (TA/CH etc) via MQTT?

Yes payload topic `root/state/chemController  {id:50, cyanuricAcid: [number], alkalinity: [number], calciumHardness:[number],cyanuricAcid: [number], borates: [number]}`  You can set one value or all of them just include the ones you want set in the payload and the others undefined.  You can set the other attributes for the chem controller the same way.

rstrouse, 2021-02-23:
Just checking.  Did you set up a feed from REM to njspc for the pH and ORP probes?
[image]

johnny2678, 2021-02-23:
> Yes payload topic `root/state/chemController`

ok, sorry - might need you to walk me through this.  If I publish using this path it creates a new topic and doesn't update the existing value.  dashPanel is currently reading 70 for `alkalinity` as this screenshot shows.

[image]

If I update the full path to alkalinity <root>/state/chemControllers/50/remchemcontrol/alkalinity, I can get MQTT to take the new value `68` but it isn't reflected in dashPanel.  Am I doing it wrong?

[image]

also, not being picky but any reason the mqtt chem is in the `key: value` format and the rest of the attributes are in the `key: {"key": value}` json format?

johnny2678, 2021-02-23:
> Just checking. Did you set up a feed from REM to njspc for the pH and ORP probes?

yes, though I didn't play with the Sampling and I didn't send to MQTT like in your screenshot - MQTT is still getting the values though

[image]

rstrouse, 2021-02-23:
Yeah that send to MQTT feed is me emitting to a different broker.  There is no need for that one.
Your payload for MQTT most likely does not have the proper format.
try `{"id":50, "alkalinity":67}`

johnny2678, 2021-02-23:
> Yeah that send to MQTT feed is me emitting to a different broker. There is no need for that one.

got it

> Your payload for MQTT most likely does not have the proper format.
> try `{"id":50, "alkalinity":67}`

That just creates a new topic at (1) instead of updating the already published topic at (2)

[image]

rstrouse, 2021-02-23:
The new topic is an incoming topic 1 and the 2 is the outgoing (emit).  Check in dashPanel for the change.  This may just wait to emit until it gets a new pH or ORP value.  It should have taken the change.

johnny2678, 2021-02-23:
dashPanel still showing the old value 70

[image]

I'll check again in the morning.  I update these chem values (TA/CH, etc) only once a week so no big deal if it doesn't update in dashPanel right away.

I'll let you go for the night - time to catch up on Wanda Vision ;)

rstrouse, 2021-02-23:
I just tried it locally and it is instant.  Are you sure you have the correct root topic.
[image]

johnny2678, 2021-02-23:
> Are you sure you have the correct root topic.

I think so.  I did prefix the default value in my config.json with `pool` 

gave me this structure in MQTT Explorer - `pool/easytouch2-4`:
[image]

so I sent the payload to `pool/easytouch2-4/state/chemController`

feels like i'm missing something obvious like a trailing slash or something

johnny2678, 2021-02-23:
I also renamed REM in dashPanel to `remchemcontrol`.

I think the default value was `rem chem 1` or something like that

rstrouse, 2021-02-23:
Yeah the name isn't important other than the original bindings included it in the topic.  The alt binding has omitted all this.

Check the njspc console to see if there are any errors when you publish the topic.  Everything looks right if the root topic is reflected.  Wait a minute /pool/easytouch2-4 is actually two topics.  I'll bet the processor doesn't know how to deal with that.

rstrouse, 2021-02-23:
Ok so pull next.  The MQTT subscription processor didn't like multi-level root topics like you have.  It should now accommodate that.

johnny2678, 2021-02-23:
yep, this worked! can set chem via MQTT with a multi-level root .

johnny2678, 2021-02-23:
Can I see what you have in your probe device feed for MQTT?  I'd like to publish to a different topic, separate from what njspc publishes...

I got this far but nothing has been created in MQTT.  Do I need to put some kind of js in the box to send the probe value?

[image]

rstrouse, 2021-02-23:
If you return nothing it will return the entire value by default so essentially what you see below is the same as not entering anything.  In your case you chose Probe pH as the send value and this will be sent as a number.  If you chose All Values it would emit the entire state for the probe which includes temperature and pH.
[image]

Make sure your MQTT connection is marked as active on the general tab.

johnny2678, 2021-02-23:
Interesting... not sure what my setup is missing

[image]

These values come in pretty frequently and I added the entry over an hour ago so I would have expected it to populate by now.

MQTT connection is marked active on the general tab

rstrouse, 2021-02-23:
Restart REM by pressing the Reload Server button on the general tab.  If you added a new MQTT server I don't think it initializes it right away.

johnny2678, 2021-02-23:
hmm, did the restart.  MQTT Values still aren't populating.

Also noticing that REM isn't triggering the Acid dosing like it was this morning

[image]

Changed REM logging to silly... I think this is the section where REM receives a new probe pH value, I can see where it does the /chemController emitting line, but there isn't a listing for the separate MQTT publish cmd

```
3|relayEquipmentManager  | verbose: Emitting: /chemController : {"acidDosing":"off","id":50}
3|relayEquipmentManager  | debug: Atlas_EZO-pH command R bytes written:1 result:7.917
3|relayEquipmentManager  | silly: Setting Dirty... true 6691
3|relayEquipmentManager  | debug: Atlas_EZO-pH command T,? bytes written:3 result:?T,21.11
3|relayEquipmentManager  | silly: Setting Dirty... true 6997
3|relayEquipmentManager  | verbose: Emitting: /chemController : {"pHLevel":7.917,"id":50}

```

Trying to troubleshoot the acid pump not dosing.  Was playing with the data smoothing values earlier, but set it back to 1 and it's been that way for a while.  Will keep digging

rstrouse, 2021-02-23:
There are 5 conditions where the acid will not dose.
1. The pH level is higher than the setpoint - Obvious
2. The body circuit is not running
3. The max limit per rolling 24 hours has been exceeded or falls within the minimum dose that can be applied in 5 seconds.
4. The pH tank is empty
5. There is a hardware error when communicating with the devices.

Check the display in dashPanel are there any errors listed on the device
[image]

johnny2678, 2021-02-23:
ok, acid is dosing again... 

[image]

I was playing with the pH feed data sampling.  Set it to 5 and then 2 thinking that would smooth out the values over readings.  Dosing didn't start working until I changed back to 1.

Still can't see the MQTT values getting published from the additional REM MQTT connection for some reason. Will keep digging.

rstrouse, 2021-02-23:
Double check the connection settings for your MQTT.  This has been flawless.  Perhaps restart REM altogether.  It is very odd that the sampling caused it to not dose as well.  njspc is totally unaware of the sampling settings.  It only responds to the information that it is given and looking at your pH rising this number came from njspc I presume.

Do you have anything else hitting that probe?  We have found that the EZO circuits don't like it when you fire requests from more than one source because of the timing that is required between initiating a read and actually reading the registers.

johnny2678, 2021-02-24:
> Do you have anything else hitting that probe? We have found that the EZO circuits don't like it when you fire requests from more than one source because of the timing that is required between initiating a read and actually reading the registers.

The yellow line is actually from my own influx script for now, not from njspc., but now that I think about it, your data smoothing function probably requires consecutive samples.  Every time there was in i2c lock from my script sampling probably restarted therefore never completing, which is why it started working again when I set it back to 1.

I am still running my old chem script and I'm working on a way to get off it.  That's where getting REM->MQTT comes in.  If I can grab the value from MQTT now that REM is reading it, then I don't need my own script anymore.

It's marked as active...

[image]

And the feed is marked as active...

[image]

Is there somewhere in the logs where it checks mqtt configuration?  When I save I see this scroll by, but looks like its just writing the config, not necessarily testing the connection...

```

3|relayEquipmentManager  | debug: Persisting Configuration data...
3|relayEquipmentManager  | verbose: 192.168.5.138 PUT /config/connection {"type":{"name":"mqttClient"},"isActive":true,"id":2,"name":"15620 MQTT","options":{"clientId":""},"protocol":"mqtt:","ipAddress":"192.168.5.140","port":1883,"userName":"john","password":"H1ll1234","sslKeyFile":"","sslCertFile":""}
3|relayEquipmentManager  | info: [6:53:29 PM] 192.168.5.138 PUT /config/connection {"type":{"name":"mqttClient"},"isActive":true,"id":2,"name":"15620 MQTT","options":{"clientId":""},"protocol":"mqtt:","ipAddress":"192.168.5.140","port":1883,"userName":"john","password":"H1ll1234","sslKeyFile":"","sslCertFile":""}
3|relayEquipmentManager  | {
3|relayEquipmentManager  |   type: { name: 'mqttClient' },
3|relayEquipmentManager  |   isActive: true,
3|relayEquipmentManager  |   id: 2,
3|relayEquipmentManager  |   name: '15620 MQTT',
3|relayEquipmentManager  |   options: { clientId: '' },
3|relayEquipmentManager  |   protocol: 'mqtt:',
3|relayEquipmentManager  |   ipAddress: '192.168.5.140',
3|relayEquipmentManager  |   port: 1883,
3|relayEquipmentManager  |   userName: 'me',
3|relayEquipmentManager  |   password: 'XXXXX',
3|relayEquipmentManager  |   sslKeyFile: '',
3|relayEquipmentManager  |   sslCertFile: ''
3|relayEquipmentManager  | }
3|relayEquipmentManager  | silly: Setting Dirty... true 952
3|relayEquipmentManager  | silly: Setting Dirty... true 953
3|relayEquipmentManager  | silly: Setting Dirty... true 953

```

After this I don't see any mention of mqtt in the silly logs.  If you get a chance, could you grab the log message your REM sends when an MQTT value is published?

rstrouse, 2021-02-24:
Ok so that pretty much tells me what is going on.  I don't have a setup where there is access applied to the mqtt server so it is likely failing with a permissions issue.  This will happen when REM starts up and you can see it in the console when you do a Reset Server.

However, I have done some research and figured out the username and password has to be passed to the connection constructor.  Pull a new REM and lets see if it connects.

johnny2678, 2021-02-24:
boom - that did it

```
 | silly: Sending on MQTT Channel {"eventName":"pool/chem/jhorp","value":734.1}
```

also rewrote my script so I rely exclusively on chem values read by REM now.  Not doing direct i2c readings anymore.

logs are looking pretty clean now!

[96 more replies]