tagyoureit/nodejs-poolController #270: Schedules running on wrong day?

johnny2678, 2021-02-23:
Not sure if this is a UI issue or a schedule issue.  Or just isolated to my environment.

3 days a week, I run my pool at a higher RPM for an hour in the morning to let the in-floor cleaning swivel returns do their thing.  I set the schedule for Mon, Weds, Fri.

However, it's running this morning (tues):
[image]

At first I thought I set the schedule wrong. The X's in the UI are a little confusing.  Is it, or is it not supposed to run on the day with the X in the circle?

But then I confirmed in settings that I set it for Mon/Weds/Fri:
[image]

Any ideas why it's running today? Tues?  This isn't isolated to today... it's been doing this for a while.  I won't run tomorrow (Weds), but will run on Thurs, etc.  No harm done as I'm still getting every other day.  

Not sure what log info I can provide after the fact.  Do you want me to run a packet capture tomorrow (weds) morning when the schedule is supposed to run so you can see what's happening?  and again Thurs morn when it does run, but isn't scheduled to? 

Low priority, just bringing it up because it's been a head scratcher for me.

johnny2678, 2021-02-27:
Another interesting observation @rstrouse @tagyoureit ... this morning (Sat) the schedule for Pool High ran, which is only supposed to run on Mod,Weds,Fri

[image]

Pool High operates at 2760 rpm so you can see that the pump is actually running the schedule, it's not just in software.

But, the OCP reports that the schedule isn't running.  Even though the corresponding Pool High feature is on.

[image]

I missed the logs/packet capture this morning when the schedule started.  Will try and get them for next time.

rstrouse, 2021-02-27:
EasyTouch does not report any indicator that a schedule happens to be in effect.  The code we were using synthesizes this effect by looking at the current time and the circuit to determine whether the schedule should be running.  However, it was not looking at the day of the week correctly.  If you pull this should be fixed in next.

Please note that there is no code that engages the circuit from njspc related to the schedule.  If the Pool High circuit is on, I can find no code where njspc turned it on.

johnny2678, 2021-02-27:
interesting.... I'll pull `next` and keep observing.  Not a big deal.  Instead of the schedule running Mon/Weds/Fri, it runs Tues/Thurs/Sat.

rstrouse, 2021-02-27:
I don't have an EasyTouch but it is very odd that it registers on your ScreenLogic as Mon/Wed/Fri and actually runs on Tues/Thurs/Sat.  I seem to remember when I had IntelliTouch I had to actually tell it what the day of the week was.  Are you sure that is set correctly on the EasyTouch panel.

johnny2678, 2021-02-27:
omg... of course it's the easiest explanation.

I didn't even think my pool could be set incorrectly.  Thought my phone app had me covered with time sync:

[image]

and then I thought i was double covered with njspc:
[image]

didn't even enter my mind that the ocp could be off by a day, but sure enough, my pool thinks it's fri 🙄

[image]

rstrouse, 2021-02-27:
Look at that you get an extra day.  With that kind of logic you should be able to schedule for the second Tuesday of the the week.

closed 2021-02-28 by johnny2678

tagyoureit, 2021-03-01:
By the way, there's a setting for that.

[image]

If you choose "server" then njsPC will check your *Touch system time (every 5 mins) and correct it when it is off by >5 mins.

johnny2678, 2021-03-01:
> BTW, how did you get it to say "Fri Feb 27". Where I live, the 27th was Saturday.??

I know, right???

My guess is it’s been like that since Feb 29th of 2020 and the extra leap day threw it off.

All of my schedules are ‘every day’ schedules except that one so it took me a while to notice.