tagyoureit/nodejs-poolController #271: Solar heat status shows as off (again!)

antamy, 2021-02-28:
Hi guys, I'm back!   After this working for a while, it has now stopped.  I was having trouble with adding a mode to the pump via dash panel (I added a new rpm setting, then every time I changed it, it would add it again).  I will look at that again and see if I can replicate it, but for now I deleted the pump and set it up again.  

Everything in the ICP is working fine.  Before this log, I loaded the ICP from the OCP and turned the pool heat off.  Then I started the log.

- everything reloads, you see the heat status is off/off.  Pump rpm is 1,400rpm / 235 w.
- on the ICP, I set heat to solar only.
- dash panel soon displays heat status solar only, but solar is off.  Pump rpm is 2,450rpm / 894 w (which is correct)
- ICP shows the pulsing sun, so the solar is working.  The valve outside is also in the correct position

Sorry for the bad news!

Anthony

antamy, 2021-03-01:
There seems to be more to this than just the solar status not displaying correctly.   It seems to stem from changing pump parameters.  The other day, I was using njspc to add and change custom speeds for my VS pump.  This resulted in the last custom speed on the list being repeatedly added, even when I would delete (in the dashPanel UI) each duplicated and save the pump.  There is no way I can see on the ICP to delete these, and I ended up with several pages of custom speeds.  This also coincided with the solar heat status not displaying.

I deleted the pump on ICP and added it again.  njspc started working again, without the solar heat status displaying.  On the ICP, I have 4 custom speeds :  Pool, Spa, Solar and Air Blower.  On dashPanel, I see 4 custom speeds:  Pool, Spa, blank and Air Blower.  If I change the "blank" on dashPanel to be "Solar" and save that pump, the speeds no longer work.  When Solar is on, the pump does not change to the Solar speed (although it did before changing it in dashPanel).

I reinstalled njspc (next branch) and dashPanel (master).  Reset my pump.  So I'm back up and running, but still dashPanel does not display what is on ICP:

[image]

In addition, now the heat mode and heat status is both off!  Even though solar is running:

[image]

And, if I click on heat mode, the config pop up correctly displays the current setting:

[image]

There is something screwy going on!

Anthony

rstrouse, 2021-03-01:
We'll have a look.  Solar is what is known as a virtual circuit.  There is no relay and it is only triggered by other things.  Interestingly on most panels we see this value as 128 on yours we see it as 158.  Either we don't have this value correct, or there is some other twist on your panel.  I'll also have a look to make sure the Touch code clears out the removed circuits.

rstrouse, 2021-03-02:
Grab njspc from next.  This should ferret out the last of the heat status issues.  However, do not edit your pumps in dashPanel.  We need to figure out why your solar circuit is being identified as 158 not 128 as we have seen in the past.

antamy, 2021-03-02:
I grabbed next, and still see Heat Mode = Off in the main display and Solar Only (correct) in the Pool Heat Settings pop up.  And Heater Status = Off when the sun is pulsing on the ICP.

rstrouse, 2021-03-02:
Well this is a mystery.  Fortunately, I have called those darn kids and they brought the mystery van with them.  We need to take a different approach and focus on the actions taking place on the ICP.  The configuration is loading just fine short of the solar virtual circuit settings but there are some differences related to them which we should be able to figure out after all the steps have been completed.

First lets make sure you are running the same version of the software I am.
**Step 1:**
Shut down njspc

**Step 2:**
Navigate to the nodejs-PoolController directory and issue the following commands
`sudo git checkout next`
`sudo git pull`
`sudo npm start`

Now we are going to focus on just the pool body.  I need you to do all of this in order so I will outline the steps that I want you to perform below.

**Step 1:**
Turn all circuits on the pool off and set all heat modes to off for the pool and the spa.

**Step 2:**
Go to the hamburger menu in dashPanel and select the logging tab then press the Capture Replay button.  However, I want you to uncheck the Capture Configuration Reload checkbox before pressing Begin Capture.

**Step 3:**
From your ICP please turn on the pool and let the filter pump come up to speed.

**Step 4:**
From your ICP set the setpoint for the pool to something higher than the current body temperature then select Heater.  Give the pool some time for the heater to engage and come on line.  Give it several seconds after the flame flickers on the ICP.

**Step 5:**
From your ICP change the heat mode for the pool to Solar Only and let the pool sunshine pulsate on the ICP.

**Step 6:**
From your ICP turn off the pool circuit.

**_The remainder of these steps are designed to capture the circuit identifiers for your virtual circuits.  This will tell us if there is an offset for your firmware version._**

**Step 7:**
From your ICP custom speeds for your filter pump. select the last slot on the first page and change the identified circuit to each the following while saving the pump after each change.
1. Solar
2. Heater
3. Freeze
If you have options for POOL HEATER, SPA HEATER, POOL SOLAR, SPA SOLAR, HEAT BOOST, PUMP +, and/or PUMP - choose these in the order you see them here and let me know whether they existed or not and if there was any variation on the name.  The order is important as I will not see the name and only the corresponding numeric value will appear in the logs.

**Step 8:**
Return to dashPanel and select the Cancel Capture button on the logging tab.  This should download a replay file to your browser.  Please upload that file here.

I appreciate your patience with this because we don't have every piece of equipment or firmware that is produced,  while there are minor changes between them, the virtual circuits are one area where they tend to have some variation.

antamy, 2021-03-02:
ok, I'm on it!  I have never got git figured out, so I just wget the next.zip file from github, unzip it and go from there.  I have been running npm install on that, before running npm start.

antamy, 2021-03-02:
Here is the capture.  For step 7, I chose Solar > Heater > Freeze > Pool Heater > Spa Heater.   The other options were not present.
[replay(2).zip](https://github.com/tagyoureit/nodejs-poolController/files/6071037/replay.2.zip)

rstrouse, 2021-03-02:
Did you uncheck the Reload Configuration checkbox prior to starting the capture?

rstrouse, 2021-03-02:
So there were no 155 messages in that file.  It turns out that you may need to exit the assign custom speeds screen after each change to the circuit name.  So can you please capture that again while pressing the save button then back between each change and don't forget to uncheck to Reload Configuration checkbox.  That adds an enormous amount of messages that we don't need for this.

antamy, 2021-03-02:
I reinstalled (with git) and re-ran.  My UI is behaving now.  But I am not sure if I really am working on the next branch (since I didn't see anything being downloaded after the initial download when I did the git clone).  I removed njspc completely, then I ran

```
pi@poolcontroller:~ $ git clone https://github.com/tagyoureit/nodejs-poolController
Cloning into 'nodejs-poolController'...
remote: Enumerating objects: 739, done.
remote: Counting objects: 100% (739/739), done.
remote: Compressing objects: 100% (357/357), done.
remote: Total 11305 (delta 444), reused 626 (delta 367), pack-reused 10566
Receiving objects: 100% (11305/11305), 6.24 MiB | 4.52 MiB/s, done.
Resolving deltas: 100% (7660/7660), done.
pi@poolcontroller:~ $ cd nodejs-poolController
pi@poolcontroller:~/nodejs-poolController $ git checkout next
Branch 'next' set up to track remote branch 'next' from 'origin'.
Switched to a new branch 'next'
pi@poolcontroller:~/nodejs-poolController $ git pull
Already up to date.
pi@poolcontroller:~/nodejs-poolController $ npm start
```
npm start failed, so I had to run npm install.  I then edited defaultConfig.json and ran npm start, which worked.  I ctrl-c'd that and ran:

```
pi@poolcontroller:~/nodejs-poolController $ git checkout next
M       defaultConfig.json
Already on 'next'
Your branch is up to date with 'origin/next'.
pi@poolcontroller:~/nodejs-poolController $ git pull
Already up to date.
pi@poolcontroller:~/nodejs-poolController $ sudo systemctl start poolController
```

The capture is a complete redo of what you asked.  **I don't know why unzipping next.zip and using the git commands give different results (see below)**, but I am assuming that my git setup is correct now.  Is there anyway to check this?  

Pump On
Heater On:

[image]

Switch to Solar:

[image]

[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6071332/replay.zip)

rstrouse, 2021-03-02:
I have no idea why unzipping next and running the git commands would be different other than it may not process the package.json when doing that.  The message stating you are up do date is the indicator that there are no differences with the repo.  If you save the config.json file from the root you will no longer need to touch the defaultConfig.json file.  That way if you recreate the directory you can then simply copy the config.json file back in and it will take your settings.

The replay you provided does now contain the information I needed regarding virtual circuit identifiers.
Just for reference on your OCP the values are as follows.
Solar = 158
Pool or Spa Heater = 157
Freeze = 159
Pool Heater = 155
Spa Heater = 156

I need to determine where there are inconsistencies with other IntelliTouch OCPs.  Which firmware are you using?  You should be able to find this in the ICP by pressing buttons 2 & 4 on the Advanced screen.  It will show you a menu with System Version Info at the top.  If you can take a picture of that screen it would be great.
[image]

antamy, 2021-03-02:
Ok, now I've figured out git, I'll just use that. Firmware :

[image]<https://user-images.githubusercontent.com/47839015/109710253-4c259c00-7b52-11eb-8862-6826e4cdf174.png>

rstrouse, 2021-03-02:
Did you forget the firmware image?

antamy, 2021-03-03:
I guess emailing a response with a photo attached didn't work!  Firmware is 1.170, inside and out.

rstrouse, 2021-03-03:
Any idea when that was released.

antamy, 2021-03-03:
> Any idea when that was released.

No idea.  This house was built 6 years ago and the pool was installed at that time.  Is it old?  I haven't looked into upgrading the firmware.

tagyoureit, 2021-03-03:
Intellitouch 1.190 was released 6/12/19.  https://www.pentair.com/en-us/education-support/residential/product-support/pentair-pool-and-spa-software-downloads.html.  Couldn't find release dates for earlier fw versions.

antamy, 2021-03-15:
I am going to close this.  The display is working fine.  I do have an odd issue no that when my pool starts up based on the schedule, the heating gets turned off, which is bizarre.  Since I'm not at the property to play around with the ICP, I'm going to ignore that until I am there again.

closed 2021-03-15 by antamy