tagyoureit/nodejs-poolController #272: Changing IntelliFlow pump RPM results in outbound message error.

brentratliff, 2021-03-01:
Changing IntelliFlow pump RPM results in outbound message error.

Changing the pump speed results in:

Message aborted after 3 attempt(s): 165,1,16,33,155,47,1,128,0,0,0,6,10,1,11,0,0,0,0,0,0,0,0,0,0,0,0,3,190,134,0,0,0,0,0,0,232,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,4,109

Stack Trace:

ApiError: Message aborted after 3 attempt(s): 165,1,16,33,155,47,1,128,0,0,0,6,10,1,11,0,0,0,0,0,0,0,0,0,0,0,0,3,190,134,0,0,0,0,0,0,232,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,4,109
    at SendRecieveBuffer.writeMessage (/home/pi/nodejs-poolController/controller/comms/Comms.ts:421:27)
    at SendRecieveBuffer.processWaitPacket (/home/pi/nodejs-poolController/controller/comms/Comms.ts:371:29)
    at SendRecieveBuffer.processOutbound (/home/pi/nodejs-poolController/controller/comms/Comms.ts:379:30)
    at SendRecieveBuffer.processPackets (/home/pi/nodejs-poolController/controller/comms/Comms.ts:362:21)
    at Timeout. (/home/pi/nodejs-poolController/controller/comms/Comms.ts:393:59)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)

**Pool Equipment**
 - EasyTouch2 8
 - Controller Model: EasyTouch2 8P
 - Controller Firmware Version: ?
 - Pump(s) manufacturer and model: IntelliFlow
 - Chlorinator: None
 - Heater(s): Pentair 400 natural gas
 - Chemical controller: None
 - Valves: Intellivalve x 2
 - 
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6062652/replay.zip)

 - OS: Windows 10
 - Chrome
 - Version: latest

I can still change the pump speed in ScreenLogic. It appears to be changing the pump speed via PoolController as well. I changed it with PC, then verified it via SL. Yesterday, however, it was reporting the incorrect speed and wattage. Today it is reporting the same as SL.

rstrouse, 2021-03-02:
Are you changing the rpm via dashPanel, webClient, or through one of the interfaces.  Also, are one of the circuits a Solar circuit on the pump?  I know strange question.

brentratliff, 2021-03-02:
Via dashPanel. I'm not sure which circuit the pump was installed on. I did not program it. It looks like it may be Aux 6. I don't know how to tell if it is a solar circuit. I'm having an IntelliChem installed in a couple weeks and I'll ask the installer.

brentratliff, 2021-03-04:
Now my circuits will randomly disappear. I delted the data directory and config.js and reran. Everything worked for a bit and then I start getting:

[3/4/2021, 6:38:55 PM] error: Cannot read property 'circuits' of undefined

And I lose my pump, heater, and IntelliChem.

brentratliff, 2021-03-05:
Looks like I'm getting the same errors in SL. I think the wireless protocol adapter is failing. I'm having it replaced tomorrow.

brentratliff, 2021-03-05:
So after further troubleshooting, it appears that when PC is running, ScreenLogic no longer works for turning on circuits. As soon as I power down the Pi, ScreenLogic starts working again. The Pi is tied into the bus at the protocol adapter.

tagyoureit, 2021-03-05:
There is something off with that packet. Will try to look at it soon but I’ve been completely slammed recently.

brentratliff, 2021-03-05:
Thanks, no hurry, SL is working for now and I can control it through HomeKit. I just like the UI and architecture of PC better and will eventually broadcast MQTT to HomeSeer with it. Great project.

brentratliff, 2021-03-12:
More info, the Intellichem packets seem to be crossed. When I changed the alkalinity, the tank levels of both decrease. When I try to change the level of the tanks,  I get a stack trace error. I'm on 6.5.2 of Next.

rstrouse, 2021-03-14:
I don't have IntelliChem so it would be good to get the stack trace error.  It sounds like the alkalinity and tank levels are not correct for EasyTouch.  These attributes work correctly on IntelliCenter so I need some more information.  It would be good if you could capture changing these values from EasyTouch with a replay -- without the Capture Configuration Reload checkbox checked.  I only want to see the messages that are sent when you change the data from EasyTouch.

brentratliff, 2021-03-14:
I don't control the IntelliChem through EasyTouch, it has it's own UI, but it's tied into the RS485 bus at the EasyTouch panel. There may be a way for EasyTouch to communicate with it though. I will look into it. I can control it at its UI or through ScreenLogic.

brentratliff, 2021-03-14:
Ok, I can control the tank levels and PH, ORP setpoints from the EasyTouch. The Saturation Index is available on EasyTouch as well. What's interesting is that running PC disables the ability for ScreenLogic to control certain aspects of the EasyTouch such as Intellibrite and external relays (outdoor lights). As soon as I power it down, ScreenLogic works again. This was happening before I installed the IntelliChem.

brentratliff, 2021-03-14:
I uninstalled 6.5.2 and rebuilt with a git pull next. PC now seems to play nice with ScreenLogic, but the IntelliChem issue still exists. I will do a packet capture controlling through EasyTouch.

brentratliff, 2021-03-15:
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6145123/replay.zip)

This is a capture of changing the IntelliChem tank levels and Alkalinity through the Easytouch settings menu.

brentratliff, 2021-03-16:
On a reboot of the Pi, everything reads correctly in PC including IntelliChem. I can toggle lights, pumps, cleaner, blower. As soon as I try to change pump speed or IntelliChem settings, I get the Stack Trace errors and ScreenLogic stops working.

rstrouse, 2021-03-17:
Thanks for the output.  I need to understand what the settings were that you were setting to:
So were you setting the following:
Cyanuric Acid = 44
Total Alkalinity = 20
Calcium Hardness = 250

Or did you set the TA to 160

brentratliff, 2021-03-18:
I set the TA to 160 and changed the levels in both tanks on that capture. I can also capture changing pump RPM which throws an error and locks out SL.

rstrouse, 2021-03-18:
I can't seem to find the spot where you changed the tanks.   The seem to be a solid 6 all the way through from start to finish.  Did you do it before or after the TA from 120 to 160?

Also was CYA = 44 and CH = 250 or do we have these values incorrect as well.  Also is there a setting that you have set to 20 for anything.  Perhaps max dosage volume or time?

brentratliff, 2021-03-18:
I don't think I have anything set to 20. I will go check. I am home all day and am happy to do as many captures as you need. I can't remember if I changed the tanks before or after. The CYA and CH values are correct.

rstrouse, 2021-03-18:
Just to be sure the only changes you can make to IntelliChem with EasyTouch are
1. pH Setpoint
2. ORP Setpoint
3. pH Tank Level
4. ORP Tank Level
5. CYA
6. CH
7. TA

Is there anything else for your configuration that you can change on EasyTouch.  Can you set dose priority or any of the dosage limits from EasyTouch?

rstrouse, 2021-03-18:
Do you by chance have your pool volume set to 16kgal on the IntelliChem?

brentratliff, 2021-03-18:
That is exactly correct. I cannot change dosing information or any of the other SI values from the EasyTouch. I can also change the pump speed on the IntelliFlow pump that was causing the error also.

brentratliff, 2021-03-18:
Yes, it is set to 16K gallons

brentratliff, 2021-03-18:
I noticed that PC is also always display dosing information as 0ml. I have it set to Oz in IntelliChem.

rstrouse, 2021-03-18:
@tagyoureit is planning on looking at the pump speed change issue for you.  I am reviewing all the messages for IntelliChem.  With the exception of the value that is 20 (this could be a flag), I think I have found our misalignment.

I don't see any incoming data for the dosing information from IntelliChem.  That is unfortunate because I imagine it would be great to get dose amounts.  Perhaps this data does exist and we just don't know it because the outputs we do get might not be dosing at the moment it was captured.  I don't have IntelliChem but if you are game perhaps we can find this information.

brentratliff, 2021-03-18:
Thanks for all the help. I am game to try to find the info. I know it's in IntelliCenter because it's monitored to prevent over dosing and I can change it. I am not sure it is broadcasted however.

rstrouse, 2021-03-18:
Ok so here are the steps.  First stop njspc then restart it.  This will clear prior log data and start with a clean slate for this session.

1. Open Messsage Manager by navigating to http://[your dashPanel IP]/messageManager.html
2. Open the dialog under hamburger menu and go to the logging tab.
3. Select all the options as you see here.  We are setting this up so that only the IntelliChem information is captured.  After you have made the selections click outside of the dialog to dismiss it.
[image]
4. Click on the icon identified by the arrow.  It should turn hot pink and messages to and from IntelliChem should start streaming into the messages list.
[image]
5. Go back to the log tab under the hamburger menu and click the Log to [File] button under the Packets section.  This will start writing what you are seeing in the browser to a file on njspc.
6. Change each value on EasyTouch and keep track of what you changed in the sequence you changed it.  Make sure you include changes to the max 24 hour dose setting from IntelliChem.  Let me know what you changed it to.
7. Make IntelliChem administer an acid dose.  Let me know what the dose amount should be.  Let's hope it lets us know what it is dosing and how much time is left.

brentratliff, 2021-03-19:
Great, I will do this tomorrow.

rstrouse, 2021-03-19:
One more thing.  Don't forget to uncheck the log to file.  You don't wan the logs to fill up your pi.  The resultant log file will be located in the /logs directory with a file name containing the date of when njspc was restarted.

brentratliff, 2021-03-20:
I tried this and it didn't log to file. There were a ton of IntelliChem packets going out on the console.

brentratliff, 2021-03-20:
I checked file before and unchecked after as instructed.

rstrouse, 2021-03-20:
Did you look in the nodejs-PoolController /logs directory on the pi?  It will not be in the dashPanel directory.

brentratliff, 2021-03-20:
Yes, the log file is there, but there are no IntelliChem logs.

brentratliff, 2021-03-20:
I figured it out. I set the app log to file, not the packet log to file. I'll redo it.

rstrouse, 2021-03-20:
That is the console log.  We are looking for the file named packetLog(2021-03-20_xx-xx-xx).log.  I figured you might have clicked the wrong log to file.

brentratliff, 2021-03-20:
From EasyTouch
PH -> 7.4
ORP -> 700
PH level 3->4
ORP level 4->3
CH ->250
TA->160
CYA->44

Accidentally turned pump off which reboot IntelliChem

Auto ORP Dose of 49oz or 1/11th of that

PH 11->10 doses per day 
PH 49->48oz
PH mixing time 25->24 minutes
ORP mixing time 19->20 minutes
ORP doses 11->10
ORP 49->48oz

PH manual dose of 3-5 seconds

[packetLog(2021-03-20_11-16-07).log](https://github.com/tagyoureit/nodejs-poolController/files/6176178/packetLog.2021-03-20_11-16-07.log)

.

brentratliff, 2021-03-20:
Quick look by me, I don't see any dosing information in the packets.

rstrouse, 2021-03-20:
Thanks for that.  I will take a look at this over the weekend.  Yeah that dosing information will be on that action 18 message.  If you want to see only the messages that have changed you can click on the not equal icon on the list.  When you click on the message it will show you which bytes changed from the last message sent.  The action 146 message is you changing something on EasyTouch.  The action 210 is just EasyTouch asking IntelliChem to cough up its status on the action 18 message.

If you are interested the first 4 bytes are the pH and ORP.  For pH you take `(byte(4)*256 + byte(5))/100` to give you `2*256+228/100 = 7.4` and ORP is simply `2*256+188 = 700`.  So the known values of 18 are (the ???? are our suspicions):
```
        //      0-1 : pH Level
        //      2-3 : ORP Level
        //      4-5 : pH Setpoint
        //      6-7 : ORP Setpoint
        //      8 : Unknown = 0
        //      9 : pH Dose time remaining (min) = 0x60??????
        //      10 : pH Dose time remaining (secs) = ?????
        //      11-12 : Pool volume 16x1000 + 0x100 = 16,000 gallons.
        //      13 : ORP Dose time remaining (min) = ??????
        //      14 : Unknown = 0
        //      15 : ORP Dose time remaining (secs) = ?????
        //      16 : Unknown = 0
        //      17 : pH Dose volume (unknown units) = ?????
        //      18 : Unknown = 0
        //      19 : ORP Dose volume (unknown units) = ?????
        //      20 : pH tank level 1-7 = 6
        //      21 : ORP tank level 1-7 = 6
        //      22 : LSI = 3 & 0x80 === 0x80 ? 256 - 3 / -100 : 3/100 = .03
        //      23-24 : Calcium Hardness
        //      25 : Unknown = 0
        //      26 : CYA
        //      27-28 : Alkalinity
        //      29 : Unknown = 20 
        //      30 : Water flow alarm = 0 = ok, 1 = no flow.
        //      31 : Water Temperature
        //      32 : Alarms = 8 = (no alarm)
        //      33 : pH Lockout, Daily Limit Reached, Invalid Setup, Chlorinator Comm error
        //      34 : Dosing Status
        //      35 : Delays = 0
        //      36-37 : Firmware = 80,1 = 1.080
        //      38 : Water Chemistry Warning
        //      39 : Unknown
        //      40 : Unknown

```

brentratliff, 2021-03-21:
The only thing I can find that I have set to 20 is the mixing time for PH

brentratliff, 2021-03-21:
I will double check the mixing times and post them

tagyoureit, 2021-03-22:
Just want to double check... all the problems with setting the pump circuits speeds have disappeared with your new protocol adapter?

brentratliff, 2021-03-22:
I didn't end up replacing the protocol adapter. I did replace the wireless connection kit that sends the raw RS485 data from the EasyTouch to the protocol adapter. It didn't really start working correctly (with exception of IntelliChem) until I did a complete uninstall and reinstall of njspc. It seems to be working correctly now. I am still trying to figure out how sending bad packets from njspc blocks ScreenLogic from controlling some circuits.

rstrouse, 2021-03-23:
So are the pump functions working?  Also do you remember how much chlorine dosed during the period you captured these messages.  The messages seem to indicate 240 in whatever units.  This appears to be mL which would = 8.11oz of chlorine is that correct?  Your pH dose was for 7 seconds and it dosed what appears to be about 1/2 oz if we have the units correct.  This pans out to a pump that puts out around 128mL/min.

brentratliff, 2021-03-23:
Yes, the pump functions are working since a clean install of 6.5.2 next. Those numbers make sense to me although they are not recorded in ScreeLogic or visible on IntelliChem.

brentratliff, 2021-03-23:
I have the dosing set to 10/48. I’m not sure if that’s 10 doses a day for a total of 48 in 24 hours.

brentratliff, 2021-03-23:
48oz

rstrouse, 2021-03-23:
Ok pull njspc next and give it a go.  I think I have decoded the messages as we know them.  There are some bytes we know nothing about but they are probably for CO2 tank dosing (they are always 0).  Unfortunately, IntelliChem does not appear to report the dose target.  It only counts up until it dosing status changes.  Kinda sucks for determining what is going on.  It's really too bad that it doesn't report the target or you would know how much longer the dose is going to take and how much more it is going to put in.

This is a major difference with REM Chem where the dosing targets are determined at the start of the dose then re-evaluated as the dose is taking place.  It would require a pi and some Atlas gear but the flow cell, tanks, and pumps you have can be used.

brentratliff, 2021-03-23:
Thanks for the update. I wonder if it didn't report the target since I dosed it manually with start and stop. The first dosing recording was an automated one with a preset target. I think IntelliChem does re-evaluate and change as it gets close. It also reduces the dosing to 25% while in spa mode.

brentratliff, 2021-03-23:
SI changes all work now and don't affect the tank levels. Still getting errors when setting tank levels via dashPanel.

**ORP**
TypeError: Cannot read property 'setpoint' of undefined
    at ChemControllerCommands.setIntelliChemStateAsync (/home/pi/nodejs-poolController/controller/boards/SystemBoard.ts:3414:45)
    at ChemControllerCommands.setChemControllerStateAsync (/home/pi/nodejs-poolController/controller/boards/SystemBoard.ts:3508:110)
    at /home/pi/nodejs-poolController/web/services/state/State.ts:31:61
    at Layer.handle [as handle_request] (/home/pi/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at next (/home/pi/nodejs-poolController/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/home/pi/nodejs-poolController/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/home/pi/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at /home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:335:12)
    at next (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:275:10)
    at /home/pi/nodejs-poolController/web/Server.ts:336:21
    at Layer.handle [as handle_request] (/home/pi/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:317:13)
    at /home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:284:7
    at Function.process_params (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:335:12)
    at next (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:275:10)

**PH**

TypeError: Cannot read property 'setpoint' of undefined
    at ChemControllerCommands.setIntelliChemStateAsync (/home/pi/nodejs-poolController/controller/boards/SystemBoard.ts:3415:47)
    at ChemControllerCommands.setChemControllerStateAsync (/home/pi/nodejs-poolController/controller/boards/SystemBoard.ts:3508:110)
    at /home/pi/nodejs-poolController/web/services/state/State.ts:31:61
    at Layer.handle [as handle_request] (/home/pi/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at next (/home/pi/nodejs-poolController/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/home/pi/nodejs-poolController/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/home/pi/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at /home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:335:12)
    at next (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:275:10)
    at /home/pi/nodejs-poolController/web/Server.ts:336:21
    at Layer.handle [as handle_request] (/home/pi/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:317:13)
    at /home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:284:7
    at Function.process_params (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:335:12)
    at next (/home/pi/nodejs-poolController/node_modules/express/lib/router/index.js:275:10)

brentratliff, 2021-03-23:
Water balance seems to reverted. Usually reports around 0. Showing 258.4.

rstrouse, 2021-03-23:
Please pull **dashPanel** and **njspc**.  Don't forget to **hard** **refresh** your browser. Use CTRL + F5 or equivelant for dashPanel.  I needed an extra set of parens on the SI math so that calc should be fixed.  

Unfortunately, there is no place in the messages coming out of IntelliChem where the current dose amount or time is emitted during the dose.  I did figure out that 20 is actually the salt level reported by your controller.  It is 20 x 50 = 1,000ppm.  In your case you are not using a chlorinator so this is probably the default level.
```
        //      Bytes - Descrption
        //      0-1 : pH byte(0) x 256 + byte(1) / 100
        //      2-3 : ORP byte(2) x 256 + byte(3)
        //      4-5 : pH Setpoint : byte(4) x 256 + byte(5) / 100
        //      6-7 : ORP Setpoint : byte(6) x 256 + byte(7)
        //      8 : Unknown = 0
        //      9 : Unknown = 0
        //      10 : Unknown = 0
        //      11-12 : pH Dose time seconds. The number of seconds since the dose started. byte(11) x 256 + byte(12)
        //      13 : Unknown = 0
        //      14-15 : ORP Dose time seconds.  The number of seconds since the dose started. byte(14) x 256 + byte(15)
        //      16-17 : pH Dose volume (unknown units) - These appear to be mL.
        //      18-19 : ORP Dose volume (unknown units) - These appear to be mL
        //      20 : pH tank level 1-7
        //      21 : ORP tank level 1-7
        //      22 : LSI. (byte(22) & 0x80) === 0x80 ? (256 - byte(22) / -100 : byte(22)/100
        //      23-24 : Calcium Hardness = byte(23) x 256 + byte(24) = 250
        //      25 : Unknown = 0 (probably reserved CYA byte so the controller is always dealing with integers)
        //      26 : CYA Max value = 210.
        //      27-28 : Alkalinity = byte(27) x 256 + byte(28)
        //      29 : Salt level = byte(29) x 50
        //      30 : Unknown = 0
        //      31 : Temperature
        //      32 : Alarms
        //      33 : Warnings pH Lockout, Daily Limit Reached, Invalid Setup, Chlorinator Comm error
        //      34 : Dosing Status (pH Monitoring, ORP Mixing)
        //      35 : Delays
        //      36-37 : Firmware = 80,1 = 1.080
        //      38 : Water Chemistry Warning (Corrosion...)
        //      39 : Unknown = 0
        //      40 : Unknown = 0
```

It looks like the ph and orp objects are not being sent from dashPanel which is why I am suspecting the dashPanel version.

brentratliff, 2021-03-23:
I was already up to date on dashPanel. Now, water balance is correct and changing levels no longer causes an error. It does however, keep resetting the level of both PH and ORP on the save of either.

brentratliff, 2021-03-23:
It looks like it saves the new value and then decreases both tanks by on a second later.

rstrouse, 2021-03-23:
> I was already up to date on dashPanel. 

Did you perform a hard refresh of the browser on dashPanel?

> It does however, keep resetting the level of both PH and ORP on the save of either.

Do you mean that when you change the pH or ORP setpoint the levels change to the setpoint?

> It looks like it saves the new value and then decreases both tanks by on a second later.

I think we got some bum advice on the tank levels.  The code looks like it is shifting the values by one so that it is never really 0.  When you captured the log for me the reading that came back was:
ORP Tank Level = 4
pH Tank Level = 6

At the time of the capture was the tank completely full? 100%

brentratliff, 2021-03-23:
All the setpoints are working perfectly. I did do a hard reset of the browser. The tank levels are what are strange. When I increase the level in either tank, it saves correctly and then decreases both tanks by one. They were not both full in the packet capture. I think ph-3, orp-4 so it may be doing exactly that.

rstrouse, 2021-03-23:
Is the max tank level 6?

brentratliff, 2021-03-23:
I changed the levels from 3->4 on PH and 4-3 on ORP. The max level is 6.

brentratliff, 2021-03-23:
I just got this error, decreasing the Acid level:

ApiError: Message aborted after 2 attempt(s): 165,0,144,16,146,21,2,238,2,198,2,2,0,249,0,43,0,150,20,0,0,0,0,0,0,0,0,5,118
    at SendRecieveBuffer.writeMessage (/home/pi/nodejs-poolController/controller/comms/Comms.ts:341:27)
    at SendRecieveBuffer.processWaitPacket (/home/pi/nodejs-poolController/controller/comms/Comms.ts:291:29)
    at SendRecieveBuffer.processOutbound (/home/pi/nodejs-poolController/controller/comms/Comms.ts:299:30)
    at SendRecieveBuffer.processPackets (/home/pi/nodejs-poolController/controller/comms/Comms.ts:282:21)
    at Timeout._onTimeout (/home/pi/nodejs-poolController/controller/comms/Comms.ts:254:109)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)

rstrouse, 2021-03-23:
You mean the Acid tank level?

brentratliff, 2021-03-23:
It lets me change the levels intermittently now, but with any save to either tank,  will decrease both tanks my one.

_You mean the Acid tank level?_
Yes

rstrouse, 2021-03-23:
When you increase the acid tank to 6 using EasyTouch what is displayed in dashPanel?

[27 more replies]

closed 2021-03-28 by tagyoureit