tagyoureit/nodejs-poolController #273: Connection to system extra equipment? ID showing as internal?

DrJeff, 2021-03-09:
Lost connection to RS485

Getting no comms to the equipment I have, used to work tried changing the RS485 adapter to no avail

```

> nodejs-poolcontroller@6.5.1 start
> npm run build && node dist/app.js

> nodejs-poolcontroller@6.5.1 build
> tsc

Init state for Pool Controller
[3/9/2021, 11:09:13 AM] info: Initializing Nixie Controller
[3/9/2021, 11:09:13 AM] info: Nixie Controller Initialized
[3/9/2021, 11:09:13 AM] info: Server is now listening on 0.0.0.0:4220
[3/9/2021, 11:09:13 AM] info: Serial port: /dev/tty.wchusbserial14610 request to open successful
[3/9/2021, 11:09:20 AM] info: Searching chlorinators, pumps and chem controllers
RESETTING DATA
[3/9/2021, 11:09:20 AM] warn: dt:Tue Mar 09 2021 11:09:20 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[3/9/2021, 11:09:20 AM] warn: dt:Tue Mar 09 2021 11:09:20 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[3/9/2021, 11:09:20 AM] info: Searching for a pump at address... 96
[3/9/2021, 11:09:20 AM] info: Searching for a pump at address... 97
[3/9/2021, 11:09:21 AM] info: New socket client connected _7bAXF3FeYGVu3vYAAAA -- 192.168.0.53
[3/9/2021, 11:09:24 AM] info: New socket client connected otm5GUzGcFHKhbBjAAAB -- 192.168.0.53
[3/9/2021, 11:09:24 AM] info: [11:09:24 AM] 192.168.0.53 GET /state/all?null {}
[3/9/2021, 11:09:24 AM] warn: Message aborted after 4 attempt(s): 16,2,80,0,0,98,16,3
[3/9/2021, 11:09:24 AM] warn: No Chlorinator Found
[3/9/2021, 11:09:27 AM] warn: Message aborted after 2 attempt(s): 165,0,96,33,4,1,255,2,42
[3/9/2021, 11:09:27 AM] info: No pump found at address 96: Message aborted after 2 attempt(s): 165,0,96,33,4,1,255,2,42
[3/9/2021, 11:09:29 AM] warn: Message aborted after 2 attempt(s): 165,0,97,33,4,1,255,2,43
[3/9/2021, 11:09:29 AM] info: No pump found at address 97: Message aborted after 2 attempt(s): 165,0,97,33,4,1,255,2,43
[3/9/2021, 11:09:31 AM] warn: Message aborted after 2 attempt(s): 165,0,144,16,210,1,210,2,234
[3/9/2021, 11:09:31 AM] info: No chemController found at address 144: Message aborted after 2 attempt(s): 165,0,144,16,210,1,210,2,234
[3/9/2021, 11:09:33 AM] warn: Message aborted after 2 attempt(s): 165,0,145,16,210,1,210,2,235
[3/9/2021, 11:09:33 AM] info: No chemController found at address 145: Message aborted after 2 attempt(s): 165,0,145,16,210,1,210,2,235
[3/9/2021, 11:09:35 AM] warn: Message aborted after 2 attempt(s): 165,0,146,16,210,1,210,2,236
[3/9/2021, 11:09:35 AM] info: No chemController found at address 146: Message aborted after 2 attempt(s): 165,0,146,16,210,1,210,2,236
[3/9/2021, 11:09:37 AM] warn: Message aborted after 2 attempt(s): 165,0,147,16,210,1,210,2,237
[3/9/2021, 11:09:37 AM] info: No chemController found at address 147: Message aborted after 2 attempt(s): 165,0,147,16,210,1,210,2,237
```

**Pool Equipment**
 - Controller: Virtual
 - Controller Model:NA
 - Controller Firmware Version: NA
 - Pump(s) manufacturer and model: Intelliflo VS
 - Chlorinator: IntelliChlor-60

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser Chrome

**Smartphone (please complete the following information):**
 - Device: iPhone Xs
 - OS: 14

DrJeff, 2021-03-09:
[replay (2).zip](https://github.com/tagyoureit/nodejs-poolController/files/6110904/replay.2.zip)

rstrouse, 2021-03-09:
Ok so there is no communication on the bus other than what njspc is speaking so let's be sure of your RS485 adapter.  I would have expected invalid messages if the adapter was translating bits.  Right now njspc is shouting at your equipment and your equipment is giving it the silent treatment.

You cannot use a TTY connection as RS485 is a differential voltage topology and TTY will set one or the other pin high or low.  From looking at your config file it looks like this is a TTY interface.  There are some TTY USB adapters that do both but they must be configured to one or the other.  Which adapter do you have?

DrJeff, 2021-03-09:
connects to njspc now communicate im not sure it used to but not now

rstrouse, 2021-03-09:
Is it this?
https://www.amazon.com/JBtek-Converter-Adapter-ch340T-Supported/dp/B00NKAJGZM/ref=asc_df_B00NKAJGZM/?tag=hyprod-20&linkCode=df0&hvadid=309773039951&hvpos=&hvnetw=g&hvrand=6136622847125386675&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9032458&hvtargid=pla-599566681644&psc=1

Also the driver interface needs to be set to 9600, No Parity, 8 bits, 1 stop bit.

DrJeff, 2021-03-09:
yup I have that one setup on the mac so how would I change the driver interfacae? I tried to view with coolterm with those settings but nothing

rstrouse, 2021-03-09:
Ok so the adapter should have detected the driver interface as soon as you plugged it in and .  I am pretty sure the endpoints you listed above are not correct and should be something more like /dev/ttyUSB0 or /dev/ttyUSBn (where n is the port number).  From what I read out there MAC OS has support built in from FTDI and CH340.

I am not a MAC guy but when you plug this adapter into win10 it shows up as COM3 for me which is what I put in my config.json for the rs485Port entry.
[image]

DrJeff, 2021-03-09:
No they are correct on linux they are connecting as well and are listed as /dev/ttyUSB0 but no coms as well but on mac they aren't USB0 or USB1 they are 
/dev/tty.usbserial-14610         /dev/ttyqb                       /dev/ttys007                     /dev/ttytc                       /dev/ttyv9
/dev/tty.wchusbserial14610

DrJeff, 2021-03-09:
Ok here is on Pi with USB0
```
> nodejs-poolcontroller@6.5.1 start /home/pi/PoolController
> npm run build && node dist/app.js

> nodejs-poolcontroller@6.5.1 build /home/pi/PoolController
> tsc

Init state for Pool Controller
[3/9/2021, 12:39:55 PM] info: Initializing Nixie Controller
[3/9/2021, 12:39:55 PM] info: Nixie Controller Initialized
[3/9/2021, 12:39:55 PM] info: Server is now listening on 0.0.0.0:4200
[3/9/2021, 12:39:55 PM] info: Serial port: /dev/ttyUSB0 request to open successful
[3/9/2021, 12:40:20 PM] info: [12:40:20 PM] 192.168.0.53 GET /state/all?null {}
[3/9/2021, 12:40:20 PM] info: New socket client connected -N2IOy8UsZEovlVAAAAA -- 192.168.0.53
[3/9/2021, 12:40:20 PM] info: [12:40:20 PM] 192.168.0.53 GET /state/all?null {}
[3/9/2021, 12:40:32 PM] info: [12:40:32 PM] 192.168.0.53 GET /config/all?null {}
[3/9/2021, 12:40:32 PM] info: [12:40:32 PM] 192.168.0.53 GET /config/options/general?null {}
[3/9/2021, 12:40:34 PM] info: [12:40:34 PM] 192.168.0.53 GET /config/all?null {}
[3/9/2021, 12:40:34 PM] info: [12:40:34 PM] 192.168.0.53 GET /config/options/general?null {}
[3/9/2021, 12:40:36 PM] info: [12:40:36 PM] 192.168.0.53 GET /config/options/pumps?null {}
[3/9/2021, 12:40:52 PM] info: [12:40:52 PM] 192.168.0.53 GET /state/all?null {}
[3/9/2021, 12:40:52 PM] info: New socket client connected mkpEYxhRrndAv7S4AAAB -- 192.168.0.53
[3/9/2021, 12:40:52 PM] info: [12:40:52 PM] 192.168.0.53 GET /state/all?null {}
[3/9/2021, 12:40:56 PM] info: [12:40:56 PM] 192.168.0.53 GET /config/all?null {}
[3/9/2021, 12:40:56 PM] info: [12:40:56 PM] 192.168.0.53 GET /config/options/general?null {}
[3/9/2021, 12:40:58 PM] info: [12:40:58 PM] 192.168.0.53 GET /config/options/pumps?null {}
[3/9/2021, 12:41:10 PM] info: [12:41:10 PM] 192.168.0.53 GET /config/options/pumps?null {}
[3/9/2021, 12:41:10 PM] info: [12:41:10 PM] 192.168.0.53 PUT /config/pump {"id":-1,"name":"Pump 1","type":128,"address":96,"primingTime":0,"primingSpeed":0,"minSpeed":450,"maxSpeed":3450,"minFlow":0,"maxFlow":130,"circuits":[{"circuit":6,"speed":3450,"units":0}]}
```

rstrouse, 2021-03-09:
In that output I see it actually opening the serial port.   Rename the poolConfog.json in the data directory and restart njspc.  Don't manually add the pump lets see if it finds it on its own.

rstrouse, 2021-03-09:
The drivers for this device can be found here.
http://www.wch.cn/download/ch341ser_exe.html

DrJeff, 2021-03-09:
It is opening the usb but not finding I've been deleting both configs in the data folder but will not create the pump
 Those drivers are already in Linux unless I'm wrong. On the Mac I believe I added them. But it wouldn't see it in njspc if no drivers correct?

tagyoureit, 2021-03-09:
@DrJeff One unfortunate thing I noticed on Mac is the 6 digit suffix in the name changes every time you unplug/replug in the adapter (and probably restart as well).  Check to make sure it hasn't changed.  Because it's been a while (and I have 10 mins in between my sales meetings) I went outside with my Mac, shutdown my OCP and screenlogic, and started up njsPC.  It did find the equipment.  So that validates the code is still good. It's either your adapter or wiring...

```
Init state for Pool Controller
[3/9/2021, 1:51:25 PM] warn: HTTPS not enabled because key or crt file is missing.
[3/9/2021, 1:51:25 PM] info: Init rem interface: Relay Equipment Manager
[3/9/2021, 1:51:25 PM] info: Opening Relay Equipment Manager socket on http://10.0.0.171:8080
[3/9/2021, 1:51:25 PM] info: Init http interface: Rules Manager
[3/9/2021, 1:51:25 PM] info: Initializing Nixie Controller
[3/9/2021, 1:51:25 PM] info: Nixie Controller Initialized
[3/9/2021, 1:51:25 PM] info: Server is now listening on 0.0.0.0:4200
[3/9/2021, 1:51:25 PM] info: Serial port: /dev/tty.usbserial-142320 request to open successful
[3/9/2021, 1:51:25 PM] info: Relay Equipment Manager socket connected
[3/9/2021, 1:51:32 PM] info: Searching chlorinators, pumps and chem controllers
RESETTING DATA
[3/9/2021, 1:51:32 PM] warn: dt:Tue Mar 09 2021 13:51:32 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[3/9/2021, 1:51:32 PM] warn: dt:Tue Mar 09 2021 13:51:32 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[3/9/2021, 1:51:32 PM] info: Searching for a pump at address... 96
[3/9/2021, 1:51:32 PM] info: Searching for a pump at address... 97
[3/9/2021, 1:51:32 PM] verbose: Wrote packet [16,2,80,0,0,98,16,3].  Retries remaining: 3
[3/9/2021, 1:51:33 PM] silly: Retrying outbound message after 1.034secs with 3 attempt(s) left. - 16,2,80,0,0,98,16,3
[3/9/2021, 1:51:33 PM] verbose: Wrote packet [16,2,80,0,0,98,16,3].  Retries remaining: 2
[3/9/2021, 1:51:34 PM] silly: Retrying outbound message after 1.028secs with 2 attempt(s) left. - 16,2,80,0,0,98,16,3
[3/9/2021, 1:51:34 PM] verbose: Wrote packet [16,2,80,0,0,98,16,3].  Retries remaining: 1
[3/9/2021, 1:51:35 PM] silly: Retrying outbound message after 1.025secs with 1 attempt(s) left. - 16,2,80,0,0,98,16,3
[3/9/2021, 1:51:35 PM] verbose: Wrote packet [16,2,80,0,0,98,16,3].  Retries remaining: 0
[3/9/2021, 1:51:36 PM] silly: Retrying outbound message after 1.028secs with 0 attempt(s) left. - 16,2,80,0,0,98,16,3
[3/9/2021, 1:51:36 PM] warn: Message aborted after 4 attempt(s): 16,2,80,0,0,98,16,3
[3/9/2021, 1:51:36 PM] warn: No Chlorinator Found
[3/9/2021, 1:51:36 PM] verbose: Wrote packet [255,0,255,165,0,96,33,4,1,255,2,42].  Retries remaining: 1
[3/9/2021, 1:51:36 PM] verbose: Wrote packet [255,0,255,165,0,97,33,4,1,255,2,43].  Retries remaining: 1
[3/9/2021, 1:51:37 PM] verbose: Wrote packet [255,0,255,165,0,144,16,210,1,210,2,234].  Retries remaining: 1
[3/9/2021, 1:51:38 PM] silly: Retrying outbound message after 1.031secs with 1 attempt(s) left. - 165,0,144,16,210,1,210,2,234
[3/9/2021, 1:51:38 PM] verbose: Wrote packet [255,0,255,165,0,144,16,210,1,210,2,234].  Retries remaining: 0
[3/9/2021, 1:51:39 PM] silly: Retrying outbound message after 1.026secs with 0 attempt(s) left. - 165,0,144,16,210,1,210,2,234
[3/9/2021, 1:51:39 PM] warn: Message aborted after 2 attempt(s): 165,0,144,16,210,1,210,2,234
[3/9/2021, 1:51:39 PM] info: No chemController found at address 144: Message aborted after 2 attempt(s): 165,0,144,16,210,1,210,2,234
[3/9/2021, 1:51:39 PM] verbose: Wrote packet [255,0,255,165,0,145,16,210,1,210,2,235].  Retries remaining: 1
[3/9/2021, 1:51:40 PM] silly: Retrying outbound message after 1.026secs with 1 attempt(s) left. - 165,0,145,16,210,1,210,2,235
[3/9/2021, 1:51:40 PM] verbose: Wrote packet [255,0,255,165,0,145,16,210,1,210,2,235].  Retries remaining: 0
[3/9/2021, 1:51:41 PM] silly: Retrying outbound message after 1.03secs with 0 attempt(s) left. - 165,0,145,16,210,1,210,2,235
[3/9/2021, 1:51:41 PM] warn: Message aborted after 2 attempt(s): 165,0,145,16,210,1,210,2,235
[3/9/2021, 1:51:41 PM] info: No chemController found at address 145: Message aborted after 2 attempt(s): 165,0,145,16,210,1,210,2,235
[3/9/2021, 1:51:41 PM] verbose: Wrote packet [255,0,255,165,0,146,16,210,1,210,2,236].  Retries remaining: 1
[3/9/2021, 1:51:42 PM] silly: Retrying outbound message after 1.029secs with 1 attempt(s) left. - 165,0,146,16,210,1,210,2,236
[3/9/2021, 1:51:42 PM] verbose: Wrote packet [255,0,255,165,0,146,16,210,1,210,2,236].  Retries remaining: 0
[3/9/2021, 1:51:43 PM] silly: Retrying outbound message after 1.026secs with 0 attempt(s) left. - 165,0,146,16,210,1,210,2,236
[3/9/2021, 1:51:43 PM] warn: Message aborted after 2 attempt(s): 165,0,146,16,210,1,210,2,236
[3/9/2021, 1:51:43 PM] info: No chemController found at address 146: Message aborted after 2 attempt(s): 165,0,146,16,210,1,210,2,236
[3/9/2021, 1:51:43 PM] verbose: Wrote packet [255,0,255,165,0,147,16,210,1,210,2,237].  Retries remaining: 1
[3/9/2021, 1:51:44 PM] silly: Retrying outbound message after 1.018secs with 1 attempt(s) left. - 165,0,147,16,210,1,210,2,237
[3/9/2021, 1:51:44 PM] verbose: Wrote packet [255,0,255,165,0,147,16,210,1,210,2,237].  Retries remaining: 0
[... cut ...]
[3/9/2021, 1:51:45 PM] warn: Message aborted after 2 attempt(s): 165,0,147,16,210,1,210,2,237
[3/9/2021, 1:51:45 PM] info: No chemController found at address 147: Message aborted after 2 attempt(s): 165,0,147,16,210,1,210,2,237
[3/9/2021, 1:51:45 PM] verbose: Wrote packet [255,0,255,165,0,96,33,7,0,1,45].  Retries remaining: 1
[3/9/2021, 1:51:45 PM] info: Found pump at 1 address 96
[3/9/2021, 1:51:45 PM] verbose: Wrote packet [255,0,255,165,0,97,33,7,0,1,46].  Retries remaining: 1
[3/9/2021, 1:51:45 PM] info: Found pump at 2 address 97
```

DrJeff, 2021-03-10:
Yes the equipment is powered I also changed over to Linux (pi) so there wouldn't be any thoughts to the Mac being the issue. But Still like there is no connection to Pump or Chlorinator. I'm leaning to having a bad cable but 2 pieces of equipment so both bad? So more like another bad RS485? But just ordered it new maybe I should try another RS485 how about this one? https://www.amazon.com/Serial-Converter-Adapter-Supports-Windows/dp/B076WVFXN8/ref=sr_1_4?dchild=1&keywords=rs485&qid=1615314456&sr=8-4
[replay (3).zip](https://github.com/tagyoureit/nodejs-poolController/files/6112703/replay.3.zip)

DrJeff, 2021-03-10:
If you want to just see it here is the start up with silly on
```
pi@raspberrypi:~/PoolController $ npm start

> nodejs-poolcontroller@6.5.1 start /home/pi/PoolController
> npm run build && node dist/app.js

> nodejs-poolcontroller@6.5.1 build /home/pi/PoolController
> tsc

[3/9/2021, 6:22:09 PM] info: Starting Replay Capture.
Init state for Pool Controller
[3/9/2021, 6:22:10 PM] info: Initializing Nixie Controller
[3/9/2021, 6:22:10 PM] info: Nixie Controller Initialized
[3/9/2021, 6:22:10 PM] info: Server is now listening on 0.0.0.0:4200
[3/9/2021, 6:22:10 PM] info: Serial port: /dev/ttyUSB0 request to open successful
[3/9/2021, 6:22:16 PM] info: Searching chlorinators, pumps and chem controllers
RESETTING DATA
[3/9/2021, 6:22:16 PM] warn: dt:Tue Mar 09 2021 18:22:16 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[3/9/2021, 6:22:16 PM] warn: dt:Tue Mar 09 2021 18:22:16 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[3/9/2021, 6:22:16 PM] info: Searching for a pump at address... 96
[3/9/2021, 6:22:16 PM] info: Searching for a pump at address... 97
[3/9/2021, 6:22:16 PM] info: {"id":1,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,0],[0],[98,16,3]],"ts":"2021-03-09T18:22:16.722-0800"}
[3/9/2021, 6:22:16 PM] verbose: Wrote packet [16,2,80,0,0,98,16,3].  Retries remaining: 3
[3/9/2021, 6:22:17 PM] silly: Retrying outbound message after 1.013secs with 3 attempt(s) left. - 16,2,80,0,0,98,16,3
[3/9/2021, 6:22:17 PM] info: {"id":1,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,0],[0],[98,16,3]],"ts":"2021-03-09T18:22:17.739-0800"}
[3/9/2021, 6:22:17 PM] verbose: Wrote packet [16,2,80,0,0,98,16,3].  Retries remaining: 2
[3/9/2021, 6:22:18 PM] silly: Retrying outbound message after 1.007secs with 2 attempt(s) left. - 16,2,80,0,0,98,16,3
[3/9/2021, 6:22:18 PM] info: {"id":1,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,0],[0],[98,16,3]],"ts":"2021-03-09T18:22:18.749-0800"}
[3/9/2021, 6:22:18 PM] verbose: Wrote packet [16,2,80,0,0,98,16,3].  Retries remaining: 1
[3/9/2021, 6:22:20 PM] silly: Retrying outbound message after 1.385secs with 1 attempt(s) left. - 16,2,80,0,0,98,16,3
[3/9/2021, 6:22:20 PM] info: {"id":1,"valid":true,"dir":"out","proto":"chlorinator","pkt":[[],[],[16,2,80,0],[0],[98,16,3]],"ts":"2021-03-09T18:22:20.138-0800"}
[3/9/2021, 6:22:23 PM] verbose: Wrote packet [16,2,80,0,0,98,16,3].  Retries remaining: 0
[3/9/2021, 6:22:23 PM] silly: Retrying outbound message after 3.564secs with 0 attempt(s) left. - 16,2,80,0,0,98,16,3
[3/9/2021, 6:22:23 PM] warn: Message aborted after 4 attempt(s): 16,2,80,0,0,98,16,3
[3/9/2021, 6:22:23 PM] warn: No Chlorinator Found
[3/9/2021, 6:22:23 PM] info: {"id":2,"valid":true,"dir":"out","proto":"pump","pkt":[[],[255,0,255],[165,0,96,33,4,1],[255],[2,42]],"ts":"2021-03-09T18:22:23.948-0800"}
[3/9/2021, 6:22:23 PM] verbose: Wrote packet [255,0,255,165,0,96,33,4,1,255,2,42].  Retries remaining: 1
[3/9/2021, 6:22:24 PM] silly: Retrying outbound message after 1.003secs with 1 attempt(s) left. - 165,0,96,33,4,1,255,2,42
[3/9/2021, 6:22:24 PM] info: {"id":2,"valid":true,"dir":"out","proto":"pump","pkt":[[],[255,0,255],[165,0,96,33,4,1],[255],[2,42]],"ts":"2021-03-09T18:22:24.954-0800"}
[3/9/2021, 6:22:24 PM] verbose: Wrote packet [255,0,255,165,0,96,33,4,1,255,2,42].  Retries remaining: 0
[3/9/2021, 6:22:25 PM] silly: Retrying outbound message after 1.005secs with 0 attempt(s) left. - 165,0,96,33,4,1,255,2,42
[3/9/2021, 6:22:25 PM] warn: Message aborted after 2 attempt(s): 165,0,96,33,4,1,255,2,42
[3/9/2021, 6:22:25 PM] info: No pump found at address 96: Message aborted after 2 attempt(s): 165,0,96,33,4,1,255,2,42
[3/9/2021, 6:22:26 PM] info: {"id":3,"valid":true,"dir":"out","proto":"pump","pkt":[[],[255,0,255],[165,0,97,33,4,1],[255],[2,43]],"ts":"2021-03-09T18:22:26.072-0800"}
[3/9/2021, 6:22:26 PM] verbose: Wrote packet [255,0,255,165,0,97,33,4,1,255,2,43].  Retries remaining: 1
[3/9/2021, 6:22:26 PM] info: New socket client connected qcM-MoD3YebdnJPIAAAA -- 192.168.0.53
[3/9/2021, 6:22:26 PM] info: [6:22:26 PM] 192.168.0.53 GET /state/all?null {}
[3/9/2021, 6:22:27 PM] silly: Retrying outbound message after 1.021secs with 1 attempt(s) left. - 165,0,97,33,4,1,255,2,43
[3/9/2021, 6:22:27 PM] info: {"id":3,"valid":true,"dir":"out","proto":"pump","pkt":[[],[255,0,255],[165,0,97,33,4,1],[255],[2,43]],"ts":"2021-03-09T18:22:27.095-0800"}
[3/9/2021, 6:22:27 PM] verbose: Wrote packet [255,0,255,165,0,97,33,4,1,255,2,43].  Retries remaining: 0
[3/9/2021, 6:22:28 PM] silly: Retrying outbound message after 1.007secs with 0 attempt(s) left. - 165,0,97,33,4,1,255,2,43
[3/9/2021, 6:22:28 PM] warn: Message aborted after 2 attempt(s): 165,0,97,33,4,1,255,2,43
[3/9/2021, 6:22:28 PM] info: No pump found at address 97: Message aborted after 2 attempt(s): 165,0,97,33,4,1,255,2,43
[3/9/2021, 6:22:28 PM] info: {"id":4,"valid":true,"dir":"out","proto":"intellichem","pkt":[[],[255,0,255],[165,0,144,16,210,1],[210],[2,234]],"ts":"2021-03-09T18:22:28.216-0800"}
[3/9/2021, 6:22:28 PM] verbose: Wrote packet [255,0,255,165,0,144,16,210,1,210,2,234].  Retries remaining: 1
[... cut ...]
[3/9/2021, 6:22:39 PM] info: No chemController found at address 147: Message aborted after 2 attempt(s): 165,0,147,16,210,1,210,2,237
[3/9/2021, 6:23:00 PM] warn: dt:Tue Mar 09 2021 18:22:16 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[3/9/2021, 6:23:00 PM] warn: dt:Tue Mar 09 2021 18:22:16 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
```

DrJeff, 2021-03-10:
Thanks, I will order another one and try. Is there a way to set the rs485 to network, I have a wiznet Wiz108 lying around I can try it out

tagyoureit, 2021-03-10:
This app doesn’t talk to the network protocol so that wouldn’t do much good. 

@rstrouse  is it worth trying the captureChanges script?  (Not at my computer now.)

rstrouse, 2021-03-10:
@tagyoureit -- perhaps.  We should include that as a utility.  I don't think I even have a copy of that floating around anymore.  It's maddening how the registers on these devices can get out of whack.

Actually it may talk to that Wiznet device (I assume you have an SR model).  You will need to configure it as 9600 8N1 then set it up in config.json  where the netHost is the IP address and the netPort is the port assigned to the device.
```
  "controller": {
    "comms": {
      "rs485Port": "/dev/ttyUSB0",
      "mockPort": false,
      "netConnect": true,
      "netHost": "<the ip address of the Wiznet device>",
      "netPort": 9801,
      "inactivityRetry": 10,
      "portSettings": {
        "baudRate": 9600,
        "dataBits": 8,
        "parity": "none",
        "stopBits": 1,
        "flowControl": false,
        "autoOpen": false,
        "lock": false
      }
    }

```

DrJeff, 2021-03-10:
Ok will go grab it and try.......

DrJeff, 2021-03-10:
Yes looks like 3 bad adapters go figure the Wiznet one is working, had to fire up windows to setup the wiznet device on network. Looks like some issues but lets see what you think
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6113078/replay.zip)

DrJeff, 2021-03-10:
It Works!!!! Yes
Lots of non-complete messages so hoping it is due to the network connection. I'm still going to order another adapter and try it directly connected. Now for some REM help I will go and create a ticket over there. Thanks Guys! here is another log
[replay (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/6113339/replay.1.zip)

tagyoureit, 2021-03-10:
[FixSP.zip](https://github.com/tagyoureit/nodejs-poolController/files/6113406/FixSP.zip)

You can try this with the old adapters.  Sometime I can run this and close it and the port works; sometime I can run it and leave it open and the port works.  Don't worry about what it spits out (mostly garbage) but it will try to reset the port registers.

`node captureChanges --port /dev/tty.[port-name]`

Glad it's working!  The `...hasn't given a complete header` just means that the serial port is processing the packets in dribs and drabs and is sending them over to njsPC.  I personally think we should silence that message.  

A message has truly `failed` when you see something like this:
```
[3/9/2021, 8:59:29 PM] info: {"id":519,"valid":false,"dir":"in","proto":"chlorinator","pkt":[[],[],[16,2,0,18],[66,128],[230,16,3]],"ts": "2021-03-09T20:59:29.046-0800"}
```
I made that up, but basically it will say `valid: false`

DrJeff, 2021-03-10:
Awesome! I will try to resurrect some of these dead adapters
😄

closed 2021-03-10 by DrJeff