tagyoureit/nodejs-poolController #274: Use Nixie for chemical controller and whatever else I can

DrJeff, 2021-03-14:
How do I use the nixie controller, how to get started?
@tagyoureit

tagyoureit, 2021-03-14:
What do you want to measure and what equipment do/will you have to accomplish that?  

The two most common are pH and ORP and Nixie + REM can also help you monitor pretty much all Atlas Scientific probes plus a few different temp sensors.

DrJeff, 2021-03-14:
I have atlas PH and ORP embedded chips (cheap probes) temps are dallas dsb1820 but for the nixie I wanted to control my valves, (relays) pool fill, Heater, Aux pumps (spa pump, spa jets) I'm running two bodies of water that are separated (2 filters) but can be one (spa spillway). So I filter and run spillway but there is a set of valves that run the heater for pool only or spa only the purpose is so when we have parties the spa is heated and filtered independently and the pool is filtered and circulated independently. If I can just pass all the values from PH, ORP, Temps, Valve control from MQTT that will work because I have all this working with Esphome, Node Red, and Home Assistant but wanted a "Real" controller and my PH ORP needs better logic.

tagyoureit, 2021-03-14:
So your setup allows you to basically switch between a dual body system and a shared system?  How many pumps do you have; assuming 2?

What reads directly from your Atlas probes now?  Is it Atlas -> EspHome -> Node Red -> REM/njsPC?

Do you have your valves controlled by relays?

DrJeff, 2021-03-14:
> So your setup allows you to basically switch between a dual body system and a shared system? How many pumps do you have; assuming 2?
3
> 
> What reads directly from your Atlas probes now? Is it Atlas -> EspHome -> Node Red -> REM/njsPC?
esphome
> 
> Do you have your valves controlled by relays?
yes esphome for now

DrJeff, 2021-03-14:
I did order a tentacle but the reason I went with esphome was that my pi would overheat in the summer in a sealed box. but the microcontroller works fine at the extreme temperatures.

also above I forgot....
And I forgot Microbrite Color I currentlly use 5 of microbrites 3 pool, 1 waterfall, 1 spa. These are controlled by esphome with a relay board also have yard lights connected it. I did have a chlorine pump

I also read pressure with a transducer to the esphome. So if I could just send raw data into something that works also. Thanks I got the temps updating in njspc now.

tagyoureit, 2021-03-15:
Ultimately, you'll want to control everything through REM/Nixie.  MQTT just isn't reliable enough for communications to control the pool equipment and monitor the sensors.  

@rstrouse is driving some big changes to allow controlling the relay and also how we are configuring equipment.

In the meantime, I'd suggest firing up [REM](https://github.com/rstrouse/relayEquipmentManager) and trying to connect your existing sensors to it so we can understand what gaps are there.  (dsb1820 needs to be added.)

You may also want to get an actively vented [nema enclosure](https://www.streakwave.com/hana-wireless-hw-n14-v-frp-nema-14x12x7-vented-mounting-plate).  I'm not sure if we'll be supporting esp8266 as a base system.  Currently we support Pi, Beaglebone Black, and Orange Pi.

DrJeff, 2021-03-15:
Yes I understand the use of REM but need to understand how to use Nixie I guess. 
I can connect all the relays (Blower, spa spillover, waterfall, acid pump, chlorine pump, valve top, valve before heater, valve after heater, spa heater, fill line, solar valve, spa pump, spa jets, yard lights1, yard lights2, yard lights3, pool deep led, pool shallow led1, pool shallow led2, waterfall led, spa led) 
I just saw that I can use my existing MCP23017 I2c I think for the relays through I2c! 😃 

Can lightshows be controlled here also, I have the MicroBrites, and various yard lights LED on transformer currently using esphome for lighting.

Need to connect the temperature sensors dallas (spa_temp, air_temp, solar_temp, pool _temp) 

The pressure transducer I think I see how via ADS1115 i2c?

I see how the EZO sensors (ORP, PH) connect to REM but not where I do the connection to njspc for setting the pump (acid on and off) and tank levels and such.

Now to tie it all together is where I'm lost. I use Home Assistant as the brains I guess, It decides what gets turned on based on temps, days, and physical actions. It also uses Node Red and Esphome for collection and distribution of it's logic.
So all this can be replaced with Nixie? I guess thats the question and How?

tagyoureit, 2021-03-17:
> Can lightshows be controlled here also, I have the MicroBrites, and various yard lights LED on transformer currently using esphome for lighting.

The manual says you can control Microbrites by setting them as Intellibrite, so yes, this should work.  

> I see how the EZO sensors (ORP, PH) connect to REM but not where I do the connection to njspc for setting the pump (acid on and off) and tank levels and such.

You need to add the following in your njsPC `config.json` in the `interfaces` section:

```
      "rem": {
        "name": "Relay Equipment Manager",
        "type": "rem",
        "enabled": true,
        "options": {
          "protocol": "http://",
          "host": "your.ip.address.001",
          "port": 8080,
          "headers": {
            "content-type": "application/json"
          }
        },
```

Then setup the controller in settings.  Chemistry -> Add Controller -> Type (REM Chem).
[image]

DrJeff, 2021-03-17:
> The manual says you can control Microbrites by setting them as Intellibrite, so yes, this should work.

But how to add to the virtual controller they are just relays in REM

Cool didn't even notice the REM Chem! 😁  Thanks Really don't understand how to use/install Nixie

tagyoureit, 2021-03-17:
The code today allows you to control pumps, circuits, and Intellichlor in a standalone manner. 

Rem and Nixie act as a replacement for Intellichem or a home grown chemical controller. 

We are working on allowing Nixie to be a full virtual controller that can replace IntelliTouch/EasyTouch/IntelliCenter but it isn't quite there yet. You'll need to be a little patient for that part.

DrJeff, 2021-03-17:
> You'll need to be a little patient for that part.

I get it now, Nixie is already doing its thing for now! 😀  Any reason you know why I keep killing RS485 dongles

tagyoureit, 2021-03-17:
> Any reason you know why I keep killing RS485 dongles

My best guess is either surges on the data line or incorrectly (or not) terminated lines.  Could also happen if the host system has a power failure.  It happens to me, too, and I haven't pinpointed why, either.

DrJeff, 2021-03-28:
Trying to use REM Chemical controller with Intellichlor IC60 checked the box to use IC60 for ORP but not sure what and how to setup properly for the IC60 to turn on off automatically (IC60 100%?), Acid pump is working nicely 😀

johnny2678, 2021-03-28:
> not sure what and how to setup properly for the IC60 to turn on off automatically (IC60 100%?)

I don't think this has been coded yet.

DrJeff, 2021-04-27:
Ok I think this is a Nixie thing. I can't get my pump to operate with Nixie Dual Body here is the replay. No errors for pump but not turning on?
[replay (9).zip](https://github.com/tagyoureit/nodejs-poolController/files/6386875/replay.9.zip)

Something funky with the heaters also I think it's the Bodies distinction or better yet the Circuits if I set Body1 as Pool or Spa and Body2 as Pool or Spa I end up with both Bodies as Pool in circuits setup under the type. So the second circuit is always set type as Pool (hardcoded?).

DrJeff, 2021-04-28:
Now I assume the watersensor1 is for body1 and watersensor2 is for body2
??
@rstrouse should i be putting these questions in 
https://github.com/rstrouse/relayEquipmentManager/issues/3

rstrouse, 2021-04-28:
If you are configured as a dual body then this is definitely true.  Watersensor1 will map to the first body and watersensor2 will map to body 2.  Are you sure you are a dual body system or do they share a filter?  You can have multiple pumps and a spillway where the intake and return valves will automatically rotate.

DrJeff, 2021-04-28:
I built it with dual body but I can share a heater and spillway

Two filters for parties!

DrJeff, 2021-04-28:
Lots of valves (4) that lay dormant unless I want to use Gas Heater to heat pool

rstrouse, 2021-04-28:
No you can assign it either way.  However based on what you have described you have a shared system.  I have valves that isolate the spa and water control is on a separate pump.   There are two filters but a single chemistry system.  I can set the valves in such a way to run the spillway or isolate the bodies.  In the end the water is connected between them.

DrJeff, 2021-04-28:
Ok yes but the valves are set to stay in an isolated state. So I will re-setup as a shared body because you are correct sanitation is shared that is why I set a spillover event unless a party then the chlorinator is used to feed constantly to the spa but under daily it's shared body with SWG. 

But my heater, and main pump wasn't working in the Dual body so I will try again in shared body

rstrouse, 2021-04-28:
Yeah I’ll replicate that setup.  For sure it should work with the pumps and heaters.  Look at your plumbing because the Intake and Return valves are the key.  These dictate where the water is drawn from and where it goes.

DrJeff, 2021-04-28:
> Yeah I’ll replicate that setup. For sure it should work with the pumps and heaters. Look at your plumbing because the Intake and Return valves are the key. These dictate where the water is drawn from and where it goes.

Yes I get it Thanks

DrJeff, 2021-04-28:
No that way wont work because it relies on one pump to share the bodies(due to sanitation/filtration), so if you turn on the spa you lose the pool circulation. My pool might be connected but it is treated as separate bodies and filters unless spillover is wanted or use of gas heater for the pool (next to never).

rstrouse, 2021-04-28:
Please make sure you have the latest from njsPC, dashPanel, and REM.  There was an issue a couple of pushes ago where the heaters were being misreported.

DrJeff, 2021-04-28:
With Dual body heaters work fine, but the Intelliflo isn't working I can create but it has no comms?
[image]
[image]

, can't start low, med, high features to set pump at those speeds.

rstrouse, 2021-04-28:
Hmmm.  Write the log to a file then post the console.  Not sure how there is a comms error with setting a feature.

DrJeff, 2021-04-28:
With capture my config was changed to Single body and lost setup??

DrJeff, 2021-04-28:
added Zip
[replay10.zip](https://github.com/tagyoureit/nodejs-poolController/files/6389128/replay10.zip)

tagyoureit, 2021-04-28:
The challenge here is that @DrJeff's pool is neither a shared body nor a dual body.  A shared body can have intake/return valves and be in pool, spa or spillway mode.  The dual body systems can operate two distinct sets of equipment.  

The setup here can be either a shared body or a dual body (and maybe it's also a Transformer and can turn into Optimus Prime?).  

DrJeff has pair of intake/returns in that it keeps the body separate or creates a spillway.  Also, it sounds like one filter can be included/excluded on/off depending on bather load? 

DrJeff - can you share a schematic/drawing of your pool bodies, valves, filters, etc?  I think that would help.

DrJeff, 2021-04-28:
> The challenge here is that @DrJeff's pool is neither a shared body nor a dual body. A shared body can have intake/return valves and be in pool, spa or spillway mode. The dual body systems can [...]

lol 🤣 

> 
> DrJeff has pair of intake/returns in that it keeps the body separate or creates a spillway. Also, it sounds like one filter can be included/excluded on/off depending on bather load?

There are 2 filters main one is a DE the secondary for the spa is a cartridge
Under Normal operations non spa use both bodies connected via spillway so I can use SWG to chlorinate both. When Optimus Prime is over we can use separate chlorine injection non monitored system for now just small feed keep bather load under control and still getting filtered with its own filter. And the pool will be operating the whole time. 

> 
> DrJeff - can you share a schematic/drawing of your pool bodies, valves, filters, etc? I think that would help.

I will try to get a pic of it tomorrow but to put it simple the heater has a bypasses 3 valves to force the water from pool around it and the spa through it. That pump (heater for spa) is used for one bank of jets in spa and there is another pump for the other bank of jets in the spa. The spa return (spillway) from the pool body is on the bottom of the spa which can be switched on demand.

rstrouse, 2021-04-28:
Alright so pull njspc.  Funny my test scenario had all the bits aligned in a way that it just worked.  When I replicated your Transformer it all went to hell and for the tiniest of reasons to boot.   Turns out we were still relying on the pump id to determine whether it was to be controlled by Nixie.

I am still interested in how you are keeping water levels where the dual bodies are connected yet not connected.  If you draw water out of one body and it doesn't make it to the place where it came from, you empty one and overflow the other.

DrJeff, 2021-04-28:
Ok pulled it!
Now I see pump messages but failing to communicate.
```
1|poolController  | [4/28/2021, 9:44:58 AM] info: {"id":712,"valid":true,"dir":"out","proto":"pump","pkt":[[],[255,0,255],[165,0,96,33,10,4],[2,196,13,122],[2,129]],"ts":"2021-04-28T09:44:58.031-0700"}
1|poolController  | [4/28/2021, 9:44:58 AM] verbose: Wrote packet [255,0,255,165,0,96,33,10,4,2,196,13,122,2,129].  Retries remaining: 1
1|poolController  | [4/28/2021, 9:44:58 AM] info: {"id":713,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,33,96,255,1],[1],[2,39]],"ts": "2021-04-28T09:44:58.086-0700"}
1|poolController  | [4/28/2021, 9:44:59 AM] silly: Retrying outbound message after 1.06secs with 1 attempt(s) left. - 165,0,96,33,10,4,2,196,13,122,2,129
1|poolController  | [4/28/2021, 9:44:59 AM] info: {"id":712,"valid":true,"dir":"out","proto":"pump","pkt":[[],[255,0,255],[165,0,96,33,10,4],[2,196,13,122],[2,129]],"ts":"2021-04-28T09:44:59.091-0700"}
1|poolController  | [4/28/2021, 9:44:59 AM] verbose: Wrote packet [255,0,255,165,0,96,33,10,4,2,196,13,122,2,129].  Retries remaining: 0
1|poolController  | [4/28/2021, 9:44:59 AM] info: {"id":714,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,33,96,255,1],[1],[2,39]],"ts": "2021-04-28T09:44:59.127-0700"}
1|poolController  | [4/28/2021, 9:45:00 AM] silly: Retrying outbound message after 1.042secs with 0 attempt(s) left. - 165,0,96,33,10,4,2,196,13,122,2,129
1|poolController  | [4/28/2021, 9:45:00 AM] warn: Message aborted after 2 attempt(s): 165,0,96,33,10,4,2,196,13,122,2,129
1|poolController  | [4/28/2021, 9:45:00 AM] error: Message aborted after 2 attempt(s): 165,0,96,33,10,4,2,196,13,122,2,129
1|poolController  | [4/28/2021, 9:45:00 AM] silly: Heater Solar is associated with Pool
1|poolController  | [4/28/2021, 9:45:00 AM] debug: Heater Type: solar Mode:off Temp: 55 Setpoint: 91 Status: 0
1|poolController  | [4/28/2021, 9:45:00 AM] silly: Heater Gas Heater is not associated with Pool
1|poolController  | [4/28/2021, 9:45:00 AM] silly: MQTT send:
1|poolController  | topic: virtual-controller/state/time
1|poolController  | message: 2021-04-28T09:45:00.457-0700
1|poolController  | opts:{"retain":true,"qos":0}
1|poolController  | [4/28/2021, 9:45:00 AM] info: {"id":715,"valid":true,"dir":"out","proto":"pump","pkt":[[],[255,0,255],[165,0,96,33,4,1],[255],[2,42]],"ts":"2021-04-28T09:45:00.637-0700"}
1|poolController  | [4/28/2021, 9:45:00 AM] verbose: Wrote packet [255,0,255,165,0,96,33,4,1,255,2,42].  Retries remaining: 1
1|poolController  | [4/28/2021, 9:45:00 AM] info: {"id":716,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,33,96,4,1],[255],[2,42]],"ts": "2021-04-28T09:45:00.687-0700"}
1|poolController  | [4/28/2021, 9:45:01 AM] info: {"id":717,"valid":true,"dir":"out","proto":"pump","pkt":[[],[255,0,255],[165,0,96,33,6,1],[10],[1,55]],"ts":"2021-04-28T09:45:01.190-0700"}
1|poolController  | [4/28/2021, 9:45:01 AM] verbose: Wrote packet [255,0,255,165,0,96,33,6,1,10,1,55].  Retries remaining: 1
1|poolController  | [4/28/2021, 9:45:01 AM] info: {"id":718,"valid":true,"dir":"in","proto":"pump","pkt":[[],[255,0,255],[165,0,33,96,6,1],[10],[1,55]],"ts": "2021-04-28T09:45:01.227-0700"}
1|poolController  | [4/28/2021, 9:45:03 AM] info: {"id":719,"valid":true,"dir":"ou
```

rstrouse, 2021-04-28:
It is definitely sending valid messages to the pump but not getting a response.  Are your comms working?  Do you know how to inspect the pump messages in messageManager?

DrJeff, 2021-04-28:
Ok thats what I thought but to the second part Nope, is it in the wiki?
[image]

rstrouse, 2021-04-28:
Ok, thanks for that.  It appears there is model confusion on the pump.  It is thinking that it is a different pump model and sending a 10 (which sets flow) instead of a 1 (which sets rpm).   Your pump is appropriately responding with hey dummy I don't do the flow thingy.

pull njspc the correct message signature should be sent now.

DrJeff, 2021-04-28:
Oh you guys are helping me more! Pulled it, now pump went incognito no talking.
[image]
[image]

rstrouse, 2021-04-28:
Say what!  Restart njspc.  Then run a replay while turning on the pool circuit.  Do not have it capture the configuration reload or you will be setting it all back up again.

DrJeff, 2021-04-28:
Yes I found the intellibrite, did you program the color changes yet I get the relay on but not the cycling of the relay for the color/mode changes.

Ok that still went bad it changed to Nixie Single Body, and yes I did't turn on capture until after restart.
[replay11.zip](https://github.com/tagyoureit/nodejs-poolController/files/6393817/replay11.zip)

DrJeff, 2021-04-28:
It's like everthing is there but Controller has to be selected again and all circuits, features have to be resaved

But pump still no go
Should I kill my data files State and Config and do a clean setup?

rstrouse, 2021-04-28:
> Yes I found the intellibrite, did you program the color changes yet I get the relay on but not the cycling of the relay for the color/mode changes.

Yes it should be sending the proper number of on/offs.  That is if you have an updated REM.

Uggh.  I suggest you set it up from scratch.   Boo.  I need to kill that function for anything but EasyTouch and IntelliTouch.

DrJeff, 2021-04-28:
> Yes it should be sending the proper number of on/offs. That is if you have an updated REM.

I will make sure i pulled it but it should be

> Uggh. I suggest you set it up from scratch. Boo. I need to kill that function for anything but EasyTouch and IntelliTouch.

No biggie i will set up and the just send the log file

rstrouse, 2021-04-28:
There is a checkbox when you click capture replay that you should uncheck in your case.  Unfortunately, we can't seem to get folks with the IntelliTouch and EasyTouch to check it for us when they give us a replay so we check it by default.   For IntelliCenter it doesn't matter because it tells us about all the equipment, including the things that are not there.

rstrouse, 2021-04-28:
On the IntelliBrite thing let us know if the timing is too fast.  I think we have it matching pretty close to what it should be but it's hard to tell if we have the on/off speed too fast for the transformer.   If you go too slow the delay is horrendous.

DrJeff, 2021-04-28:
> There is a checkbox when you click capture replay that you should uncheck in your case. Unfortunately, we can't seem to get folks with the IntelliTouch and EasyTouch to check it for us when they [...]

I definitely can uncheck that! No problem. It's working now (it's not reporting status under pump header) got to add all the rest of the goodies
[Uploading replay13.zip…]()

DrJeff, 2021-04-28:
> On the IntelliBrite thing let us know if the timing is too fast. I think we have it matching pretty close to what it should be but it's hard to tell if we have the on/off speed too fast for the transformer. If you go too slow the delay is horrendous.

I have the latest REM are you sure the code is there all I have is switching the relay on once even if color wheel selected and mode or color no toggling of relay

rstrouse, 2021-04-28:
I'll have a look at your relay.  Pull njspc I found another reference for the pump status where it was reliant on the id and not the address.

DrJeff, 2021-04-28:
Oops! lost pump again stopped njspc, did a `git pull` then restarted njspc
[replay14.zip](https://github.com/tagyoureit/nodejs-poolController/files/6394355/replay14.zip)

rstrouse, 2021-04-28:
Alright so I looked at the light relay.  You have an MCP23017 base relay correct?  I checked out the mockup and I got the correct number of flashes for it.  It should only send this when you change the theme by clicking on the palette icon and choosing a theme.  Is that what you are doing?

Also is you uncheck show as feature then it will put those into the lights section.

DrJeff, 2021-04-28:
> Alright so I looked at the light relay. You have an MCP23017 base relay correct? 

Yes

> I checked out the mockup and I got the correct number of flashes for it. It should only send this when you change the theme by clicking on the palette icon and choosing a theme. Is that what you are doing?

Yes, but doesn't change relays this is mocked up until functional 

> 
> Also is you uncheck show as feature then it will put those into the lights section.

The Checkbox can be unchecked but when saved just rechecks

rstrouse, 2021-04-28:
Can you take another look at the message manager for me related to the pumps?  All indications are that this should be doing its thing.

As for the relay.  I think it may just be too fast through the cycle for it.  I'll add some delays for that relay type.

DrJeff, 2021-04-28:
> Can you take another look at the message manager for me related to the pumps? All indications are that this should be doing its thing.
> 

Nope it's blank and yes I selected the top right to start it

> As for the relay. I think it may just be too fast through the cycle for it. I'll add some delays for that relay type.

Oh for this MCP20317 probably needs 1 sec delays long but thats what I was doing for the previous setup

rstrouse, 2021-04-28:
> Oh for this MCP20317 probably needs 1 sec delays long but thats what I was doing for the previous setup

Wow that is really slow.  To set white that would mean 14 pairs + an on at the beginning and end so you would be looking at 30 seconds to change the theme.  Holy yikes I thought 100ms on my setup was slow but it works.

DrJeff, 2021-04-28:
> Wow that is really slow. To set white that would mean 14 pairs + an on at the beginning and end so you would be looking at 30 seconds to change the theme. Holy yikes I thought 100ms on my setup was [...]

Yup I took it down to 250ms but had issues upped it to 750ms mostly stable but absolutely solid at 1000ms, the delay was ok because I don't do much change. let's do 500ms and try?

DrJeff, 2021-04-28:
As for the pump thing I tried pulling the USB and it finds the connection but no messages are generated like earlier....
[image]

rstrouse, 2021-04-28:
I'll add some code to allow you to do a cycle delay.  That is horrible.  My relay transitions are capable of around 500/sec but the lights themselves or the transformer isn't that quick.

Still, can't seem to replicate the pump refusal.  In fact the message are indicating that it is attempting to make a run at the pump.   Can you stop njspc and copy the poolState.json that I included here over your poolState.json.

[poolState.zip](https://github.com/tagyoureit/nodejs-poolController/files/6394583/poolState.zip)

DrJeff, 2021-04-28:
You da Man! that worked

rstrouse, 2021-04-28:
I removed an artifact for the pump state that was created by the reload.  Near as I can figure it seems that it didn't like an inactive pump state hanging out in there.  I am hardening the code right now to make sure it cleans those up on failure.

rstrouse, 2021-04-28:
Pull njspc and REM.  I added the ability to add additional time to the relay for sequences.  Don't forget to click on the save device after making your changes.
[image]

DrJeff, 2021-04-28:
Ok in your ways to fixing the pump is the chlorinator having the same id problem?

[16 more replies]

closed 2021-06-16 by tagyoureit