tagyoureit/nodejs-poolController #275: njspc <-> REM socket timeouts

johnny2678, 2021-03-27:
Hi @rstrouse ,
Any updates on the socket.io updates in njspc?

I still get random disconnects and the only way to re-establish is to restart njspc.  Not sure how to convey that but I did find this in the logs this morning.

```
1|poolController  | [3/27/2021, 10:17:44 AM] verbose: REM server request initiated. GET /status/device/i2c:1:2 
1|poolController  | [3/27/2021, 10:17:44 AM] verbose: REM server request returned. GET /status/device/i2c:1:2 
1|poolController  | [3/27/2021, 10:17:45 AM] verbose: Chem mixing acid remaining: 42sec
1|poolController  | [3/27/2021, 10:17:46 AM] verbose: Chem mixing acid remaining: 41sec
1|poolController  | [3/27/2021, 10:17:47 AM] info: [10:17:47 AM] 192.168.5.140 GET /state/all/ {}
1|poolController  | [3/27/2021, 10:17:47 AM] verbose: Chem mixing acid remaining: 40sec
1|poolController  | [3/27/2021, 10:17:48 AM] error: Error Sending REM Request: GET /status/device/i2c:1:2 Request timeout
1|poolController  | [3/27/2021, 10:17:48 AM] verbose: REM server request initiated. GET /status/device/gpio:1:36 
1|poolController  | [3/27/2021, 10:17:48 AM] verbose: REM server request returned. GET /status/device/gpio:1:36 
1|poolController  | [3/27/2021, 10:17:48 AM] verbose: Chem mixing acid remaining: 39sec
1|poolController  | [3/27/2021, 10:17:49 AM] verbose: Chem mixing acid remaining: 38sec
1|poolController  | [3/27/2021, 10:17:50 AM] verbose: Chem mixing acid remaining: 37sec
1|poolController  | [3/27/2021, 10:17:52 AM] verbose: Chem mixing acid remaining: 36sec
1|poolController  | [3/27/2021, 10:17:52 AM] verbose: REM server request initiated. PUT /feed/device/i2c:1:2 {"tempF":80}
1|poolController  | [3/27/2021, 10:17:52 AM] verbose: REM server request returned. PUT /feed/device/i2c:1:2 {"tempF":80}
1|poolController  | [3/27/2021, 10:17:53 AM] verbose: Chem mixing acid remaining: 35sec
1|poolController  | [3/27/2021, 10:17:53 AM] error: Error sending Request: PUT /feed/device/i2c:1:2 read ECONNRESET
1|poolController  | [3/27/2021, 10:17:54 AM] verbose: Chem mixing acid remaining: 34sec
1|poolController  | [3/27/2021, 10:17:55 AM] verbose: Chem mixing acid remaining: 33sec
1|poolController  | [3/27/2021, 10:17:56 AM] verbose: Chem mixing acid remaining: 32sec
1|poolController  | [3/27/2021, 10:17:57 AM] verbose: Chem mixing acid remaining: 31sec
1|poolController  | [3/27/2021, 10:17:58 AM] verbose: Chem mixing acid remaining: 30sec
1|poolController  | [3/27/2021, 10:17:59 AM] verbose: Chem mixing acid remaining: 29sec
1|poolController  | [3/27/2021, 10:18:00 AM] verbose: Chem mixing acid remaining: 28sec
1|poolController  | [3/27/2021, 10:18:02 AM] verbose: Chem mixing acid remaining: 27sec
```

Usually I know when it's time to reset because acid dosing stops even though pH is above the threshold (it's NOT up against the daily acid dosing max) AND when the pH measurements look like they are coming in less frequently.

An njspc restart clears this up but then it happens again every 12-36 hours-ish.

Any thing else I can provide to help troubleshoot?  I'm on next 6.5.2

johnny2678, 2021-03-27:
just restarted again and the hang up happened after only 5 minutes and didn't re-establish.  I was actually watching the njspc logs when I caught this:

```
1|poolController  | [3/27/2021, 11:58:55 AM] verbose: REM server request initiated. PUT /state/device/gpio:1:36 {"state":false}
1|poolController  | [3/27/2021, 11:58:55 AM] verbose: REM server request returned. PUT /state/device/gpio:1:36 {"state":false}
1|poolController  | [3/27/2021, 11:58:56 AM] error: Error sending Request: PUT /state/device/gpio:1:36 socket hang up
1|poolController  | [3/27/2021, 11:58:59 AM] verbose: REM server request initiated. GET /status/device/i2c:1:1 
1|poolController  | [3/27/2021, 11:58:59 AM] verbose: REM server request returned. GET /status/device/i2c:1:1 
1|poolController  | [3/27/2021, 11:58:59 AM] error: Error sending Request: GET /status/device/i2c:1:1 socket hang up
1|poolController  | [3/27/2021, 11:59:03 AM] info: [11:59:03 AM] 192.168.5.140 GET /state/all/ {}
1|poolController  | [3/27/2021, 11:59:18 AM] info: [11:59:18 AM] 192.168.5.140 GET /state/all/ {}
```

tagyoureit, 2021-03-27:
Have you pulled the latest code?  @rstrouse added code to catch a reject ~2 weeks ago and I haven't had a single timeout since!

johnny2678, 2021-03-27:
ok, not sure why I didn't just try and pull ;) 

updated njspc, dashPanel, and REM to the latest.  It's dosed once already without losing connection... will keep an eye on it.

Only odd thing I see on dashPanel is I now have some new, blank features that aren't listed in setup:

[image]

tagyoureit, 2021-03-27:
What's in your `poolConfig.json`?  Can you post your config files?

tagyoureit, 2021-03-27:
Ok, weird.  I just did a pull and also have an extra circuit.  Will have to see what @rstrouse messed up... ;)

[image]

johnny2678, 2021-03-27:
@rstrouse can't catch a break.  The two of you have cooked up a distributed, open source masterpiece and then wonks like me show up with tiny little nits 😂

the circuits and the features sections look the same to me.  Don't see any of those extras.

```
  "circuits": [
    {
      "id": 1,
      "master": 0,
      "name": "Spa",
      "nameId": 72,
      "type": 1,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 360,
      "dontStop": false
    },
    {
      "id": 2,
      "master": 0,
      "name": "Lights",
      "nameId": 47,
      "type": 16,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 240,
      "dontStop": false
    },
    {
      "id": 3,
      "master": 0,
      "name": "Air Blower",
      "nameId": 2,
      "type": 0,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 60,
      "dontStop": false
    },
    {
      "id": 4,
      "master": 0,
      "name": "High Speed",
      "nameId": 43,
      "type": 0,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 45,
      "dontStop": false
    },
    {
      "id": 5,
      "master": 0,
      "name": "AUX 4",
      "nameId": 6,
      "type": 0,
      "isActive": true,
      "freeze": false,
      "showInFeatures": false,
      "eggTimer": 720,
      "dontStop": false
    },
    {
      "id": 6,
      "master": 0,
      "name": "Pool",
      "nameId": 61,
      "type": 2,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 960,
      "dontStop": false
    }
  ],
  "features": [
    {
      "id": 11,
      "master": 0,
      "name": "Pool Low",
      "nameId": 64,
      "type": 0,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 720,
      "dontStop": false
    },
    {
      "id": 12,
      "master": 0,
      "nameId": 206,
      "type": 0,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 720,
      "dontStop": false,
      "name": "Pool Lower"
    },
    {
      "id": 13,
      "master": 0,
      "name": "Pool Lowest",
      "nameId": 200,
      "type": 0,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 720
    },
    {
      "id": 15,
      "master": 0,
      "name": "Pool High",
      "nameId": 202,
      "type": 0,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 45,
      "dontStop": false
    },
    {
      "id": 16,
      "master": 0,
      "name": "Pool Higher",
      "nameId": 203,
      "type": 0,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 45,
      "dontStop": false
    },
    {
      "id": 20,
      "master": 0,
      "name": "AUX EXTRA",
      "nameId": 93,
      "type": 0,
      "isActive": true,
      "freeze": false,
      "showInFeatures": false,
      "eggTimer": 720,
      "dontStop": false
    }
  ],

```
full file attached (changed to .txt)

[poolConfig.txt](https://github.com/tagyoureit/nodejs-poolController/files/6216175/poolConfig.txt)

tagyoureit, 2021-03-28:
A while back we were getting heater data before circuit data and that caused a problem, so we reversed the order.  I see why my extra circuit appeared; it was actually the solar (aux extra) circuit but we don't disable it (mark it invalid) until after we get the heater data.  Going through all of this, I did see another place where we would remove the circuits data from poolConfig.json but not poolState.json.  All this to say, you may see some extra circuits briefly during the initialization but they should disappear and not come back.  Pull and try again.  If it still happens, attach a packet capture and I'll go through it more closely.

johnny2678, 2021-03-28:
yea, I'm still seeing them.

logs attached.

not sure if it's related to this issue but the last pull cleared out my body sizes and my acid tank capacity so I had a bunch of errors at startup.  Once I re-entered in setup pH dosing seemed to return to normal but now this is repeating in the console:

```
1|poolController  | failed on alarm: flow/@bind=data.alarms.flow.val;
1|poolController  | failed on chemController:alarm: flow desc/@bind=data.alarms.flow.desc; {"id":50,"saturationIndex":0.036,"flowDetected":true,"orp":{"probe":{"saltLevel":0,"level":732.9},"tank":{"capacity":0,"level":0,"units":{"val":1,"name":"gal","desc":"Gallons"},"alarmEmptyEnabled":false,"alarmEmptyLevel":20},"dosingTimeRemaining":0,"delayTimeRemaining":0,"dosingVolumeRemaining":0,"doseVolume":0,"doseTime":0,"lockout":false,"level":732.9,"mixTimeRemaining":0,"dailyLimitReached":false,"manualDosing":false,"flowDelay":false,"dosingStatus":{"val":1,"name":"monitoring","desc":"Monitoring"},"enabled":true,"dailyVolumeDosed":0,"pump":{"isDosing":false},"setpoint":720},"ph":{"probe":{"temperature":81,"level":7.598,"tempUnits":{"val":0,"name":"F","desc":"Fahrenheit"}},"tank":{"capacity":3.5,"level":3.0944623340757995,"units":{"val":1,"name":"gal","desc":"Gallons"},"alarmEmptyEnabled":true,"alarmEmptyLevel":20},"dosingTimeRemaining":0,"delayTimeRemaining":0,"dosingVolumeRemaining":0,"doseVolume":23,"doseTime":12,"lockout":false,"level":7.598,"mixTimeRemaining":148,"dailyLimitReached":false,"manualDosing":false,"flowDelay":false,"dosingStatus":{"val":2,"name":"mixing","desc":"Mixing"},"enabled":true,"dailyVolumeDosed":1481,"demand":95,"pump":{"isDosing":false},"setpoint":7.57,"volumeDosed":0},"flowSensor":{},"type":{"val":4,"name":"rem","desc":"REM Chem","ph":{"min":6.8,"max":8},"hasAddress":false},"alarms":{"comms":{"val":0,"name":"ok","desc":"Ok"},"flowSensorFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"orpProbeFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"chlorFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"orpPumpFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"pHProbeFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"pHPumpFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"bodyFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"orpTank":{"val":0,"name":"ok","desc":"Ok - No alarm"},"orp":{"val":0,"name":"ok","desc":"Ok - No alarm"},"pHTank":{"val":0,"name":"ok","desc":"Ok - No alarm"},"pH":{"val":0,"name":"ok","desc":"Ok - No alarm"}},"status":{"val":0,"name":"ok","desc":"Ok"},"lastComm":1616934672626,"isBodyOn":true,"warnings":{"orpDailyLimitReached":{"val":0,"name":"ok","desc":"Ok - No limits reached"},"chlorinatorCommError":{"val":0,"name":"ok","desc":"Ok - No Warning"},"pHLockout":{"val":0,"name":"ok","desc":"Ok - No limits reached"},"pHDailyLimitReached":{"val":0,"name":"ok","desc":"Ok - No limits reached"},"waterChemistry":{"val":0,"name":"ok","desc":"Ok - No Warning"}},"name":"REM ChemControl","isActive":true}
1|poolController  | failed on alarm: flow/@bind=data.alarms.flow.val;
1|poolController  | failed on chemController:alarm: flow desc/@bind=data.alarms.flow.desc; {"id":50,"saturationIndex":0.036,"flowDetected":true,"orp":{"probe":{"saltLevel":0,"level":732.9},"tank":{"capacity":0,"level":0,"units":{"val":1,"name":"gal","desc":"Gallons"},"alarmEmptyEnabled":false,"alarmEmptyLevel":20},"dosingTimeRemaining":0,"delayTimeRemaining":0,"dosingVolumeRemaining":0,"doseVolume":0,"doseTime":0,"lockout":false,"level":732.9,"mixTimeRemaining":0,"dailyLimitReached":false,"manualDosing":false,"flowDelay":false,"dosingStatus":{"val":1,"name":"monitoring","desc":"Monitoring"},"enabled":true,"dailyVolumeDosed":0,"pump":{"isDosing":false},"setpoint":720},"ph":{"probe":{"temperature":81,"level":7.598,"tempUnits":{"val":0,"name":"F","desc":"Fahrenheit"}},"tank":{"capacity":3.5,"level":3.0944623340757995,"units":{"val":1,"name":"gal","desc":"Gallons"},"alarmEmptyEnabled":true,"alarmEmptyLevel":20},"dosingTimeRemaining":0,"delayTimeRemaining":0,"dosingVolumeRemaining":0,"doseVolume":23,"doseTime":12,"lockout":false,"level":7.598,"mixTimeRemaining":147,"dailyLimitReached":false,"manualDosing":false,"flowDelay":false,"dosingStatus":{"val":2,"name":"mixing","desc":"Mixing"},"enabled":true,"dailyVolumeDosed":1481,"demand":95,"pump":{"isDosing":false},"setpoint":7.57,"volumeDosed":0},"flowSensor":{},"type":{"val":4,"name":"rem","desc":"REM Chem","ph":{"min":6.8,"max":8},"hasAddress":false},"alarms":{"comms":{"val":0,"name":"ok","desc":"Ok"},"flowSensorFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"orpProbeFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"chlorFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"orpPumpFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"pHProbeFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"pHPumpFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"bodyFault":{"val":0,"name":"ok","desc":"Ok - No Faults"},"orpTank":{"val":0,"name":"ok","desc":"Ok - No alarm"},"orp":{"val":0,"name":"ok","desc":"Ok - No alarm"},"pHTank":{"val":0,"name":"ok","desc":"Ok - No alarm"},"pH":{"val":0,"name":"ok","desc":"Ok - No alarm"}},"status":{"val":0,"name":"ok","desc":"Ok"},"lastComm":1616934672626,"isBodyOn":true,"warnings":{"orpDailyLimitReached":{"val":0,"name":"ok","desc":"Ok - No limits reached"},"chlorinatorCommError":{"val":0,"name":"ok","desc":"Ok - No Warning"},"pHLockout":{"val":0,"name":"ok","desc":"Ok - No limits reached"},"pHDailyLimitReached":{"val":0,"name":"ok"," [... cut ...] k - No Warning"}},"name":"REM ChemControl","isActive":true}
1|poolController  | failed on alarm: flow/@bind=data.alarms.flow.val;

```
Going to try one more restart to see if it clears... nope. still there.

[replay (2).zip](https://github.com/tagyoureit/nodejs-poolController/files/6217282/replay.2.zip)
[poolConfig.txt](https://github.com/tagyoureit/nodejs-poolController/files/6217296/poolConfig.txt)
[poolState.txt](https://github.com/tagyoureit/nodejs-poolController/files/6217297/poolState.txt)

johnny2678, 2021-03-29:
NICE, how'd the flow cell install go @rstrouse ?

don't recall changing anything on the influx binding for flow alarms but it looks like it cleared today so you can disregard.

rstrouse, 2021-03-29:
It was a comedy and a tragedy all rolled into one.  I decided that I was going to thread the intake into a T that returns from the solar before it goes into the heater.  So I got my handy dandy 3/8" NPT tap out and proceeded to drill into the fitting.  Then as if I were standing on a thin sheet of ice I hear a crack.  Near as I can figure the T had a whole lot of stress on it and the pipe in the fitting cracked.  

So the whole morning yesterday was cutting out the old plumbing between the heater and the filter.  The whole run was off by misaligned by over 6 inches and the reason for the crack.  I wanted to hang the plumber but that was a good 11 years ago when he put it in.

So after repairing the plumbing,  I installed the flow cell.  The fittings all worked like a charm, the crack was gone and it appeared like I was in business.  However, the Atlas pH and ORP probes are not behaving.  I soldered on some SMA connectors so there must be something in those that isn't correct.  So this afternoon I will pull the probes back out again and redo the connectors.  I know the pH is capable of reading because I also have a lab grade probe and it worked like a charm.

Your alarms would have reset the first time it processed alarms.  Once the values are set it no longer tries to bind against an undefined flow alarm.

johnny2678, 2021-03-29:
> So after repairing the plumbing, I installed the flow cell. The fittings all worked like a charm, the crack was gone and it appeared like I was in business. However, the Atlas pH and ORP probes are [...]

Oof on the crack - that was my biggest hangup on this whole DIY endeavor was 'what do I do if I spring a leak or break the plumbing'?  I bought all kinds of PVC connectors, sleeves, extra 2" pipe etc to have on hand just in case.  I took 2 years thinking it out with diagrams, drawings, project plans, etc. and then I did it all on one afternoon 😂

Those atlas guys get fussy about soldering /extending their probe wires in their datasheets.  They push their SMA extenders, which I did use when a recently moved my project box and they worked great.
https://atlas-scientific.com/cables/3-meter-sma-male-to-sma-female-extension-cable/

rstrouse, 2021-03-29:
Yeah but the industrial probes come with bare wires and no connectors.  Soldering on an SMA connector isn't rocket science.  You just need a good solder station because of the heat mass and a whole lot of flux.  Maybe I didn't get the connector clean enough.

johnny2678, 2021-03-29:
ahh didn't know that about the industrial probes.  Good thing I'm cheap because my soldering skills are terrible. Guess you can strip and try again?  Atlas support is pretty knowledgable too - they should be for the price of those industrial probes.

rstrouse, 2021-03-30:
Turns out I just needed a bit more solder heat and thorough cleaning of the flux and the probes came to life.  I think the issue was flux in the connector.

johnny2678, 2021-03-31:
glad to say that the REM <-> njspc connection has been rock solid for 2 days now!  Still have the extra blank features listed in dashPanel but that's not a big deal.  Might try and do a packet capture at some point but I'll start another issue.  Thanks!

[image]

closed 2021-03-31 by johnny2678