tagyoureit/nodejs-poolController #276: How to adjust Pump speed via MQTT (or via API)

theRealMCQwerty, 2021-03-27:
Trying to figure out how to change the RPM of my InteliFlo XF VSF pump via MQTT.
I know that pumps speed is based on which PumpCircuit is active, se really my question is "how do I change the properties of a Pump Circuit via the API?"

Obviously this can easily be achieved via DashPanel from the pump settings be ensuring an associated FeatureCircuit is turned on and then adjusting the flow rate directly so I am just missing where this feature is hiding in the API and hence how to control it via MQTT.

**Pool Equipment**
 - Controller: IntelliCenter
 - Controller Model: IntelliCenter i8PS
 - Controller Firmware Version: 1.047
 - Pump(s) manufacturer and model: IntelliFlo XF VSF 

Thanks

DrJeff, 2021-03-28:
You need to setup Features and assign them to pump speeds
[image]
[image]
Then you can call the feature to activate that speed/rpm
`virtual-controller/state/circuits/setState` with a payload to the feature you are switching on or off `{"id":8, "isOn":"off"}`

tagyoureit, 2021-03-28:
Does that solution work for you?  Intellicenter should have a ton of free circuits you can assign different speeds to and then trigger those.  There is a direct API for changing the speed associated with a pump circuit but it isn't a great API for MQTT because you'd need to send all of the data for all of the pump circuits.

theRealMCQwerty, 2021-03-28:
@tagyoureit unfortunately it doesn't. I have 2 pumps and 24 actuated valve amongst other things so I am already out of feature circuits. I need to be able to send an absolute value for a given pump in the same way as you can adjust the GPM or RMP value of an already assigned pump circuit via the DashPanel UI.
I don't mind how messy it is, I'm sure I could work with it I just can't figure it out. Could you provide me with an example of what is available now?
Also, is it a no-go for MQTT or just something that would need implementing (understanding the complexity)? I can work with just the API but MQTT is a bit tidier via Home Assistant.

Thanks!

rstrouse, 2021-03-29:
The problem is that unless the configuration is changed, IntelliCenter will override it.  There is no way to take pump control away from IntelliCenter.  Every 2 seconds of so it will send an overriding RPM request directly to the pump that is the speed for the fastest currently running circuit.  

The only way to change the speed of a pump is to send the pump circuit message to IntelliCenter (which would modify the circuits attached to the pump) and let IntelliCenter sort it out.  When you click on the pump in dashPanel that is what it does is send the pump circuit message that includes all circuits assigned to the pump.  We could add an MQTT endpoint to do that but it would still be at the mercy of IntelliCenter.

theRealMCQwerty, 2021-03-29:
@rstrouse thanks for the info. I plan on only having one pump circuit and then adjusting the speed of that circuit so that I have 'control' over the speed and don't get overridden by another circuit.
I would turn the circuit off whenever I wanted the pump completely off and whenever I wanted the pump on at a specific speed I would turn the circuit on and provide the speed I wanted. I wouldn't just turn the circuit on because I wouldn't rely on an preset speed still being there, I would always provide a speed.

Would that work? I see that there needs to be a lot of caveats and warnings with this functionality but it would be great if I could achieve the control described above.

tagyoureit, 2021-03-29:
I was in the middle of writing this, but now have to change my response.  :)

You can change the pump circuit(s) at the [/config/pump](https://tagyoureit.github.io/nodejs-poolcontroller-api/#/config/put_config_pump_) endpoint.  As long as you only have one pump circuit, it won't be a problem.  If you have multiple (some point in the future), but only send one pump circuit it will delete the rest.  This endpoint is currently designed as an all-or-nothing approach to adjusting these types of circuits.

I only say (now, said) it isn't great for MQTT because it's mostly designed for small, fast messages.  Some MQTT brokers have a very small limit; others allow you to go up to 256mb full payload capacity.  Since you are only using one pump circuit this is likely moot at this point, anyhow.

theRealMCQwerty, 2021-03-29:
Excellent, now it all makes sense. I should have been able to figure this out by watching the events in NJSPC while changing the speed in DashPanel 🤦‍♂️ but now @tagyoureit and @rstrouse have explained what is going on it has become clear 😄.

I can work with API for now but if there were scope for an MQTT option in the future that would be great.
Thanks!

closed 2021-06-16 by tagyoureit