tagyoureit/nodejs-poolController #279: Unable to create new features or add a schedule entry

2delarosa, 2021-04-15:
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6321595/replay.zip)
This started in 6.x where I was unable to create a new feature entry.  This will be brief since I now lost all my feature settings.  Going to reboot to see if the settings recover.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch dashPanel
2. Click on Settings
2. Click on Circuits
3. Click on Features
4. Create and feature and try saving it.
4. See error

**Expected behavior**
New feature entry
**Screenshots**
[image]

If applicable, add screenshots to help explain your problem.

**Packet Capture**
Follow the instructions to complete a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) and attach the resulting zip/log files.

**Pool Equipment**
 - Controller: Virtual
 - Pump(s) manufacturer and model: [e.g. IntelliFlow 2 VST 011056]
 - Heater(s): [e.g. gas, ]
 - Valves: [e.g. 3rd party]
 - Any other relevant equipment:

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

2delarosa, 2021-04-16:
Here's a screenshot when I try to create a schedule entry

[image]

rstrouse, 2021-04-16:
Yeah you won't be able to create a schedule from there just yet.  I have a model of this working but it needs some time to coalesce.  Can I get you to upload the poolConfig.json and poolState.json that you currently have?

2delarosa, 2021-04-16:
Sure.  See attached.
[poolConfig.json.txt](https://github.com/tagyoureit/nodejs-poolController/files/6321813/poolConfig.json.txt)
[poolState.json.txt](https://github.com/tagyoureit/nodejs-poolController/files/6321814/poolState.json.txt)

2delarosa, 2021-04-16:
I copied the config.json from the nodejs-poolController folder and renamed it so I could upload it.

[nsjcPCconfig.json.txt](https://github.com/tagyoureit/nodejs-poolController/files/6327120/nsjcPCconfig.json.txt)

2delarosa, 2021-04-17:
Attached is the latest npm install log output

[npm install log.txt](https://github.com/tagyoureit/nodejs-poolController/files/6330157/npm.install.log.txt)

rstrouse, 2021-04-17:
It looks like serial port is trying to target node 14 when you type node --version you get 12.x correct?

2delarosa, 2021-04-17:
node --version is v14.16.1

As requested, I've attached the poolConfig.json and poolState.json files

[poolConfig.json.mar.txt](https://github.com/tagyoureit/nodejs-poolController/files/6330249/poolConfig.json.mar.txt)
[poolConfig.json.txt](https://github.com/tagyoureit/nodejs-poolController/files/6330250/poolConfig.json.txt)
[poolState.json.txt](https://github.com/tagyoureit/nodejs-poolController/files/6330251/poolState.json.txt)

rstrouse, 2021-04-18:
Why did you upgrade to node 14.x?  That is why serial port isn't compiling.   14 was just moved to LTS and they haven't created binaries for all their libraries yet.  I wish they wouldn't do that.  You may need to downgrade to 12.x.

Anyhow before you downgrade unzip the two files contained in this file to your data directory and lets see if we get a different result.
[poolState.zip](https://github.com/tagyoureit/nodejs-poolController/files/6330274/poolState.zip)

2delarosa, 2021-04-18:
I don't recall upgrading to node 14.x unless rPI OS or Openhab 3.0 required it.  Is there a history where one can find out when it was upgraded to 14.x.  Please post a link for the downgrade instructions in case I need to downgrade it

2delarosa, 2021-04-18:
No more multiple pump entries but still no communication with pump.

2delarosa, 2021-04-18:
Okay, I downgraded node js to 12.x, removed the njsPC folder, clone the repository, npm install, npm run rebuild, and pm2 start njsPC.  Still didn't work.  :(

[npm install n12.x log.txt](https://github.com/tagyoureit/nodejs-poolController/files/6330448/npm.install.n12.x.log.txt)

tagyoureit, 2021-04-18:
That looks fine per the install logs.

Can you start with `npm run start` instead of starting with PM2?

2delarosa, 2021-04-18:
I will once I finished updating the webClient.  I didn't want to launch the webClient but I figured it wouldn't hurt to verify the existing behavior.  So I went to the webClient folder and did a 
`git reset --hard`
`git checkout master`
`git pull`
`npm i`
`npm start&`

Still building something

> > nodejs-poolcontroller-webclient@6.1.0 start /home/openhabian/nodejs-poolController-webClient
> npm run cleanup && npm run build && npm run start:cached

> nodejs-poolcontroller-webclient@6.1.0 cleanup /home/openhabian/nodejs-poolController-webClient
> rimraf dist

> nodejs-poolcontroller-webclient@6.1.0 build /home/openhabian/nodejs-poolController-webClient
> tsc && npm run build:parcel

> nodejs-poolcontroller-webclient@6.1.0 build:parcel /home/openhabian/nodejs-poolController-webClient
> parcel build --public-url . web/dashboard/index.html --out-dir dist/web

⠸ Building app.tsx...Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
⠦ Building app.tsx...Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
⠹ Building objectWithoutPropertiesLoose.js...Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
⠇ Building ReactPropTypesSecret.js...Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
⠋ Building react-data-grid.js...Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
⚠️  Could not load existing sourcemap of "../../node_modules/react-data-grid/dist/react-data-grid.js".
⠸ Building index.js...

2delarosa, 2021-04-18:
it took quite some time to start but it has the same expected behavior.  When I click on edit pumps, I get the following error

```
Something went wrong.
Details
TypeError: Cannot read property 'type' of undefined

    in c
    in r
    in div
    in p
    ...
```

Will stop it now and get back to the njsPC folder to run the `npm run start` command

2delarosa, 2021-04-18:
Here's the serial output.  The -dashPanel is still showing an anomaly in the pump section plus I don't see any activity on the serial port.

openhabian@casajuarez:~ $ cd nodejs-poolController
openhabian@casajuarez:~/nodejs-poolController $ npm run start

> nodejs-poolcontroller@7.0.0 start /home/openhabian/nodejs-poolController
> npm run build && node dist/app.js

> nodejs-poolcontroller@7.0.0 build /home/openhabian/nodejs-poolController
> tsc

Init state for Pool Controller
[4/18/2021, 10:10:42 AM] info: Starting up SSDP server
[4/18/2021, 10:10:42 AM] info: Initializing Nixie Controller
[4/18/2021, 10:10:42 AM] info: Nixie Controller Initialized
[4/18/2021, 10:10:42 AM] info: Server is now listening on 0.0.0.0:4200
[4/18/2021, 10:10:42 AM] info: Serial port: /dev/ttyUSB0 request to open successful
[4/18/2021, 10:10:45 AM] info: New socket client connected wB3gaWcz8nzdLt3CAAAA -- 192.168.100.39
[4/18/2021, 10:10:51 AM] info: New socket client connected haiXbEOMjazNbX4oAAAC -- 192.168.99.2
[4/18/2021, 10:10:51 AM] info: [10:10:51 AM] 192.168.99.2 GET /state/all?null {}
[4/18/2021, 10:10:58 AM] info: [10:10:58 AM] 192.168.100.39 GET /state/features {}
[4/18/2021, 10:11:04 AM] info: [10:11:04 AM] 192.168.99.2 GET /state/all?null {}
[4/18/2021, 10:11:04 AM] info: New socket client connected VRW7Az9KZr97sjWNAAAE -- 192.168.99.2
[4/18/2021, 10:11:04 AM] info: [10:11:04 AM] 192.168.99.2 GET /state/all?null {}
[4/18/2021, 10:11:07 AM] info: [10:11:07 AM] 192.168.99.2 PUT /state/circuit/setState {"id":7,"state":true}
[4/18/2021, 10:11:28 AM] info: [10:11:28 AM] 192.168.100.39 GET /state/features {}
[4/18/2021, 10:11:41 AM] info: [10:11:41 AM] 192.168.99.2 GET /state/pump/2?null {}
[4/18/2021, 10:11:58 AM] info: [10:11:58 AM] 192.168.100.39 GET /state/features {}
[4/18/2021, 10:12:28 AM] info: [10:12:28 AM] 192.168.100.39 GET /state/features {}
[4/18/2021, 10:12:58 AM] info: [10:12:58 AM] 192.168.100.39 GET /state/features {}
[4/18/2021, 10:13:02 AM] info: [10:13:02 AM] 192.168.99.2 GET /config/all?null {}
[4/18/2021, 10:13:02 AM] info: [10:13:02 AM] 192.168.99.2 GET /config/options/general?null {}
[4/18/2021, 10:13:05 AM] info: [10:13:05 AM] 192.168.99.2 GET /config/options/pumps?null {}
[4/18/2021, 10:13:15 AM] info: [10:13:15 AM] 192.168.99.2 GET /app/options/interfaces?null {}
[4/18/2021, 10:13:22 AM] info: [10:13:22 AM] 192.168.99.2 GET /state/all?null {}
[4/18/2021, 10:13:22 AM] info: New socket client connected bxUox9DZeo6kwzsSAAAG -- 192.168.99.2
[4/18/2021, 10:13:22 AM] info: [10:13:22 AM] 192.168.99.2 GET /state/all?null {}
[4/18/2021, 10:13:24 AM] info: [10:13:24 AM] 192.168.99.2 PUT /state/circuit/setState {"id":6,"state":true}
[4/18/2021, 10:13:28 AM] info: [10:13:28 AM] 192.168.100.39 GET /state/features {}
[4/18/2021, 10:13:41 AM] info: [10:13:41 AM] 192.168.99.2 GET /config/all?null {}
[4/18/2021, 10:13:41 AM] info: [10:13:41 AM] 192.168.99.2 GET /config/options/general?null {}
[4/18/2021, 10:13:43 AM] info: [10:13:43 AM] 192.168.99.2 GET /config/options/bodies?null {}
[4/18/2021, 10:13:44 AM] info: [10:13:44 AM] 192.168.99.2 GET /config/options/circuits?null {}
[4/18/2021, 10:13:47 AM] info: [10:13:47 AM] 192.168.99.2 GET /config/options/pumps?null {}
[4/18/2021, 10:13:52 AM] info: [10:13:52 AM] 192.168.99.2 GET /app/config/log {}
[4/18/2021, 10:13:52 AM] info: [10:13:52 AM] 192.168.99.2 GET /config/all?null {}
[4/18/2021, 10:13:58 AM] info: [10:13:58 AM] 192.168.100.39 GET /state/features {}

[image]

rstrouse, 2021-04-18:
Ok I checked in a whole lot of fixes for the virtual controller setup.  The issue here appears to be related to settings in the configuration where we were attempting to automatically detect pumps on the the bus.  This code has now been deprecated to only look for pumps and chem equipment that you manually add from the config pages which I fixed for the virtual contoller.

You should now be able to add pumps, heaters, features, chlorinators, and chem controllers as well as set the items on the setup pages.  You still cannot add schedules because the schedule controller is not yet complete.  It has dependencies that are related to the Nixie controller functions that are coming in future releases.

2delarosa, 2021-04-18:
Okay, will try it now.  Thanks for the serial port activity clarification.  I was wondering if something was wrong with RS-485 adapter.  Will turn on the logging level to verbose after I do the updates.  Just to confirm, I will do the following steps in each of the two folders

`git pull`
`npm install`
`npm run rebuild`

This should update the respective dist.js files that can be then called by PM2

rstrouse, 2021-04-18:
I don't run npm run rebuild I typically run npm start.  I don't know if npm run rebuild actually starts the app.  If it does and you run pm2 start then you will have 2 instances going.

2delarosa, 2021-04-18:
Apps updated.   Did not restore the old poolConfig.json, poolState.json, and the config.json files.  I went and deleted the pump entries and started a new entry but it does not delete them.  I'm thinking that I may have to restore the above files since I use mqtt to verify communication to the pump

[image]
[image]

[npm start njsPC with silly logging.txt](https://github.com/tagyoureit/nodejs-poolController/files/6331930/npm.start.njsPC.with.silly.logging.txt)

2delarosa, 2021-04-18:
I just confirmed speaking to the guests that I can control the pump.  Feature 1 is set to 2900 or 3000 rpms and they confirmed it was loud.  I then disabled it and it dropped either to 2350 or 2500 rpms which they was much quieter.  The reason for the two values is that I have multiple pump entries

rstrouse, 2021-04-18:
I just posted up a fix for the pump circuit display you should be able to see the assignments by tapping or clicking on the pump.  Also if you only have one pump you should only have one entry.  If there is more than one entry for a single pump it will send conflicting messages to it.  Worst case scenario is to blow away the poolState.json and poolConfig.json files then re-add your pumps and features.  This works for the virtual controller instance I set up.
[image]

2delarosa, 2021-04-18:
Is the update on dashPanel, njsPC, or both?

rstrouse, 2021-04-18:
The changes were in njsPC.

tagyoureit, 2021-04-22:
> Feature 1 is set to 2900 or 3000 rpms and they confirmed it was loud. I then disabled it and it dropped either to 2350 or 2500 rpms which they was much quieter.

Pumps themselves should never be "loud".  You might want to check for a blown gasket when you are there in person.  My neighbor had one for a while and it was incredibly annoying.  A change in ~400 rpm shouldn't make a huge difference in the decibels produced by the pump.

Also, I tested this locally and have confirmed everything is working.  Please let us know if you have any other issues.

closed 2021-04-22 by tagyoureit