tagyoureit/nodejs-poolController #281: Equipment discovery issues

emes, 2021-04-28:
I jumped on to issue https://github.com/tagyoureit/nodejs-poolController/issues/278 and @rstrouse fixed.  

What I did not notice was that my pool circuit disappeared.  Body was of course there, but no circuit.  I deleted data/* and re-ran with commit tagged 7.0.1 and have included capture.  Just to be clear, missing pool circuit is the only issue at this release.

With fc0153c384a136e0ccb813ed3a470963da003de2 (realize that this is not released/being worked on, so just as fyi) I'm seeing slightly different issues:

- unused spa body is back
- unused spa circuit is back
- pool circuit still missing
- pump has lost it's "pool" speed and "pool heater" circuits
- a non-existent "intake" valve is back (in addition to the two valves both associated with pool heater)

[image]

[image]

[replay-7.0.1.zip](https://github.com/tagyoureit/nodejs-poolController/files/6394289/replay-7.0.1.zip)

[replay-fc0153c.zip](https://github.com/tagyoureit/nodejs-poolController/files/6394291/replay-fc0153c.zip)

**Pool Equipment**
 - Controller: IntelliTouch
 - Controller Model: i5+3S
 - Controller Firmware Version: 1.160
 - Pump(s) manufacturer and model: IntelliFlow VS
 - Heater(s): gas
 - Valves: Intellivalve (2 both dedicated to pool heater op)

rstrouse, 2021-04-29:
Thanks I’ll have a look.  Just to be sure your model is IntelliTouch i5+3S not i7+3

emes, 2021-04-29:
Right and thanks!

rstrouse, 2021-04-29:
Go ahead and pull njspc.  I am pretty sure we got the model bytes reversed.  We needed to remap a few things to make room for the Nixie extensions and the two bytes got mixed up.

emes, 2021-04-29:
Cool.  All looks good except now it's not seeing circuit id 5 (Aux 4).  Everything else is perfect.  Added a new capture.

[image]

[replay-3ef365e.zip](https://github.com/tagyoureit/nodejs-poolController/files/6396542/replay-3ef365e.zip)

emes, 2021-04-29:
Also noticing that in poolConfig circuit 5 is included in invalidIds:

```
"invalidIds": [
        1,
        5,
        7,
        8,
        9,
        10,
        16,
        17,
        18,
        19
      ]
```

emes, 2021-04-29:
Minor adjustment.  Works for me.
```
diff --git a/controller/boards/IntelliTouchBoard.ts b/controller/boards/IntelliTouchBoard.ts
index 7227c4f..90fdb2d 100644
--- a/controller/boards/IntelliTouchBoard.ts
+++ b/controller/boards/IntelliTouchBoard.ts
@@ -67,9 +67,8 @@ export class IntelliTouchBoard extends EasyTouchBoard {
         if (!eq.shared) sys.board.equipmentIds.invalidIds.merge([1]);
         // Add in all the invalid ids from the base personality board.
         sys.board.equipmentIds.invalidIds.set([16, 17, 18]); // These appear to alway be invalid in IntelliTouch.
-        for (let i = 5; i <= 9; i++) {
-            // This will add all the invalid ids between 5 and 9 that are omitted for IntelliTouch models.
-            if (i === 6) continue;
+        for (let i = 7; i <= 9; i++) {
+            // This will add all the invalid ids between 7 and 9 that are omitted for IntelliTouch models.
             if (i >= eq.maxCircuits - 1) sys.board.equipmentIds.invalidIds.merge([i]);
         }
         // This code should be repeated if we ever see a panel with more than one expansion panel.
@@ -152,4 +151,4 @@ class ITTouchCircuitCommands extends TouchCircuitCommands {
             catch (err) { reject(err); }
         });
     } 
```

rstrouse, 2021-04-29:
Perfect! so it look like in IntelliTouch 2-6 are always active.  Anything above that to 9 is model dependent.  Thanks

closed 2021-04-29 by emes