tagyoureit/nodejs-poolController #282: Not Detecting Intellicenter

Tegguy, 2021-04-28:
This is a brand new setup and when I issue the NPM Start& command it does not detect the pool controller. I'm hooked up to a shared RS-485 port on the intellicenter using JBTek RS-485 adapter into the raspberry pi. I have checked I'm using the right wiring but plan to triple check everything this weekend.

**Pool Equipment**
 - Controller: IntelliCenter
 - Controller Model: Intellicenter i8PS
 - Controller Firmware Version: 
 - Pump(s) manufacturer and model: 2X IntelliFlow 2 VST 011056
 - Chlorinator: IntelliChlor-40
 - Heater(s): AquaCal Heatpump (Heat/Cool)
 - Valves: Intellivalve
 - Any other relevant equipment:
[image]

tagyoureit, 2021-04-28:
I don't see any errors there. If you log into dashPanel or look at the two json files in the data directory you'll likely see data. 

The logging is most likely turned off which is why you aren't seeing anything in the console.

tagyoureit, 2021-04-28:
I do see the Nixie Single Body... try to swap the wires and/or make sure the right port is listed in config.json.

Tegguy, 2021-04-30:
I have tried swapping the wires (+/-), Swapping cables, swaping ethernet pairs, going directly into the RS-485 port as the only item and no luck. I have verified the USB port in the config.json matches the raspberrypi.

I'm going to triple check I have the right ethernet pairs but I'm 99% sure I do.

Tegguy, 2021-04-30:
Triple checked the wires I'm using and I have the correct 2

rstrouse, 2021-04-30:
We just had another person where the adapter was printed incorrectly.  After you make a change you must shut down njspc, delete the files in the data directory and restart njspc in that order.  If it cannot find an OCP it locks itself onto the Nixie controller and stops looking for IntelliCenter.

Tegguy, 2021-05-01:
I have been deleting these files and rebooting the raspberry pi in-between
each change

Tegguy, 2021-05-01:
How do you shutdown the njspc?

rstrouse, 2021-05-01:
If you are starting njsPC via npm start simply close the console window or press ctrl + c.   If you are using pm2 to start it then pm2 stop [whatever you named it]

tagyoureit, 2021-05-01:
You could also check out the troubleshooting tips in the [wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/RS-485-Adapter-Details).

Tegguy, 2021-05-03:
I ended up buying a new rs485 converter so I could see the activity lights
and my 5 minute test with it this morning shows it might solve the problem
since it detected the intellicenter. I will work on it more after work
today but it could be two bad rs485 converters that was the problem all
along (laptop testing pointed this was as well)

rstrouse, 2021-05-03:
There seems to have been a rash of bum adapters lately.

emes, 2021-05-04:
Yup. They seem to work until you touch them. Good thing they're cheap.

rstrouse, 2021-05-04:
I have to wonder if some of them need the eject command before removal.

Tegguy, 2021-05-04:
Problem appears to be resolved with the new rs485 converter

closed 2021-05-04 by Tegguy