tagyoureit/nodejs-poolController #283: Pump Status byte 7 (flow) not being read

jjsander, 2021-04-29:
Byte 7 of the Pump Status is flatlined at zero, both in the MQTT messages where I first discovered the issue as well as API calls. 

curl -X GET localhost:4200/state/pump/1 returns a value of '0' for flow every time. All other parameters appear to be correct in the message response. 

I did verify via Message Manager in the poolController-dashPanel interface that the byte is coming over correctly, in my case '45' [gpm]. 

All that said, thank you for this wonderful app! I may dig around the code to try to figure out where this value gets picked up, but I am a total novice at this stuff so I'm probably more dangerous than helpful...

rstrouse, 2021-04-29:
Which pump model do you have?  Do you have IntelliTouch, IntelliCenter, EasyTouch, or none of the above.

jjsander, 2021-04-29:
I have an EasyTouch panel with an Intelliflow VS pump.

rstrouse, 2021-04-29:
Are you sure it is a VS pump because the flow rate byte(7) is only valid for VSF and VF pumps?

jjsander, 2021-04-29:
No, I am not sure...I was just trying to get into Pentair's warranty site because I know I have the model stored in there by serial number. Apparently they changed things up on that site though... My original contract "spec" was for a "Pentair Intelliflo Variable Speed Pump" and the dashPanel is picking it up as a "Intellflo VS". If I can't find the model info on the Pentair site, I'll pull it off the nameplate when I get home this evening. Sorry about that!

Whatever it is, though, the byte is definitely coming across the RS-485 protocol and it *appears* to be tracking with speed. I suspect it's a calculated value, though, and not instrumentation feedback.

rstrouse, 2021-04-29:
If it is a VS pump it has been our experience that that byte is noise.  If it is a VSF or VF then it contains the flow rate.  If that is the case you may not have it set up correctly in EasyTouch which would make it not right in dashPanel.

You will find the code for this in `/controller/comms/status/PumpStateMessage.ts` line 45.  Byte(5) and byte(6) contain the encoded speed and byte(3) and byte(4) contain the encoded energy (watts).

rstrouse, 2021-05-12:
@jjsander did you figure out which pump you have?  Is this squared away for you?

jjsander, 2021-05-12:
Sorry, yes... It's an Intelliflo VSF. So that may be why I was getting data
on that byte.

jjsander, 2021-05-12:
Ok, I apologize I missed your message from 13 days ago. You are absolutely spot-on, it was set to vs in the easy touch, probably since installation. All is working now, thank you and sorry!

closed 2021-05-12 by jjsander