tagyoureit/nodejs-poolController #284: Can't read data from IntelliCenter

Candurin, 2021-04-29:
**Describe the bug**
I am so close. I have activity LEDs on the USB adapter now (https://www.amazon.com/gp/product/B07SD65BVF).  However, when I start the NodeJS Pool Controller I get stuck on this warning in a continuous loop 
"**[4/29/2021, 10:24:54 AM] warn: The body temperature for Pool cannot be determined. Heater status for this body cannot be calculated.**"

Then I noticed that when I start dashpanel, I see this: 

**info: Server is now listening on 0.0.0.0:5150
192.168.1.48 GET /config/web.services?null
192.168.1.48 GET /config/web.services?null
192.168.1.48 GET /
192.168.1.48 GET /config/web.services?null
192.168.1.48 GET /config/web.services?null
192.168.1.48 GET /options?null
192.168.1.48 GET /config/findPoolControllers?null
done searching for poolController
192.168.1.48 PUT /config/web.services
192.168.1.48 GET /config/web.services?null
192.168.1.48 GET /config/web.services?null
192.168.1.48 GET /options?null
192.168.1.48 GET /
192.168.1.48 GET /config/web.services?null
192.168.1.48 GET /config/web.services?null**

Which seems odd because my Pi is statically set at 192.168.1.50

**Screenshots**

[image]

**Pool Equipment**
 - Controller: IntelliCenter
 - Controller Model: IntelliCenter i5PS
 - Controller Firmware Version: 1.047
 - Pump(s) manufacturer and model: IntelliFlow 2 VST 011056
 - Chlorinator: IntelliChlor-40
 - Heater(s): MasterTemp 300
 - Chemical controller: N/A
 - Valves: Manual

**Desktop (please complete the following information):**
 - OS: macOS Big Sur (11.2)
 - Browser: Safari

**Smartphone (please complete the following information):**
 - Device: iPhone 11
 - OS: 14.5
 - Browser: Safari

rstrouse, 2021-04-29:
Alright now that you have communication with the RS485, perform the following steps.  

1. Shut down njspc.
2. From the directory where nodejs-poolController is installed pull the latest code `git pull` just to make sure you have all the updates.
3. Delete the files in the data directory.  There should be 2 of them poolState.json and poolConfig.json.
4. Restart njspc `npm start`

Now that IntelliCenter is broadcasting status njspc should be able to hear the controller on the bus and create the necessary items to operate as a slave to IntelliCenter.

Candurin, 2021-04-29:
Followed all your steps.  No dice:

[image]

rstrouse, 2021-04-29:
Hmmm.  IntelliCenter is powered up when you start?

Candurin, 2021-04-29:
It's running.  I can access it from the app and website.  

Do you think I should move the communications from the IntelliChlor "daughter card" comm port and combine with another device on the IntelliCenter comm port (perhaps the one shared by the pump)?

rstrouse, 2021-04-29:
This actually looks like the data + and - are reversed.  What will happen then is every message sent will be invalid.  Double check the wiring.  If the connections are good and tight reverse the green and yellow wires on the adapter.

Candurin, 2021-04-29:
Changed yellow and green at the USB adapter:[image]

It still doesn't say IntelliCenter, but that much closer!

rstrouse, 2021-04-29:
After you did that did you delete the files in the data directory and restart?
If you did then lets make sure we are addressing the right USB port.  Sometimes these adapters don't install at ttyUSB0.

Do you know how to list the available ports on the pi?

Candurin, 2021-04-29:
Hold that thought!  I clicked on "reload configuration".  Voila![image]

Which is crazy because I have it connected Yellow to A+ and Green to B- on my adapter.  Maybe the adapter is wired incorrectly?

Either way, thanks!

rstrouse, 2021-04-29:
The reason why we have you switch the wires is because those adapters are rarely documented well and sometimes incorrectly.  Glad you are up and running.  Enjoy!

Candurin, 2021-04-29:
Any ideas how to make my pi autostart the controller and dash panel?  I edited the local file but no luck there.

rstrouse, 2021-04-29:
I use PM2 to start mine.  Check out the wiki for options
https://github.com/tagyoureit/nodejs-poolController/wiki/Automatically-start-at-boot---PM2-&-Systemd

Candurin, 2021-04-29:
@rstrouse PM2 works great!  Everything up and running and auto-starting on reboot.  Again, the help was much appreciated.

rstrouse, 2021-04-30:
I knew we would get you there.

closed 2021-04-30 by rstrouse