tagyoureit/nodejs-poolController #285: Ability to specify which temp probe to use for body temp

theRealMCQwerty, 2021-04-29:
FEATRURE REQUEST

I have 3 temp sensors, one each attached to `Air`, `Water` and `Solar`.
I don't have a solar heater but I do have both the 'water' and 'solar' probes installed in various parts of my setup.
My pad layout is non-standard and as such there are times when there is no flow past the 'water' sensor but there is flow past the 'solar' sensor. Therefore, I would like to programmatically tell the IC which input 'water'/'solar' to use as the probe to read for the water body temperature.

For clarity, this is not as simple as using 'water' for Pool and 'solar' for Spa, it depends on the position of actuators and which pump is running etc. So I would like to control this via the API and perform the logic as to which sensor should be used elsewhere.

Would that be possible?
Also, would it be possible to add additional temp probes and via REM and optionally feed that data to the IC and have it see it as the 'water' temp?

rstrouse, 2021-04-30:
The problem is that IC will not accept an external feed of a temperature sensor.  Your only bet would be to do a bit of hardware shenanigans and fool IC into reading one sensor or the other.  It would be a relatively simple circuit and I think the logic can be done in REM using the appropriate triggers.  The solar sensor does you no good when it comes to the water temp.  Only watersensor 1 is valid for your setup.

theRealMCQwerty, 2021-04-30:
`Your only bet would be to do a bit of hardware shenanigans`
The idea has crossed my mind and your response has convinced me to do it. Need to get the wire cutters out anyway to try out the sweet new virtual light circuit control! 😄

closed 2021-04-30 by theRealMCQwerty