tagyoureit/nodejs-poolController #287: Pool “on/off” button not working

Candurin, 2021-04-30:
Hopefully an easy fix.  The “power button”:

[image]

desn’t seem to “stick.  I see it temporarily flip the SWCG off but then everything comes back on.  The toggle works on the intellicenter app, just not using the dashPanel.

It’s not critical, but, would be nice to have that functionality.

rstrouse, 2021-04-30:
If you hit this button again.  Does it go off.  If it does, this is a problem that also plagues the wireless remote for IntelliCenter.  What happens is that the command to shut it down works and completes but the indoor controller turns it back on.  I believe this is a bug in the IntelliCenter firmware related to their schedule restart function.  For now if it comes back on within a second or two turn it back off and it will stay off (the indoor control panel will not turn it back on).  It is annoying but if they don't fix it on the next firmware release I will put some code in njspc that can play the game when an ICP turns it back on.  This bug was introduced in 1.040 and remained in 1.047.

Candurin, 2021-05-01:
Tried to double tap it.  Exhibits the same behavior.  Goes off and then kicks back on.  It doesn’t seem to turn the pump off, just the swcg, which also kicks back on immediately.

rstrouse, 2021-05-01:
Don’t double tap.  Wait a second and do it again.  This is the same thing that happens with the wireless remote.  I may need to just resend when the controller turns it back on.

Candurin, 2021-05-01:
Unfortunately, no combination works.  Always automatically turns right back on.

I can do it from the IntelliCenter app for the time being.  I’d be glad to test out whatever you need.

rstrouse, 2021-05-01:
I will put together some instructions to get a better picture of what might be going on in your system that is different than mine.

rstrouse, 2021-05-01:
Ok so follow these instructions carefully so I can decipher what is going on.  We are going to capture the information on the RS485 bus to analyze what all the controllers are doing on it.  I suspect that I will find njspc turning the circuit off then another controller on the bus turning it back on.  This is consistent with what happens on i10PS when the schedules have a hand in setting it.

### Step 1:
Turn on the pool from IntelliCenter.  We want to start with the pool being on.

### Step 2:
In dashPanel click on the hamburger menu and select the Logging tab.  Click the Capture Replay button.  When the dialog comes up, **Please uncheck the Capture Configuration Reload** checkbox.  If you don't then IntelliCenter will begin reloading the configuration from the panel.
[image]

### Step 3:
Click the Begin Capture button then click on the background somewhere to dismiss the Settings popover.

### Step 4:
Click Pool to turn it off.  Wait a few seconds and try it one more time.

### Step 7:
Click the hamburger menu in the upper left again and select the Logging Tab.  Then click the Cancel Capture button.  This will download a zip file to your browser which I would like you to upload here.
[image]

Candurin, 2021-05-01:
Ok.  I’m attaching the file as requested.

thanks!

> 
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6410098/replay.zip)

rstrouse, 2021-05-01:
I see what is going on.  I will have a fix shortly.

Candurin, 2021-05-01:
Awesome!  Is it something I’m doing?

rstrouse, 2021-05-01:
You need to publish both port 4200 and 5150 in your port forwarding to get it to work.

Candurin, 2021-05-01:
> You need to publish both port 4200 and 5150 in your port forwarding to get it to work.

I’ll play around some more later.  Both are forwarded but no luck.  Connects just fine when using the external address while on home network, just not while away.

rstrouse, 2021-05-01:
Go ahead and pull njspc.  It should now initialize the pool circuit when starting up on an i5P

DrJeff, 2021-05-01:
> > You need to publish both port 4200 and 5150 in your port forwarding to get it to work.
> 
> I’ll play around some more later. Both are forwarded but no luck. Connects just fine when using the external address while on home network, just not while away.

On your router you have the external IP addressed forwarded to internal Pi with those two ports, What router model? Maybe I can help

Candurin, 2021-05-01:
> On your router you have the external IP addressed forwarded to internal Pi with those two ports, What router model? Maybe I can help

Netgear R7800 (x4s).

Candurin, 2021-05-01:
> Go ahead and pull njspc. It should now initialize the pool circuit when starting up on an i5P

It works!  Again thanks!

rstrouse, 2021-05-03:
Awesome.  Glad we could get you going.  Closing this issue open another if you need more help.

closed 2021-05-03 by rstrouse