tagyoureit/nodejs-poolController #288: Can't detect the intellitouch panel in 7.0.1

fresnoboy, 2021-05-01:
**Describe the bug**
The Panel cannot be detected at start.   Also, the README needs to be updated with doing the npm install with sudo, and how to clear the config once the nixie controller is selected when the intellitouch panel is not detected during initial setup.

**To Reproduce**
Steps to reproduce the behavior:
Attempt to install and connect to an Intellitouch panel.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Packet Capture**
Follow the instructions to complete a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) and attach the resulting zip/log files.

**Pool Equipment**
 - Controller: Intellitouch i9+3
 -
 - Controller Firmware Version: ? (how to tell?
 - Pump(s) manufacturer and model: STA RITE INTELLIFLOW 8 X 160 MULTI SPEED MAIN CIRCULATION PUMP and SPA jet pump
 - Chlorinator: Intellichlor IC40
 - Heater(s): gas 
 - Valves: ?
 - Any other relevant equipment: SwimJets, deckjets, floor cleaner

**Desktop (please complete the following information):**
 - OS: Raspbian
 - Browser Chrome
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

fresnoboy, 2021-05-01:
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/6410177/replay.zip)

This is the captured traffic.

rstrouse, 2021-05-01:
This is extremely helpful.

Run the following commands in the nodejs-poolController directory after you shutdown njspc.
`git checkout master`
`git pull`
`npm install`
`npm start`

rstrouse, 2021-05-01:
After you have done that I would like you to capture another replay.  When you do it leave the capture configuration reload checkbox checked.

fresnoboy, 2021-05-01:
I assume you mean sudo npm install, right?

rstrouse, 2021-05-01:
Yes.  That whole thing has to do with how the repo was installed.  Sadly, the pi user and the root user are linked but it creates the directory owners as root is sudo is used in the beginning.  Most of the time this is because the OS has items that can only be accessed by root.

fresnoboy, 2021-05-01:
Ok, did that, and since I have the screenlogic controller, I cycled a bunch of devices to try and get some fullsome traffic, except for the deck jets that I don't like to run with the pool cover closed.

Attached.  
[replay (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/6410232/replay.1.zip)

fresnoboy, 2021-05-01:
BTW, I still don't understand what you mean by njspc.  Is this installed with the poolcontroller or a nodejs feature?  The README doesn't mention this at all.

rstrouse, 2021-05-01:
Interesting there is no traffic in the message log.  There should be messages if the comm port is communicating in message log.

njspc is just easier to type than nodejs-poolController.

Lets do another replay capture.  Only this time start the capture then shut down njspc then restart it.  You can leave dashPanel open but after it reconnects give it a couple of seconds then hit the Cancel Capture button.  This will force it to go back to the beginning and report in great detail what is occurring on the bus.

fresnoboy, 2021-05-01:
[replay (3).zip](https://github.com/tagyoureit/nodejs-poolController/files/6410262/replay.3.zip)

Done as requested.

fresnoboy, 2021-05-01:
Sorry, did you need me to remove files in the data directory before doing each start and capture?

rstrouse, 2021-05-01:
Yes please.  Soon you will no longer need to do this.

fresnoboy, 2021-05-01:
Ok.  Here you go.   
[replay (5).zip](https://github.com/tagyoureit/nodejs-poolController/files/6410275/replay.5.zip)

BTW, aren't the capture logs stored in the data directory that I am clearing on restart?

rstrouse, 2021-05-01:
The captured logs are actually stored in the logs directory.  There is no communication coming from your adapter whatsoever.  It is opening the port but there isn't even garbage transmitted on it.  This is what happens when both wires on the adapter are either not connected or dead.  If it was connected incorrectly I would at least be seeing invalid packets.  In this case it is crickets.

Double check to ensure everything is hooked up securely.  There is the possibility that the directory clone is bungled and we can try from start with git clone.  Delete the old directory and do the following commands.  This time without sudo so it doesn't set the directory owners as root.
`git clone https://github.com/tagyoureit/nodejs-poolController.git`
`cd nodejs-poolController`
`git pull`
`npm install`
`npm start`

fresnoboy, 2021-05-01:
This is quite odd.  It was working using the old 6.5.1 code before.  I may try and do that pull you mentioned on gitter and see if that sees traffic or not.  I checked the wiring before and it looks properly cabled.  I will take a multimeter and check continuity later, but the screenlogic controller is wired into the intellitouch panel, and the Rpi RS485 interface is cabled into the same jack the screenlogic controller is.

Will work on this and get back to you.

rstrouse, 2021-05-01:
Yeah if it were bad communication that would be one thing but the port is completely silent.

Are you sure its on `/dev/ttyUSB0` it opens but that really doesn't mean anything because the port is always there.

fresnoboy, 2021-05-01:
This is the startup under 6.5.1:

[5/1/2021, 2:29:24 PM] info: Starting up SSDP server
[5/1/2021, 2:29:24 PM] info: Initializing Nixie Controller
[5/1/2021, 2:29:24 PM] info: Nixie Controller Initialized
[5/1/2021, 2:29:24 PM] info: Server is now listening on 0.0.0.0:4200
[5/1/2021, 2:29:24 PM] info: Serial port: /dev/ttyUSB0 request to open successful
[5/1/2021, 2:29:31 PM] info: Searching chlorinators, pumps and chem controllers
RESETTING DATA
[5/1/2021, 2:29:31 PM] warn: dt:Sat May 01 2021 14:29:31 GMT-0700 (Pacific Daylight Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[5/1/2021, 2:29:31 PM] warn: dt:Sat May 01 2021 14:29:31 GMT-0700 (Pacific Daylight Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[5/1/2021, 2:29:31 PM] info: Searching for a pump at address... 96
[5/1/2021, 2:29:31 PM] info: Searching for a pump at address... 97
[5/1/2021, 2:29:35 PM] warn: Message aborted after 4 attempt(s): 16,2,80,0,0,98,16,3
[5/1/2021, 2:29:35 PM] warn: No Chlorinator Found
[5/1/2021, 2:29:37 PM] warn: Message aborted after 2 attempt(s): 165,0,96,33,4,1,255,2,42
[5/1/2021, 2:29:37 PM] info: No pump found at address 96: Message aborted after 2 attempt(s): 165,0,96,33,4,1,255,2,42
[5/1/2021, 2:29:40 PM] warn: Message aborted after 2 attempt(s): 165,0,97,33,4,1,255,2,43
[5/1/2021, 2:29:40 PM] info: No pump found at address 97: Message aborted after 2 attempt(s): 165,0,97,33,4,1,255,2,43
[5/1/2021, 2:29:42 PM] warn: Message aborted after 2 attempt(s): 165,0,144,16,210,1,210,2,234
[5/1/2021, 2:29:42 PM] info: No chemController found at address 144: Message aborted after 2 attempt(s): 165,0,144,16,210,1,210,2,234
[5/1/2021, 2:29:42 PM] info: New socket client connected 2fKGSpTUtiyH5SyVAAAA -- 10.1.10.68
[5/1/2021, 2:29:42 PM] info: [2:29:42 PM] 10.1.10.68 GET /state/all?null {}
[5/1/2021, 2:29:44 PM] warn: Message aborted after 2 attempt(s): 165,0,145,16,210,1,210,2,235
[5/1/2021, 2:29:44 PM] info: No chemController found at address 145: Message aborted after 2 attempt(s): 165,0,145,16,210,1,210,2,235
[5/1/2021, 2:29:46 PM] warn: Message aborted after 2 attempt(s): 165,0,146,16,210,1,210,2,236
[5/1/2021, 2:29:46 PM] info: No chemController found at address 146: Message aborted after 2 attempt(s): 165,0,146,16,210,1,210,2,236
[5/1/2021, 2:29:48 PM] warn: Message aborted after 2 attempt(s): 165,0,147,16,210,1,210,2,237
[5/1/2021, 2:29:48 PM] info: No chemController found at address 147: Message aborted after 2 attempt(s): 165,0,147,16,210,1,210,2,237

What does this mean?

fresnoboy, 2021-05-01:
As for the devices, here is part of the dmesg output:

    2.303586] usb 1-1.4: new full-speed USB device number 3 using xhci_hcd
[    2.439753] usb 1-1.4: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[    2.439787] usb 1-1.4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[    2.439817] usb 1-1.4: Product: USB Serial
[    5.240215] usbserial: USB Serial support registered for generic
[    5.245818] usbserial: USB Serial support registered for ch341-uart
[    5.253912] usb 1-1.4: ch341-uart converter now attached to ttyUSB0

That would be the RS485 adapter right?

rstrouse, 2021-05-01:
Yeah it too is not communicating on the port.  All those message aborted is njspc calling out to the wild which indicates the port isn't communicating.  We got a little smarter in 7.x and didn't indiscriminately start polling devices that we didn't already know existed.

That ch341-uart is definitely the adapter.  Is it listed when you run `lsusb`

rstrouse, 2021-05-01:
Hey just for giggles, is this a pi4?  Turn off Bluetooth if it is.

fresnoboy, 2021-05-02:
Ok, I turned bluetooth off (and yes the device shows up in lsusb), rebotted, and then started the pool controller.  There are a few packets collected this go around it appears.

Attached is the reply log.

[replay (6).zip](https://github.com/tagyoureit/nodejs-poolController/files/6410752/replay.6.zip)

fresnoboy, 2021-05-02:
Ok, I checked continuity with a voltmeter, and the yellow and green wires are connected properly to the intellitouch panel, as well as the screenlogic controller that seems to work perfectly well.  So no wiring issues as far as I can tell.

Does the njspc contain any new drivers that are installed in the system?  I am trying to figure out why things are not being read at all but the hardware didn't change.

fresnoboy, 2021-05-02:
I guess next step is to reformat the raspberry pi from scratch (it's only used for this function) and try it again.  I'll use a raspberry pi 3 that I have lying around to make sure there are fewer RFI issues.  

The other thing to try would be getting another RS485 interface and seeing if it died for some reason.

rstrouse, 2021-05-03:
My guess is the Bluetooth thing.  On one of my pi’s I had an issue getting an adapter to start.  I turned off Bluetooth and it started working.  I reimaged that pi with a newer version and the issue didn’t come back.

fresnoboy, 2021-05-03:
This is very weird.  I can look at the logfile myself and it's not seeing anything.  Yet this hardware worked just fine until recently.  And the screenlogic controller is plugged into the same wires and works just fine.  I wonder if the rs485 adapter was friend somehow.

I'll get an RPi 3 configured up and see if I can have more luck with that and a new serial adapter if needed.

rstrouse, 2021-05-03:
You know this wouldn’t be the first time I have seen these adapters just fail.  I have tossed a couple of them.  However as you can imagine I have many of them going all the time.

fresnoboy, 2021-05-05:
Ok, I got a replacement RS485 controller.  I swapped out the wiring and the system came up and detected the panel just fine.  Looks like it found the pool equipment as well.  I'm not sure the IC40 SWG is properly represented - I can't turn it on, but it may be linked to the pool pump relay.

Did you want me to grab a replay log?

tagyoureit, 2021-05-05:
These have a flow sensor in them so they need the pool to be on; they are also powered based on the pool circuit of wired through the OCP. Does it work when the pool circuit is on?

fresnoboy, 2021-05-05:
Yes, that's right, but it still doesn't indicate on when the pool pump is running.  I confess I don't what understand the settings for that unit mean in terms of setpoints.  The radio button seem to act as an on/off.

tagyoureit, 2021-05-05:
It will show green when it is powered/communicating and when the set point is >0.  
[image]

And it will show off if the body is off or it is set at 0 output.
[image]

rstrouse, 2021-05-13:
Are you all squared away?   The indicators next to the chemistry, pumps, and schedules only indicate when the particular piece of equipment is running.  The are not to turn them off or on.  Your chlorinator should come on when the assigned body is turned on and the chlorinator responds that it is currently on.  This response from the chlorinator can take minutes because it will shut off the cell as it calculates the salinity and checks for flow.

closed 2021-05-23 by rstrouse